elFarto/nvidia-vaapi-driver #78: Bottom bar of video weird decoded

HE7086, 2022-04-23:
I have noticed this bug quite a while but didn't find a way before to consistently reproduce it. I messed around with it today and seemed to at least find some more reliable ways. So the issue is, the frame of video is not correctly decoded, but it only applies to the very bottom bar.

Sometimes the bottom bar is blacked out. (Or maybe the entire frame is moved up a few pixels? the player sizes are the same, so some pixels are definitely missing)
[image]
(h264 codec, video from [here](https://www.youtube.com/watch?v=ldJ-52jxL-0))

Sometimes the bottom bar is blurred. Looks like it repeats the pixels of the last correctly decoded line. It may not seem obvious in screen capture, but very clear in the video when objects are moving.
[image]
(m3u8/hls, video from [here](https://developer-tools.jwplayer.com/stream-tester))

I haven't encountered the issue with vp9/av1 codecs, only h264 is affected. Some videos trigger this bug consistently while others trigger the bug after a few refreshes. Some videos like twitch streams never triggered this for me. Dragging the progress bar sometimes solves the issue, I'm not entirely sure how this works though.

mpv does not have this issue using vaapi, have tried downloading the video and play them locally, the bottom bar is correctly rendered. Not sure whether this is a driver bug or a firefox bug, since I have been seeing this for a few months now IIRC.

here's the log playing the first video above while triggering the bug: [h264.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8547637/h264.txt)

<details>
<summary>My system info, if that helps</summary>

```
System:
  Kernel: 5.17.4-zen1-1-zen arch: x86_64 bits: 64 compiler: gcc v: 11.2.0
    Desktop: KDE Plasma v: 5.24.4 Distro: Arch Linux
Graphics:
  Device-1: NVIDIA GA102 [GeForce RTX 3080 Ti] driver: nvidia v: 510.60.02
  Display: x11 server: X.Org v: 21.1.3 with: Xwayland v: 22.1.1 driver: X:
    loaded: nvidia gpu: nvidia,nvidia-nvswitch resolution: 3840x2160~60Hz
  OpenGL: renderer: NVIDIA GeForce RTX 3080 Ti/PCIe/SSE2
    v: 4.6.0 NVIDIA 510.60.02
```

```
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
[21367-21367] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55aaf61b3df0 10
[21367-21367] ../src/vabackend.c:1645       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
[21367-21367] ../src/export-buf.c: 164       findGPUIndexFromFd Looking for GPU index: 0
[21367-21367] ../src/export-buf.c: 175       findGPUIndexFromFd Found 3 EGL devices
[21367-21367] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[21367-21367] ../src/export-buf.c: 136 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
[21367-21367] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[21367-21367] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[21367-21367] ../src/export-buf.c: 103                reconnect Reconnecting to stream
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
[21367-21367] ../src/vabackend.c:1626              nvTerminate Terminating 0x55aaf61b3df0
[21367-21367] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[21367-21367] ../src/export-buf.c:  82          releaseExporter Done releasing frames
```
</details>

elFarto, 2022-04-23:
Ah, I've been noticing this aswell recently, but was unsure if it was the driver as I was only seeing it on reddit videos. That's likely due to most of the videos sources I watch preferring VP9, not H.264.

I'm going to take a wild guess take you're using Firefox 99?

HE7086, 2022-04-23:
Currently I'm using firefox dev 100.0b9, but I recall seeing this quite a while ago. I'll try install some older version of firefox and see if that matters.

elFarto, 2022-04-23:
I've just tested the same video on FF98 and FF99. FF98 doesn't have the issue, so this issue was introduced since then. I'll raise the issue to Mozilla.

HE7086, 2022-04-23:
Ok just reinstalled FF97 from cached packages and the issue is gone. So it seems FF99+ have got the issue

elFarto, 2022-04-23:
I've raised that as [1766124](https://bugzilla.mozilla.org/show_bug.cgi?id=1766124) with Mozilla. Thanks for raising this, and for the helpful screenshots!

HE7086, 2022-06-30:
firefox 102 stable is released and this bug is [fixed](https://bugzilla.mozilla.org/show_bug.cgi?id=1759784)

closed 2022-06-30 by HE7086