elFarto/nvidia-vaapi-driver #79: Low framerate for some YouTube videos

bcrescimanno, 2022-05-03:
OS: Arch Linux
DE: Gnome on X11
Firefox version: 100.0-1 (Arch package)
nvidia-vaapi-driver: [AUR -Git package](https://aur.archlinux.org/packages/nvidia-vaapi-driver-git)
Card Model: Nvidia RTX 3080ti
Nvidia driver version: 510.68.02-1 (Arch package)

With Firefox 100 hitting in Arch today, I decided to try out this driver now that the bug that was happening FF99 was fixed. On some videos, playback is working as expected. On others, the video playback framerate is very low. I don't know if this is a specific codec issue (I don't know how to check which codec is in use on a YouTube video.

Here's an example of a [video with very slow framerate](https://www.youtube.com/watch?v=LFr0ELa6tkk) when this driver is enabled. Here's the [log output ](https://pastebin.com/UaBc86SR) from a quick session where i played a few seconds of the video.

The command line I used for this test:

`$ LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 NVD_LOG=vid.log firefox`

elFarto, 2022-05-03:
I've just tried it with FF100 downloaded from [Mozilla](https://ftp.mozilla.org/pub/firefox/releases/100.0/linux-x86_64/), and that works fine with that video. I'm a little surprised that a 24fps video would cause any problems.

The log file you attached looks fine, I see about 40ms between frames, which is exactly what you'd expect from a 24fps video.

Is the video just playing back in slow motion, or is it dropping the frames (does the video seem to skip between them)?

One thing you can try, is to use a slightly older version of this driver, either commit `51aaf5e2` or tag `v0.0.5`. I've recently committed a large change to how frames are decoded which might be the cause of the issue.

bcrescimanno, 2022-05-03:
> Is the video just playing back in slow motion, or is it dropping the frames (does the video seem to skip between them)?

Good question. I just looked again to be sure and it's a slow motion / jutter effect--but, from what I can tell, no frames are being dropped. 

I'll check that older version and see if the results are different. I'll also try to do a full system reboot in case an older nvidia driver is loading or something like that. Any other data I can provide?

elFarto, 2022-05-03:
BTW, you can right click on the Youtube video and click 'Stats for nerds' to see how it's performing/how many frames have been dropped.

bcrescimanno, 2022-05-03:
Thanks for that tip! I can confirm that no frames are dropped. I can also confirm that I'm seeing this effect for all videos at 24fps regardless of codec (same effect for a VP9 and AV1 encoded video). Anything at 30 or 60fps appears to play normally.

bcrescimanno, 2022-05-03:
OK, last update from me for a bit: I switched over to 0.5 and the issue persists so it doesn't seem to be related to the latest commit. However, I found another video by the same creator that is also 1920x1080@24fps and AV1 encoded that works fine.

[Slow Motion Video](https://youtu.be/LFr0ELa6tkk)
[Normally Playing Video](https://www.youtube.com/watch?v=kGlfjpPDQ2o)

elFarto, 2022-05-03:
Just to be clear, in the Stats for nerds, does it say AVC1 or AV01 in the slow motion video? The former is H.264, the latter is AV1.

bcrescimanno, 2022-05-03:
`Codecs av01.0.08M.08 (399) / opus (251)`

elFarto, 2022-05-03:
Looks like it could be related to this [Firefox issue](https://bugzilla.mozilla.org/show_bug.cgi?id=1758948).

bcrescimanno, 2022-05-03:
Good find; the results absolutely sound the same. I'm trying to find some AV1 encoded 30fps videos to verify that the issue applies there as well. It's super noticeable on 24fps video and I can't help but wonder why it's working for you and not me--gotta be some kind of delta there.

Fxzzi, 2022-05-30:
Also experiencing this exact issue. Hopefully it's fixed soon.

elFarto, 2022-06-19:
Looks like a fix has been committed for this issue, and is available in [this](https://archive.mozilla.org/pub/firefox/nightly/2022/06/2022-06-16-09-30-51-mozilla-central/) nightly build. If you can test it out when you get a chance, and let me know if it's fixed.

elFarto, 2022-12-07:
This issue appears to be fixed now, closing issue.

closed 2022-12-07 by elFarto