elFarto/nvidia-vaapi-driver #80: Laggy video on WhatsApp Web

guihkx, 2022-05-06:
Sometimes, videos from WhatsApp are rendered like this:

https://user-images.githubusercontent.com/626206/167218177-fa5e066d-101d-4fe5-8519-17ec1cfbcea2.mp4

This is the actual video:

https://user-images.githubusercontent.com/626206/167217760-5a682c26-7e1e-42b2-900d-e5f81c6f80c5.mp4

The command I used to get the logs:

```
$ NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox &> firefox.txt
```

- Log file: [nvidia-vaapi.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8643674/nvidia-vaapi.txt)

Other info:

- Arch Linux
- VAAPI driver: 96f0428
- Firefox 100
- NVIDIA GTX 660 (driver 470.103.01)

elFarto, 2022-05-07:
The log file is showing a lot of destroying and recreating contexts, which might be triggering a bug in the new multi-threading code. Can you try with the previous commit `51aaf5e2` and see if that has the same issue?

guihkx, 2022-05-07:
Results are the same (visually at least) with 51aaf5e2:

- [firefox.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8645611/firefox.txt)

elFarto, 2022-05-07:
Do you have an entries in `about:crashes` that correlate to you playing back that video? If you do can you submit it, and paste the link here?

guihkx, 2022-05-07:
There you go, generated this one right now: https://crash-stats.mozilla.org/report/index/a70415b1-d45e-4f4f-9302-948ee0220507

elFarto, 2022-05-07:
Well, that crash dump doesn't show it crashing in the driver, so it might be a Firefox issue, specifically [this one](https://bugzilla.mozilla.org/show_bug.cgi?id=1762725). It looks like a fix has been committed for FF101, but hasn't been included in a nightly build yet.

Once that fix is in a build, could you retest with it?

guihkx, 2022-05-11:
Firefox nightly 102.0a1 just got out and unfortunately it still crashes for me. Logs:

- [firefox-102.0a1.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8666572/firefox-102.0a1.txt)
- https://crash-stats.mozilla.org/report/index/af7c539a-8e36-4b42-81da-2bbe20220511
 
Still testing on commit 96f0428

guihkx, 2022-05-11:
Just for peace of mind I also tested on a commit earlier (51aaf5e), but it still crashed:

- [firefox-102.0a1-51aaf5e.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8666621/firefox-102.0a1-51aaf5e.txt)
- https://crash-stats.mozilla.org/report/index/489dbfa0-b5e8-406d-9c1e-a1dde0220511

elFarto, 2022-05-11:
Interesting, the errors on those log files are different with both of them crashing in this driver (well not really crashing, we forcefully exit when we get a CUDA error):
```
[90134-90431] ../src/export-buf.c: 325          allocateSurface cuda error 'out of memory' (2)
```
Can you try running `nvidia-smi pmon -s um` in a terminal while reproducing the issue? That'll show how much memory is being use just before crashing.

guihkx, 2022-05-11:
`nvidia-smi` says that command is not supported on my GPU, but I monitored using the `nvidia-settings`utility and indeed I run out of memory when the video is playing (I have 2 GB total)

[image]

One weird thing I noticed after running out of memory and something crashing in Firefox: If I close the video and try to play it back again, it works without any apparent issues. I have a new log file where I did exactly that:

- [firefox-102.0a1.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8671110/firefox-102.0a1.txt)

The second successful playback starts exactly at line 13115

elFarto, 2022-05-11:
Ok, so it seems that we're leaking some resource that leads to memory exhaustion and a crash. It working after the crash is expected, there's no actual problem playing the video, but without any memory nothing's going to work.

Unfortunately I think this is an issue that's been hanging around a while, and it may not be easy to fix. The method I'm using to get the image out of CUDA and into EGL so I can export them doesn't take kindly to you deleting it. In fact the documentation straight up says you can't do it. That coupled with your use of the 470 driver means I can't use GBM to get around it.

I'll see if I can reproduce it on the 500 series drivers.

elFarto, 2022-05-11:
Looking into it a bit more, I'm not entirely certain it is leaking images. The leak I thought I saw before seems to be caused by it crashing, and therefore not releasing them as normal.

I've created a new branch [max_instances](https://github.com/elFarto/nvidia-vaapi-driver/tree/max_instances) which has support for a new environment variable `NVD_MAX_INSTANCES`. This will stop Firefox from creating too many instances by failing to initialise. Set it to something low like 2 or 3 and see if that helps with the issue.

guihkx, 2022-05-11:
Nice, that seems to have fixed both the crash and the high VRAM usage... Are there any downsides to this? 🤔 

It doesn't crash or lag even in Firefox 100 now! Logs just in case:

- [firefox-100-stable.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8672826/firefox-100-stable.txt)

guihkx, 2022-05-11:
Wait haha

I don't think the NVIDIA VAAPI driver is actually being used though, I might've messed something up.

guihkx, 2022-05-11:
I was testing this on Firefox stable, which I had `widget.dmabuf.force-enabled` set to `fals` because of the crashes. I forgot to turn it back on.

On Firefox 100 (stable) it still crashes, but I believe it's due to the bug you mentioned earlier. Nevertheless, logs below:

- [firefox-100-stable.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8672913/firefox-100-stable.txt)
- https://crash-stats.mozilla.org/report/index/1833625e-cd6c-4329-b2af-5529e0220511

However, on Firefox nightly 102.0a1, everything seems to work:

- [firefox-102.0a1.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/8672919/firefox-102.0a1.txt)

I have set `NV_MAX_INSTANCES` to `3`. I don't actually know what this does, but as long as it doesn't affect anything else I'm very happy with this fix/workaround!

elFarto, 2022-05-11:
`NVD_MAX_INSTANCES` just limits how many concurrent contexts can be created in a single process. It's not a great fix, but it should stop Firefox from attempting to use unbounded memory, esp. useful for older GPUs with less memory on sites with lots of videos.

guihkx, 2022-05-11:
Gotcha! Would it be reasonable to turn this option into a default for R470 drivers? Unless it's too complicated to implement it of course

elFarto, 2022-05-11:
Unfortunately there's no easy way for me to detect the version of the NVIDIA driver (at least that would work reliably within Firefox's sandbox if we ever get there). And this only effects GPUs with limited memory, not the driver version specifically.

So for the moment I'll just leave it as an option.

PR #83 (2022-05-14): Add option to limit maximum instances
This PR adds the `NVD_MAX_INSTANCES` option. This option will prevent more than the specified amount of instances of the driver running within a single process. This option is mostly to prevent Firefox from created unbounded amount of instances and exhausting VRAM.

Fixes #80
closed 2022-05-14 by elFarto