elFarto/nvidia-vaapi-driver #82: vaapi version mismatch and driver errors

conorlmcbride, 2022-05-11:
System Info:
Firefox version: 100.0 (64-bit)
Ubuntu 22.04 15.05 kernel
RTX 3080
Nvidia Driver: 510.60.02
nvidia_drm.modeset=1 set
env:
```
MOZ_DISABLE_RDD_SANDBOX=1
MOZ_X11_EGL=1
LIBVA_DRIVER_NAME=nvidia
```
I've got some issues getting firefox (and other applications) to recognize the vaapi nvidia drivers correctly. I initially had some trouble removing VDPAU and installing NVDEC but **vainfo** is now reporting NVDEC is installed correctly:
<details><summary>vainfo</summary>

```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/local/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
```
</details>
However when I run firefox I can't get hardware decoding to work, and it's reporting an older version of va-api and some errors.
I have tested 1080p/4k videos in youtube, vimeo, twitch, and a few other locations with the same results. Videos play but fall back to software decoding. GPU utilization remains low.
<details><summary>firefox (vaapi 1.7.0) errors</summary>

```Gtk-Message: 11:39:27.904: Failed to load module "xapp-gtk3-module"
Gtk-Message: 11:39:27.937: Failed to load module "canberra-gtk-module"
Gtk-Message: 11:39:27.938: Failed to load module "canberra-gtk-module"
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
libva info: VA-API version 1.7.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
```
</details>

VLC at least reports the same newer vaapi version but still returns errors. I have tested various H264/HEVC files with the same results. Videos usually play with poor performance and fall back to software decoding. GPU utilization remains low. 

<details><summary>vlc (vaapi 1.14.0) errors</summary>

```
[00007fda04004910] gl gl: Initialized libplacebo v4.192.1 (API v192)
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/local/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[00007fda04004910] glconv_vaapi_x11 gl error: vaDeriveImage: operation failed
[00007fda006e7320] main video output error: video output creation failed
[00007fda18cc9c50] main decoder error: failed to create video output
[00007fda04099d90] gl gl: Initialized libplacebo v4.192.1 (API v192)
[00007fda04099d90] gl gl: Initialized libplacebo v4.192.1 (API v192)
```
</details>

I think hardware decoding is working with mpv player. I notice the "compute" type showing up in nvtop and cpu usage is fairly low for that process compared to when I use another video player. For a few videos I only see a green screen with audio, but I think that's an unrelated issue. I'm not exactly sure where to go from here.

elFarto, 2022-05-11:
It's unlikely to work under vlc, as that's not something I've been testing with. The green output with mpv is related to not having the latest git version of it, which has the fix for that issue.

Can you run Firefox with the following environment variables set: `NVD_LOG=1 MOZ_LOG="Dmabuf:5, PlatformDecoderModule:5"` and post the output?

conorlmcbride, 2022-05-11:
Ok I'm not too worried about the VLC thing I just noticed a bunch of errors after switching from VDPAU and thought they might provide a troubleshooting clue. I wanted to make sure I could actually get it working somewhere before I tackled all of the firefox stuff.

<details><summary>Here is the beginning bit I think is probably most relevant</summary>

```
Gtk-Message: 14:35:44.229: Failed to load module "xapp-gtk3-module"
Gtk-Message: 14:35:44.258: Failed to load module "canberra-gtk-module"
Gtk-Message: 14:35:44.259: Failed to load module "canberra-gtk-module"
[Parent 12939: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 12939: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 12939: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
[Parent 12939: Main Thread]: D/Dmabuf DMABuf is enabled
[Parent 12939: Main Thread]: D/Dmabuf VAAPIGetAcceleratedProfiles()
libva info: VA-API version 1.7.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
[Parent 12939: Main Thread]: D/Dmabuf   Failed to initialise VAAPI connection: unknown libva error.
[Child 13502: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 13502: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Child 13502: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD128 error Permission denied
[Child 13502: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 13502: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
```
</details>
Here is the full log where I open a firefox window, navigate to youtube, and start playing a video: https://pastebin.com/uLAdr3nw

elFarto, 2022-05-11:
Ok, something is very wrong here. Something like this should be logged directly after `libva info: Trying to open...`:
```
     14637.849173988 [59438-59438] ../src/vabackend.c:1775       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fefbdc1e980 31
```
Can you run: `ls -l /usr/lib/x86_64-linux-gnu/dri/`?

Also of concern is this `Failed to open drm render node /dev/dri/renderD128 error Permission denied`. Can you try running `NVD_LOG=1 vainfo --display drm`?

conorlmcbride, 2022-05-11:
<details><summary>ls -l /usr/lib/x86_64-linux-gnu/dri/</summary>

```
total 436652
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 crocus_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 d3d12_dri.so
-rw-r--r--  6 root root 17133096 Mar 16 04:07 i830_dri.so
-rw-r--r--  6 root root 17133096 Mar 16 04:07 i915_dri.so
-rw-r--r--  6 root root 17133096 Mar 16 04:07 i965_dri.so
-rw-r--r--  1 root root  1764040 Jul 11  2020 i965_drv_video.so
-rw-r--r--  1 root root  8004216 Apr  4 16:23 iHD_drv_video.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 iris_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 kms_swrast_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 nouveau_dri.so
-rw-r--r--  3 root root 11206848 Apr 14 05:28 nouveau_drv_video.so
-rw-r--r--  6 root root 17133096 Mar 16 04:07 nouveau_vieux_dri.so
-rwxr-xr-x  1 root root   498032 May 10 12:09 nvdec_drv_video.so
-rwxr-xr-x  1 root root   497584 May 10 12:15 nvidia_drv_video.so
-rw-r--r--  6 root root 17133096 Mar 16 04:07 r200_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 r300_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 r600_dri.so
-rw-r--r--  3 root root 11206848 Apr 14 05:28 r600_drv_video.so
-rw-r--r--  6 root root 17133096 Mar 16 04:07 radeon_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 radeonsi_dri.so
-rw-r--r--  3 root root 11206848 Apr 14 05:28 radeonsi_drv_video.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 swrast_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 virtio_gpu_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 vmwgfx_dri.so
-rw-r--r-- 12 root root 24990520 Apr 14 05:28 zink_dri.so
```
</details>
<p><p>

<details><summary>NVD_LOG=1 vainfo --display drm</summary>

```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/local/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
[19838-19838] ../src/vabackend.c:1642       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5612e6105440 31
[19838-19838] ../src/export-buf.c: 162       findGPUIndexFromFd Looking for drmDeviceIndex: 0
[19838-19838] ../src/export-buf.c: 175       findGPUIndexFromFd Found 3 EGL devices
[19838-19838] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[19838-19838] ../src/export-buf.c: 190       findGPUIndexFromFd Found drmDeviceIndex: 0
[19838-19838] ../src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
[19838-19838] ../src/export-buf.c: 270             initExporter Driver doesn't support 16-bit surfaces
[19838-19838] ../src/export-buf.c: 103                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
[19838-19838] ../src/vabackend.c:1626              nvTerminate Terminating 0x5612e6105440
[19838-19838] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
[19838-19838] ../src/export-buf.c:  82          releaseExporter Done releasing frames
```

</details>

elFarto, 2022-05-12:
Ok, I think I see the problem, or at least part of the problem. `vainfo` is looking in `/usr/local/lib/dri/` and Firefox is looking in `/usr/lib/x86_64-linux-gnu/dri/`. Did you build libva yourself? If so you'll want to remove the old install in `/usr/local` and rebuild it with `--prefix=/usr`. You'll also want to double check that it installs the lib into `/usr/lib/x86_64-linux-gnu/libva.so.*`, Ubuntu/Debian use multi-arch lib directories which may not be automatically detected when you build from source.

conorlmcbride, 2022-05-12:
I installed a handful of packages from apt including libva2, libva-dev, libva-x11-2, libva-drm2 and it installed them to `/usr/lib/x86_64-linux-gnu/` however initially I was following this guide: https://github.com/marioroy/nvidia-acceleration-on-ubuntu and it looks like it builds libva manually. I can remove the `/usr/local/lib/dri` libva but I'm not sure how to tell libva to use the correct `/usr/lib/x86...etc` one.

elFarto, 2022-05-12:
`/usr/local` is hardcoded in libva at compile time in the `scripts/100-build-libva` script. It looks like those instructions also want you to set `LIBVA_DRIVERS_PATH=/usr/local/lib/dri` which would override where libva looks for the drivers.

Those instructions seem fine, but they've very opinionated, and make some different choices, like installing this driver as `nvdec_drv_video.so`, rather than `nvidia_drv_video.so`. So it's better to follow those instructions, or ignore them completely, trying to mix and match will likely bite you in the ass.

marioroy, 2022-06-30:
Ubuntuzilla updated Firefox to 102.0. I removed the package including the apt source entry and trusted key from apt. The following instructions was tested on Ubuntu 22.04.

Remove Firefox snap package.
```bash
sudo snap remove --purge firefox
```
The NVDEC VA-API driver does not work in Firefox 102.0. Obtain and install Firefox 101.0.1 and dependencies.
```bash
cd ~/Downloads
wget http://http.us.debian.org/debian/pool/main/f/firefox/firefox_101.0.1-1_amd64.deb
wget http://http.us.debian.org/debian/pool/main/n/nss/libnss3_3.79-1_amd64.deb
wget http://http.us.debian.org/debian/pool/main/n/nspr/libnspr4_4.34-1_amd64.deb

sudo apt install ./firefox_101.0.1-1_amd64.deb ./libnss3_3.79-1_amd64.deb ./libnspr4_4.34-1_amd64.deb
```
Prevent Firefox from updating.
```bash
sudo apt-mark hold firefox
dpkg --get-selections firefox
```

Once HW acceleration is working again (hoping Firefox 103), one can remove the hold and follow the [instructions](https://www.omgubuntu.co.uk/2022/04/how-to-install-firefox-deb-apt-ubuntu-22-04) at Mozilla.

```bash
sudo apt-mark unhold firefox
dpkg --get-selections firefox
```

The [nvidia-acceleration-on-ubuntu](https://github.com/marioroy/nvidia-acceleration-on-ubuntu) repo works on Ubuntu 22.04 (running X11 with NVIDIA proprietary driver). Firefox and Chromium-based browsers utilize the video engine while playing a video.

marioroy, 2022-07-01:
Hardware decode acceleration is working in [Firefox 103.0b3](https://www.mozilla.org/en-US/firefox/103.0beta/releasenotes/) using the NVDEC VA-API driver.

marioroy, 2022-07-11:
I updated the guide at [https://github.com/marioroy/nvidia-acceleration-on-ubuntu](https://github.com/marioroy/nvidia-acceleration-on-ubuntu) to install Firefox from PPA on Ubuntu 21.10 and 22.04. Hardware acceleration is broken sadly in Firefox 102 using NVIDIA graphics. So, I extended the Epilogue section on how to obtain the latest beta. The `~/bin/run-firefox` script now defaults to `~/firefox/firefox` otherwise `/usr/bin/firefox`.

marioroy, 2022-07-12:
I purchased a RTX 3070 from eBay when the price dropped. Setting `media.ffvpx.enabled` to `false` caused AV1 decode acceleration to not work on Ubuntu. I reverted the change for my Ubuntu repo. AV1 acceleration works great.

```text
media.ffvpx.enabled                            true
media.av1.enabled                              true
```

marioroy, 2022-07-28:
Ubuntu updates has landed Firefox 103.0 for Ubuntu 20.04 and 22.04. Hardware acceleration is working again. Be sure to set `media.ffmpeg.vaapi.enabled` to `true` via `about:config`. Optionally, set `media.av1.enabled` to `false` if your graphics card cannot decode `AV1`.

elFarto, 2022-12-07:
Closing issue as it seems to be caused by old/mismatched libva versions/builds.

closed 2022-12-07 by elFarto