elFarto/nvidia-vaapi-driver #84: How to know if this is working

Fxzzi, 2022-05-15:
Whilst playing an 8K Video from youtube, here are the screenshots of nvtop and bottom.

[image]

How can I tell if it's working or not? On YT stats for nerds, it reports 1 frame being missed every 1 or 2 seconds.

joebonrichie, 2022-05-15:
When a video is playing: Look for firefox using G+C in `nvidia-smi` (graphics + compute) opposed to just graphics. Also look for 'Video Engine Utilization' in nvidia-settings for GPU n.

I don't use nvtop but presumably it should show the firefox process using Compute as a type.

Fxzzi, 2022-05-15:
[image]
How does this look?

Fxzzi, 2022-05-15:
after some further testing I have seen that at 4k, the GPU is being used but at 8K, the Type is instead graphics when it should be compute + graphics here is an example:
[image] 8K
[image] 4K

elFarto, 2022-05-16:
For 4k it looks like the GPU is being used, but 8k it's not. There are ongoing issues with 8k video and Firefox. [This](https://bugzilla.mozilla.org/show_bug.cgi?id=1750388) issue landed a change a few days ago to fallback to software decoding if hardware decoding was too slow.

Looks like nvtop has an option to show `GPU decoder rate`, enabling that should better show what's using the hardware decoder. `nvidia-smi pmon` also shows what's using the decoder.

raspin0, 2022-05-23:
[image]
Use GWE

Fxzzi, 2022-05-27:
Thanks everyone! It seems that with librewolf (my firefox fork of choice) only the compute is being shown, not graphic also, does this mean it isn't working? Or is it still working fine? The command I use to launch librewolf with vaapi is `env LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 EGL_PLATFORM=x11 librewolf`

also an unrelated problem that I fixed: librewolf would not open after a suspend with the environment variables set. This was fixed by enabling 3 systemd services; `nvidia-suspend.service`, `nvidia-resume.service` and `nvidia-hibernate.service`.

elFarto, 2022-05-27:
If you're using nvidia-smi, both G+C are needed. If you don't see any usage in the `dec` column of  `nvidia-smi pmon` or `nvidia-smi` then it's not using the hardware.

Fxzzi, 2022-05-27:
> If you're using nvidia-smi, both G+C are needed. If you don't see any usage in the `dec` column of `nvidia-smi pmon` or `nvidia-smi` then it's not using the hardware.

Okay, I guess librewolf is just broken with it. I've switched to firefox with a custom user.js and it's now showing C+G on firefox with DEC usage at about 6% for 4k.

elFarto, 2022-06-05:
I've updated the README with some instructions on how to verify if the driver is working.

closed 2022-06-05 by elFarto

pc00per, 2022-07-21:
@Fxzzi wait for librewolf v103 (or firefox-stable 103 ingeneral). v103 has patches with working hw decode, as of currently on 102 it lacks.
ETA for firefox-stable 103 is [26th of this month](https://wiki.mozilla.org/Release_Management/Calendar#Future_branch_dates). librewolf might take another week to get it merged.