elFarto/nvidia-vaapi-driver #85: state=DECODING_METADATA Decode metadata failed, shutting down decoder

freakedenough, 2022-05-21:
`besitzer@MSI:~/.firefox/firefox$ ./firefox-bin
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       785.676968193 [8084-8084] ../src/vabackend.c:1775       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f8fd08fe4c0 31
       785.676971383 [8084-8084] ../src/vabackend.c:1778       __vaDriverInit_1_0 Now have 0 (0 max) instances
       785.679577475 [8084-8084] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 0
       785.680739329 [8084-8084] ../src/export-buf.c: 172       findGPUIndexFromFd Found 3 EGL devices
       785.680802410 [8084-8084] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       785.680806650 [8084-8084] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 0
       785.680808790 [8084-8084] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
       785.752822132 [8084-8084] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
       785.752830822 [8084-8084] ../src/export-buf.c:  99                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
       785.772976836 [8084-8084] ../src/vabackend.c: 556           nvCreateConfig got profile: 6 with 0 attributes
       785.772989626 [8084-8084] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
       785.772997506 [8084-8084] ../src/vabackend.c: 556           nvCreateConfig got profile: 13 with 0 attributes
       785.773002577 [8084-8084] ../src/vabackend.c: 556           nvCreateConfig got profile: 14 with 0 attributes
       785.773007227 [8084-8084] ../src/vabackend.c: 556           nvCreateConfig got profile: 19 with 0 attributes
       785.773012777 [8084-8084] ../src/vabackend.c:1752              nvTerminate Terminating 0x7f8fd08fe4c0
       785.773017637 [8084-8084] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
       785.774086760 [8084-8084] ../src/export-buf.c:  82          releaseExporter Done releasing frames
       785.798252082 [8084-8084] ../src/vabackend.c:1766              nvTerminate Now have 0 (0 max) instances
[Child 8321, MediaDecoderStateMachine #1] WARNING: Decoder=7ff510b30900 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:369
[Child 8321, MediaDecoderStateMachine #1] WARNING: Decoder=7ff510b30900 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006): file /builds/worker/checkouts/gecko/dom/media/MediaDecoderStateMachine.cpp:3577`

im getting this erro in the console once i try to play an mp4 video - https://www.reddit.com/r/AquaticSnails/comments/uugf6j/nova_my_big_girl_spinning_a_moss_ball/ -  video doesnt play and firefox suggests me to install a decoder.

freakedenough, 2022-05-21:
firefox 100.0.2
nvidia-510 (kubuntu)
nvidia-vaapi-driver
nv-codec-headers
vdpau-va-driver-vp9

environemnt env set:
`LIBVA_DRIVER_NAME=nvidia
VDPAU_DRIVER=nvidia
MOZ_DISABLE_RDD_SANDBOX=1`

flags based on FAQ page.

elFarto, 2022-05-21:
Hmm, I think that's more a Firefox issue than a driver issue. The video is playing back fine for me. Do you have any Firefox add-ons that change what video formats it uses?

freakedenough, 2022-05-22:
i could solve it, it was an issue with the invoked ffmpeg and a libary which was broken. i figured that out by simply asking for ffmpeg version which returned this error:

besitzer@MSI:~$ ffmpeg -version
ffmpeg: error while loading shared libraries: /lib/x86_64-linux-gnu/libOpenCL.so.1: file too short

i searched for which package provides this file:
besitzer@MSI:~$ dpkg -S libOpenCL.so.1
ocl-icd-libopencl1:i386: /usr/lib/i386-linux-gnu/libOpenCL.so.1
ocl-icd-libopencl1:amd64: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.0.0
ocl-icd-libopencl1:i386: /usr/lib/i386-linux-gnu/libOpenCL.so.1.0.0
ocl-icd-libopencl1:amd64: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1

reinstalled via 
apt reinstall ocl-icd-libopencl1 
and now its working again.

closed 2022-05-22 by freakedenough