elFarto/nvidia-vaapi-driver #86: eglCreateContext: EGL_BAD_MATCH error: In eglCreateContext: Failure in argument parsing

reliant8307, 2022-05-27:
Nvidia-vaapi-driver works perfectly for watching VP9 videos on Youtube, but when I try to watch AVC1 (H264) videos, I get the following error after skipping the video. The following is a AVC1 video:

```
      3312.440103539 [11566-11572] ../src/vabackend.c:1022             nvEndPicture Decoded frame successfully to idx: 0 (0x7f4f5b9587a0)
      3312.440126922 [11566-11572] ../src/vabackend.c:1703    nvExportSurfaceHandle Exporting surface: 1 (0x7f4f5b958ac0)
      3312.440130118 [11566-11572] ../src/vabackend.c:1744    nvExportSurfaceHandle Exporting with 640 368 491520 640 327680 640
      3312.440131726 [11566-11572] ../src/vabackend.c:1052            nvSyncSurface Syncing on surface: 1 (0x7f4f5b958ac0)
      3312.440133017 [11566-11572] ../src/vabackend.c:1067            nvSyncSurface Surface 1 resolved (0x7f4f5b958ac0)
      3312.440159733 [11566-11584] ../src/vabackend.c: 357          resolveSurfaces Reading from queue: 12 13
      3312.440164528 [11566-11584] ../src/vabackend.c: 372          resolveSurfaces Mapping surface 0
      3312.440530122 [11566-11584] ../src/vabackend.c: 374          resolveSurfaces Mapped surface 0 to 205828000 (1024)
      3312.440553437 [11566-11584] ../src/vabackend.c: 378          resolveSurfaces Surface 0 exported
      3312.465684356 [11566-11573] ../src/vabackend.c:1022             nvEndPicture Decoded frame successfully to idx: 6 (0x7f4f5b958a20)
      3312.465702330 [11566-11584] ../src/vabackend.c: 357          resolveSurfaces Reading from queue: 13 14
      3312.465709228 [11566-11584] ../src/vabackend.c: 372          resolveSurfaces Mapping surface 6
[Child 11148, MediaDecoderStateMachine #1] WARNING: Decoder=7f96ce992400 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005): file /usr/local/src/hg/firefox/dom/media/MediaDecoderStateMachine.cpp:3577

      3312.488865245 [10861-10915] ../src/export-buf.c:  60                    debug [EGL] eglCreateContext: EGL_BAD_MATCH error: In eglCreateContext: Failure in argument parsing

      3312.488892946 [10861-10915] ../src/export-buf.c:  60                    debug [EGL] eglCreateContext: EGL_BAD_ATTRIBUTE error: In eglCreateContext: Failure in argument parsing

[Child 11148, MediaDecoderStateMachine #1] WARNING: Decoder=7f96ce992400 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005): file /usr/local/src/hg/firefox/dom/media/MediaDecoderStateMachine.cpp:3577
```

Any idea? Thanks!

Ps: here's my vainfo output:

```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/xorg/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.15.0.1)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD

```

elFarto, 2022-05-27:
Which version of Firefox are you using? Also, are you on X11 or Wayland?

reliant8307, 2022-05-27:
> Which version of Firefox are you using? Also, are you on X11 or Wayland?

I'm using Firefox 100.0.2 on X11.

reliant8307, 2022-05-27:
> It might be resolved by [this](https://archive.mozilla.org/pub/firefox/nightly/2022/05/2022-05-14-21-39-37-mozilla-central/) build of Firefox.

You're right, the above build fixed it. So I'm closing this issue. Thank you!

closed 2022-05-27 by reliant8307

elFarto, 2022-05-27:
Did you check that the new build was actually using the driver? Nightly builds use a different profile which might not enable it by default.

reliant8307, 2022-05-27:
Yes, I changed the **media.ffmpeg.vaapi.enabled** to "**true**", restarted Firefox and confirmed in the log it was using the driver. The MOZ_DISABLE_RDD_SANDBOX=1 environment variable was already set.

I also checked "nvidia-smi" for the "C" in Type column, so everything worked perfectly ;-) 100% sure the problem has gone.