elFarto/nvidia-vaapi-driver #89: WebGL Driver Render (FEATURE_FAILURE_WEBGL_EXHAUSTED_DRIVERS)

JustCauseWhyNot, 2022-06-03:
I'm not able to get firefox hwaccel working. I believe it's due to the error in the title. I've set in about:config all the options, and I've set environment variables in every place I can think of including in about:config. I've pasted about:support below.
 
Name | Firefox
-- | --
101.0
20220526203855
 
release
Mozilla/5.0 (X11; Linux x86_64; rv:101.0) Gecko/20100101 Firefox/101.0
Linux 5.18.1-gentoo-r2 #1 SMP PREEMPT_DYNAMIC Thu Jun 2 09:59:30 MDT 2022
Adwaita / Adwaita
1/1
1/1               Enabled by default
27
Active
Found
Found
Found
false
15.6 GB
439 GB

    Application Basics
------------------

Name: Firefox
Version: 101.0
Build ID: 20220526203855
Distribution ID:
Update Channel: release
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:101.0) Gecko/20100101 Firefox/101.0
OS: Linux 5.18.1-gentoo-r2 #1 SMP PREEMPT_DYNAMIC Thu Jun 2 09:59:30 MDT 2022
OS Theme: Adwaita / Adwaita
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: [27](https://github.com/elFarto/nvidia-vaapi-driver/issues/new#remote-processes)
Enterprise Policies: [Active](about:policies#active)
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false
Memory Size (RAM): 15.6 GB
Disk Space Available: 439 GB

Crash Reports for the Last 3 Days
---------------------------------

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org

Name: Picture-In-Picture
Version: 1.0.0
ID: pictureinpicture@mozilla.org

Name: Web Compatibility Interventions
Version: 101.7.0
ID: webcompat@mozilla.org

[... 414 lines cut ...]

Internationalization & Localization
-----------------------------------

Application Settings
Requested Locales: ["en-US"]
Available Locales: ["en-US"]
App Locales: ["en-US"]
Regional Preferences: ["en-US"]
Default Locale: "en-US"
Operating System
System Locales: ["en-US"]
Regional Preferences: ["en-US"]

Remote Debugging (Chromium Protocol)
------------------------------------

Accepting Connections: false
URL:

Printing
--------

Modified print settings
-----------------------

elFarto, 2022-06-04:
I think the cause is the inability to use EGL. Double check that `gfx.x11-egl.force-enabled` is set to `true` in about:config. You can also using something like `eglinfo` to make sure that EGL is functioning correctly.

Also, do you have the environment variable `MOZ_DISABLE_RDD_SANDBOX` set to `1`? The easiest way to ensure they're set is to run Firefox from the command line, like this:
```
MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvdec firefox
```

JustCauseWhyNot, 2022-06-04:
I rant `MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvdec firefox`, but that didn't work. I do get error with elginfo `eglinfo: eglInitialize failed`.

Here's the output of elginfo
❯ eglinfo
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base EGL_EXT_device_enumeration
    EGL_EXT_device_query EGL_KHR_client_get_all_proc_addresses
    EGL_EXT_client_extensions EGL_KHR_debug EGL_KHR_platform_x11
    EGL_EXT_platform_x11 EGL_EXT_platform_device EGL_MESA_platform_xcb
    EGL_MESA_platform_gbm EGL_KHR_platform_gbm
    EGL_MESA_platform_surfaceless

GBM platform:
eglinfo: eglInitialize failed

X11 platform:
EGL API version: 1.5
EGL vendor string: NVIDIA
EGL version string: 1.5
EGL client APIs: OpenGL_ES OpenGL
EGL extensions string:
    EGL_EXT_buffer_age EGL_EXT_client_sync
    EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_import
    EGL_EXT_image_dma_buf_import_modifiers EGL_MESA_image_dma_buf_export
    EGL_EXT_output_base EGL_EXT_stream_acquire_mode EGL_EXT_sync_reuse
    EGL_IMG_context_priority EGL_KHR_config_attribs
    EGL_KHR_create_context_no_error EGL_KHR_context_flush_control
    EGL_KHR_create_context EGL_KHR_fence_sync
    EGL_KHR_get_all_proc_addresses EGL_KHR_partial_update
    EGL_KHR_swap_buffers_with_damage EGL_KHR_no_config_context
    EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image
    EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image
    EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base
    EGL_KHR_image_pixmap EGL_KHR_reusable_sync EGL_KHR_stream
    EGL_KHR_stream_attrib EGL_KHR_stream_consumer_gltexture
    EGL_KHR_stream_cross_process_fd EGL_KHR_stream_fifo
    EGL_KHR_stream_producer_eglsurface EGL_KHR_surfaceless_context
    EGL_KHR_wait_sync EGL_NV_nvrm_fence_sync EGL_NV_post_sub_buffer
    EGL_NV_quadruple_buffer EGL_NV_stream_consumer_eglimage
    EGL_NV_stream_cross_display EGL_NV_stream_cross_object
    EGL_NV_stream_cross_process EGL_NV_stream_cross_system
    EGL_NV_stream_dma EGL_NV_stream_flush EGL_NV_stream_metadata
    EGL_NV_stream_remote EGL_NV_stream_reset EGL_NV_stream_socket
    EGL_NV_stream_socket_inet EGL_NV_stream_socket_unix
    EGL_NV_stream_sync EGL_NV_stream_fifo_next
    EGL_NV_stream_fifo_synchronous EGL_NV_stream_consumer_gltexture_yuv
    EGL_NV_stream_attrib EGL_NV_stream_origin EGL_NV_system_time
    EGL_NV_output_drm_flip_event EGL_NV_triple_buffer
    EGL_NV_robustness_video_memory_purge
Configurations:
     bf lv colorbuffer dp st  ms    vis   cav bi  renderable  supported
  id sz  l  r  g  b  a th cl ns b    id   eat nd gl es es2 vg surfaces 
---------------------------------------------------------------------
0x01 32  0  8  8  8  8 24  8  0 0 0x00--         y  y  y     pb,str
0x02 32  0 10 10 10  2 24  8  0 0 0x28--         y  y  y     win,pb,pix,str
0x03 32  0 10 10 10  2 24  8  0 0 0x7e--         y  y  y     win,pb,pix,str
0x04 32  0 10 10 10  2 24  8  0 0 0x53--         y  y  y     win,pb,pix,str
0x05 32  0  8  8  8  8 24  0  0 0 0x00--         y  y  y     pb,str
0x06 32  0 10 10 10  2 24  0  0 0 0x24--         y  y  y     win,pb,pix,str
0x07 32  0 10 10 10  2 24  0  0 0 0x7a--         y  y  y     win,pb,pix,str
0x08 32  0 10 10 10  2 24  0  0 0 0x4f--         y  y  y     win,pb,pix,str
0x09 32  0  8  8  8  8  0  0  0 0 0x00--         y  y  y     pb,str
0x0a 32  0 10 10 10  2  0  0  0 0 0x2c--         y  y  y     win,pb,pix,str
0x0b 32  0 10 10 10  2  0  0  0 0 0x82--         y  y  y     win,pb,pix,str
0x0c 32  0 10 10 10  2  0  0  0 0 0x57--         y  y  y     win,pb,pix,str
0x0d 32  0 10 10 10  2 24  8  2 1 0x38--         y  y  y     win,pb,str
0x0e 32  0 10 10 10  2 24  8  2 1 0x8e--         y  y  y     win,pb,str
0x0f 32  0 10 10 10  2 24  8  2 1 0x63--         y  y  y     win,pb,str
0x10 32  0 10 10 10  2 24  0  2 1 0x30--         y  y  y     win,pb,str
0x11 32  0 10 10 10  2 24  0  2 1 0x86--         y  y  y     win,pb,str
[... 294 lines cut ...]
0x1a 32  0  8  8  8  8  0  0  4 1 0x00--      a  y  y  y     pb
0x1b 32  0  8  8  8  8 16  0  4 1 0x00--      a  y  y  y     pb
0x1c 32  0  8  8  8  8 24  0  4 1 0x00--      a  y  y  y     pb
0x1d 32  0  8  8  8  8 24  8  4 1 0x00--      a  y  y  y     pb
0x1e 32  0  8  8  8  8 32  0  4 1 0x00--      a  y  y  y     pb
0x1f 24  0  8  8  8  0  0  0  0 0 0x00--      y  y  y  y     pb
0x20 24  0  8  8  8  0 16  0  0 0 0x00--      y  y  y  y     pb
0x21 24  0  8  8  8  0 24  0  0 0 0x00--      y  y  y  y     pb
0x22 24  0  8  8  8  0 24  8  0 0 0x00--      y  y  y  y     pb
0x23 24  0  8  8  8  0 32  0  0 0 0x00--      y  y  y  y     pb
0x24 24  0  8  8  8  0  0  0  4 1 0x00--      y  y  y  y     pb
0x25 24  0  8  8  8  0 16  0  4 1 0x00--      y  y  y  y     pb
0x26 24  0  8  8  8  0 24  0  4 1 0x00--      y  y  y  y     pb
0x27 24  0  8  8  8  0 24  8  4 1 0x00--      y  y  y  y     pb
0x28 24  0  8  8  8  0 32  0  4 1 0x00--      y  y  y  y     pb
0x29 16  0  5  6  5  0  0  0  0 0 0x00--      y  y  y  y     pb
0x2a 16  0  5  6  5  0 16  0  0 0 0x00--      y  y  y  y     pb
0x2b 16  0  5  6  5  0 24  0  0 0 0x00--      y  y  y  y     pb
0x2c 16  0  5  6  5  0 24  8  0 0 0x00--      y  y  y  y     pb
0x2d 16  0  5  6  5  0 32  0  0 0 0x00--      y  y  y  y     pb
0x2e 16  0  5  6  5  0  0  0  4 1 0x00--      y  y  y  y     pb
0x2f 16  0  5  6  5  0 16  0  4 1 0x00--      y  y  y  y     pb
0x30 16  0  5  6  5  0 24  0  4 1 0x00--      y  y  y  y     pb
0x31 16  0  5  6  5  0 24  8  4 1 0x00--      y  y  y  y     pb
0x32 16  0  5  6  5  0 32  0  4 1 0x00--      y  y  y  y     pb

elFarto, 2022-06-04:
```
GBM platform:
eglinfo: eglInitialize failed
```
This is concerning. It should work correctly for the last 470 and current 500+ driver series. The fact it isn't means there's something wrong with your system setup, however I'm not sure what it is. Debugging these issues is hard enough when you have access to the system, trying to do it over the Internet like is almost impossible.

Since this is a Firefox/Gentoo issue, not related to this driver, I'd recommend you ask on those forums as they might be able to give you more specific information.

JustCauseWhyNot, 2022-06-04:
I kinda thought that might be the issue around the time you sent the message. I'll close the issue since it's not an issue with this driver. You've been extremely helpful thank you.

JustCauseWhyNot, 2022-06-17:
Firefox about:support
```
Name | Firefox
-- | --
101.0.1
20220616180410
gentoo
Mozilla/5.0 (X11; Linux x86_64; rv:101.0) Gecko/20100101 Firefox/101.0
Linux 5.18.3-gentoo #1 SMP PREEMPT_DYNAMIC Mon Jun 13 10:38:51 MDT 2022
Adwaita / Adwaita
1/1
1/1               Enabled by default
16
Active
Found
Found
Found
false
15.6 GB
435 GB
    Application Basics
------------------

Name: Firefox
Version: 101.0.1
Build ID: 20220616180410
Distribution ID: gentoo
User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:101.0) Gecko/20100101 Firefox/101.0
OS: Linux 5.18.3-gentoo #1 SMP PREEMPT_DYNAMIC Mon Jun 13 10:38:51 MDT 2022
OS Theme: Adwaita / Adwaita
Multiprocess Windows: 1/1
Fission Windows: 1/1 Enabled by default
Remote Processes: [16](https://github.com/elFarto/nvidia-vaapi-driver/issues/89#remote-processes)
Enterprise Policies: [Active](about:policies#active)
Google Location Service Key: Found
Google Safebrowsing Key: Found
Mozilla Location Service Key: Found
Safe Mode: false
Memory Size (RAM): 15.6 GB
Disk Space Available: 435 GB

Firefox Features
----------------

Name: DoH Roll-Out
Version: 2.0.0
ID: doh-rollout@mozilla.org

Name: Firefox Screenshots
Version: 39.0.1
ID: screenshots@mozilla.org

Name: Form Autofill
Version: 1.0.1
ID: formautofill@mozilla.org

Name: Picture-In-Picture
Version: 1.0.0
ID: pictureinpicture@mozilla.org

Name: Web Compatibility Interventions
Version: 101.7.0
ID: webcompat@mozilla.org

Name: WebCompat Reporter
Version: 1.4.2
ID: webcompat-reporter@mozilla.org

Remote Features
---------------
[... 672 lines cut ...]

Printing
--------

Modified print settings
-----------------------
```

NVD_LOG=1 firefox 
```Missing chrome or resource URL: resource://gre/modules/UpdateListener.jsm
Sandbox: attempt to open unexpected file /usr/lib64/firefox/libgcc_s.so.1
Sandbox: attempt to open unexpected file /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/libgcc_s.so.1
Sandbox: attempt to open unexpected file /lib64/glibc-hwcaps/x86-64-v3/libgcc_s.so.1
Sandbox: seccomp sandbox violation: pid 14676, tid 14676, syscall 262, args 4294967196 140734005684928 140734005685104 0 4294967295 140734005684928.
Sandbox: attempt to open unexpected file /lib64/glibc-hwcaps/x86-64-v2/libgcc_s.so.1
Sandbox: seccomp sandbox violation: pid 14676, tid 14676, syscall 262, args 4294967196 140734005684928 140734005685104 0 4294967295 140734005684928.
Sandbox: attempt to open unexpected file /lib64/tls/x86_64/x86_64/libgcc_s.so.1
Sandbox: seccomp sandbox violation: pid 14676, tid 14676, syscall 262, args 4294967196 140734005684928 140734005685104 0 4294967295 140734005684928.
Sandbox: attempt to open unexpected file /lib64/tls/x86_64/libgcc_s.so.1
Sandbox: seccomp sandbox violation: pid 14676, tid 14676, syscall 262, args 4294967196 140734005684928 140734005685104 0 4294967295 140734005684928.
Sandbox: attempt to open unexpected file /lib64/tls/x86_64/libgcc_s.so.1
Sandbox: seccomp sandbox violation: pid 14676, tid 14676, syscall 262, args 4294967196 140734005684928 140734005685104 0 4294967295 140734005684928.
```

I've still not been able to get hwaccel working. But I think it might be due to x11_egl platform not working.

elFarto, 2022-06-19:
This driver is only responsible for hardware video acceleration (VA-API), not WebGL. The issue you're having is caused by Firefox being unable to initialise EGL for some reason. This is not something I can help you fix, and you're better off contacting the Firefox devs about this.

The only issue I can spot in the about related to this driver, is that you appear to be missing the environment variable `MOZ_DISABLE_RDD_SANDBOX`, which should be set to `1`.

closed 2022-06-19 by JustCauseWhyNot

Munzu, 2022-08-06:
@JustCauseWhyNot Have you been able to fix the issue? I think I'm having the same problem.

JustCauseWhyNot, 2022-08-07:
Yes I have. It was an issue with my system. I can elaborate more if you want me to.

Munzu, 2022-08-07:
Yes please

JustCauseWhyNot, 2022-08-07:
Two things to check. One is that your user is in video group just run `groups` in terminal to get output of current users groups. And also make sure /dev/nvidia*, and /dev/dri/(card0,renderD128) all are in video groups. Second is make sure you're xorg.conf is set correctly. Here's my config at /etc/X11/xorg.conf.d/20-nvidia.conf
```
Section "Device"
  Identifier "Nvidia Card"
  Driver "nvidia"
  VendorName "NVIDIA Corporation"
  Option "AddARGBGLXVisuals" "true"
  Option "AllowGLXWithComposite" "true"
  Option "AllowIndirectGLXProtocol" "false"
  Option "AllowUnofficialGLXProtocol" "false"
  Option "ColorRange" "Full"
  Option "ColorSpace" "RGB"
  Option "Coolbits" "16"
  Option "FlatPanelProperties" "Dithering = Enabled, DitheringMode = Dynamic-2x2"
  Option "HardDPMS" "false"
  Option "MultiGPU" "0"
  Option "RenderAccel" "true"
  Option "TripleBuffer" "false"
  Option "UseDisplayDevice" "DFP"
EndSection
   
Section "Extensions"
   Option "Composite" "Enable"                                                                                                               EndSection                                                                                                                               ```

JustCauseWhyNot, 2022-08-07:
If you're on gentoo let me know if you want my use flags settings.

Munzu, 2022-08-08:
Thanks, I managed to solve my problem. I sometimes switch between Wayland and X11 and I had to make sure these two environment variables were unset when using X11:
```
EGL_PLATFORM=wayland
MOZ_ENABLE_WAYLAND=1
```
This solved this problem for me: https://bugzilla.mozilla.org/show_bug.cgi?id=1765042

Then, I had to set `webgl.force-enable` to `true` in `about:config`. This was suggested here: https://support.mozilla.org/en-US/questions/1334668

Now HW decoding is working for me. I'm on Arch, currently mainly X11.

JustCauseWhyNot, 2022-08-09:
Glad it's working for you.