elFarto/nvidia-vaapi-driver #90: Some files not playing correctly with mpv

philipl, 2022-06-06:
I'm pretty sure this is a regression, but I've observed certain files failing to play back with mpv (latest master).

The examples I've found have been h.264 and vc1 - but certainly not all h.264 files. Anything downloaded from youtube by yt-dlp seems to be fine for example, but one example is [this](https://download.blender.org/durian/movies/Sintel.2010.1080p.mkv) 1080p version of Sintel.

I did some quick digging with gdb and it seems the 'bad' files are causing the hwaccel to be unloaded in mpv and then reloaded. This has something to do with format probing I think, but it leads to a call to `nvDestroyContext` and then it tries to create a new context but the expected number of surfaces is zero, instead of matching the number of render targets.

So it seems something is missing in the scenario where the context is destroyed and recreated within the same driver instance.

I need to re-generate logs and I'll add them shortly.

philipl, 2022-06-06:
So, I did the one line change to make it use num_render_targets if drv->surfaceCount is zero. This makes the error go away, but the frames show massive amounts of corruption.

An example:

[image]

philipl, 2022-06-06:
Here's a log.

[nvd.log](https://github.com/elFarto/nvidia-vaapi-driver/files/8848888/nvd.log)

philipl, 2022-06-06:
I see it repeatedly saying

```
realiseSurface No free surfaces found
```

So maybe the surfaces from the first context, which are being reused, are not actually usable. I could believe the errors here are what you'd see if you lost all your reference frames.

elFarto, 2022-06-07:
> I see it repeatedly saying
> 
> ```
> realiseSurface No free surfaces found
> ```

This is fine, the driver is just looking for cached cuArrays that have been freed to reuse them. It stops logging that after the first 4 or 5 frames once all of them have been allocated.

However, there shouldn't be any corruption. I'll take a look at the issue.

elFarto, 2022-06-07:
I can't seem to reproduce it with the mpv master version with that Sintel video. What version of ffmpeg are you using?

philipl, 2022-06-07:
I've tried your branch fix and it works. No corruption. Thanks!

PR #91 (2022-06-07): Check if surface has been used in a different context before use
Newer versions of ffmpeg appear to reuse VASurfaces across VAContexts. This means we could be presented a surface with a pictureId that is incorrect.
This change add a check to reset the pictureId and detach and backing image from the surface before we attempt to use it.

Fixes #90
commit 43eae91c (2022-06-07): Check if surface has been used before.

Newer versions of ffmpeg appear to reuse VASurfaces across VAContexts. This means we could be presented a surface with a pictureId that is incorrect.
This change add a check to reset the pictureId and detach and backing image from the surface before we attempt to use it.

Fixes #90
closed 2022-06-07 by elFarto