elFarto/nvidia-vaapi-driver #95: The driver doesn't fail against a uncompatible DRI device

tytan652, 2022-07-02:
I have a desktop machine with two GPU, Intel iGPU and NVIDIA.

If `LIBVA_DRIVER_NAME` is set to `nvidia`, vainfo will not fail against the Intel device and entrypoints of the Nvidia card.

Normal `vainfo` against my Nvidia card:
```
vainfo --display drm --device /dev/dri/renderD128
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
```
Normal `vainfo` against my Intel iGPU:
```
vainfo --display drm --device /dev/dri/renderD129
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
```
`vainfo` against my Intel iGPU with `LIBVA_DRIVER_NAME` set to `nvidia`:
```
LIBVA_DRIVER_NAME="nvidia" vainfo --display drm --device /dev/dri/renderD129
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
```
Rather than failling like any driver, this one don't and show info of the Nvidia card.

elFarto, 2022-07-02:
Yes, that's a known issue. It shouldn't be too difficult to detect if we're looking at an NVIDIA device or not.

tytan652, 2022-07-02:
> Yes, that's a known issue.

Oh, I didn't see an issue that related to this. Sorry if it's a duplicate.

elFarto, 2022-07-02:
It hasn't been raised before, but I've been aware of it, it's just not been high up on my list of things to fix as most users only have a single NVIDIA GPU.

elFarto, 2022-07-09:
I've added an option to fail initialisation if the DRM device passed to us isn't an nvidia-drm one. Can you retest this with the latest master version and check it works as expected?

tytan652, 2022-07-09:
Work as expected, thank you for implementing this.

Solved with 186e3a0594623131d9adcf3a82d6ec6831b1b5ff.

closed 2022-07-09 by tytan652