elFarto/nvidia-vaapi-driver #97: undefined reference to 'pthread_mutexattr_settype'

marioroy, 2022-07-08:
Linking is failing with master branch. The -pthread compiler flag is needed.

elFarto, 2022-07-09:
Hmm, wonder why that's just broken now, a new version of the compiler?. Anyway, I've added the option to meson.

closed 2022-07-09 by elFarto

marioroy, 2022-07-09:
It seems something changed with my Ubuntu 20.04 environment. I tried building v0.0.6 and got the same error. Thank you, for adding `threads` run-time dependency.
```text
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.16.2
Run-time dependency ffnvcodec found: YES 11.1.5.1
Run-time dependency libva found: YES 1.14.0
Run-time dependency threads found: YES
...
```