elFarto/nvidia-vaapi-driver #98: Can't seem to get it working

endorphinsu, 2022-07-09:
My firefox version is 102.0.1-1
I installed this through the AUR package (nvidia-vaapi-driver-git)

What I have done:

Changed about:config variables

My /etc/environment file includes:
LIBVA_DRIVER_NAME=nvidia
MOZ_X11_EGL=1
MOZ_DISABLE_RDD_SANDBOX=1

And i have nvidia-drm.modeset=1 in my boot parameters

Have i missed something?

Firefox log when running with "MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox"
[firefox.log](https://github.com/elFarto/nvidia-vaapi-driver/files/9076124/firefox.log)

elFarto, 2022-07-09:
Can you run that command again, but with NVD_LOG=1 set aswell?

endorphinsu, 2022-07-09:
Here you go
[firefox.log](https://github.com/elFarto/nvidia-vaapi-driver/files/9076243/firefox.log)

elFarto, 2022-07-09:
I'm not sure what's going on here, but it seems to be a Firefox issue. It never attempts to initialise the driver after the first time, which is just to see what codecs it supports.

The issue seems to be around here in the log:
```
[RDD 21415: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 21415: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 21415: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 21415: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
...
[Child 21288: MediaPDecoder #2]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Child 21288: MediaPDecoder #2]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Child 21288: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: FFmpegVideoDecoder::FFmpegVideoDecoder MIME video/avc Codec ID 27
[Child 21288: MediaPDecoder #2]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 21288: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: VA-API is disabled by preference.
[Child 21288: MediaPDecoder #3]: D/PlatformDecoderModule FFMPEG: Initialising FFmpeg decoder
[Child 21288: MediaPDecoder #3]: D/PlatformDecoderModule FFMPEG:   codec h264 : H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
```
The `XRE_IsRDDProcess 0` should really be `XRE_IsRDDProcess 1`. I would recommend you raise this as an issue with Mozilla as they'll know better why that's occurring.

One other thing you could try is playing a VP9 video, rather than a H.264 one. That uses a slightly different codepath that might work better.

pc00per, 2022-07-09:
I've same issue on stable firefox. Try beta.

endorphinsu, 2022-07-09:
I installed firefox-beta-bin for the AUR and did the same about:config changes as before and it seems to be working now!

closed 2022-07-09 by endorphinsu

marioroy, 2022-07-09:
To have Firefox choose the VP9 codepath for AV1 content, set `media.av1.enabled` to `false` in about:config.