elFarto/nvidia-vaapi-driver #99: Error upon resume from sleep: "glxtest: VA-API test failed: process crashed"

Roman2K, 2022-07-10:
The driver worked very well with Firefox Developer Edition on a desktop PC with a GTX 1080Ti, even on [8K60](https://www.youtube.com/watch?v=vJz8QzO1VzQ) videos (thanks for your work!).
 
Today upon resuming the PC from sleep/suspend, Firefox was crashed. I noticed choppy 4K videos so I checked if NVD was still working but got the following errors.

Firefox log:

```
[GFX1-]: glxtest: VA-API test failed: process crashed. Please check your VA-API drivers.
Missing chrome or resource URL: resource://gre/modules/UpdateListener.jsm
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
[2022-07-10T09:35:39Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-07-10T09:35:39Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-07-10T09:35:39Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-07-10T09:35:39Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[Child 16222, MediaDecoderStateMachine #1] WARNING: Decoder=7f0aabbbcd00 Decode error: NS_ERROR_DOM_MEDIA_REMOTE_DECODER_CRASHED_RDD_OR_GPU_ERR (0x806e0010) - HandleRejectionError: file /build/firefox-developer-edition/src/firefox-103.0/dom/media/MediaDecoderStateMachineBase.cpp:151
```

NVD log:

```
       918.264714391 [4560-4560] ../src/vabackend.c:1778       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7ff1efe22d40 31
       918.264808772 [4560-4560] ../src/vabackend.c:1781       __vaDriverInit_1_0 Now have 0 (0 max) instances
       918.270594136 [4560-4560] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 0
       918.272119106 [4560-4560] ../src/export-buf.c: 172       findGPUIndexFromFd Found 3 EGL devices
       918.272215346 [4560-4560] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       918.272228066 [4560-4560] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 0
       918.272233616 [4560-4560] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
       918.332888669 [4560-4560] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
       918.332914129 [4560-4560] ../src/export-buf.c:  99                reconnect Reconnecting to stream
       918.354337147 [4560-4560] ../src/vabackend.c: 556           nvCreateConfig got profile: 6 with 0 attributes
       918.354358638 [4560-4560] ../src/vabackend.c: 556           nvCreateConfig got profile: 7 with 0 attributes
       918.354364288 [4560-4560] ../src/vabackend.c: 556           nvCreateConfig got profile: 13 with 0 attributes
       918.354369128 [4560-4560] ../src/vabackend.c: 556           nvCreateConfig got profile: 19 with 0 attributes
       918.354374418 [4560-4560] ../src/vabackend.c:1755              nvTerminate Terminating 0x7ff1efe22d40
       918.354379008 [4560-4560] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
       918.355975097 [4560-4560] ../src/export-buf.c:  82          releaseExporter Done releasing frames
       918.382002653 [4560-4560] ../src/vabackend.c:1769              nvTerminate Now have 0 (0 max) instances
       922.603034919 [4560-4621] ../src/export-buf.c:  60                    debug [EGL] eglCreateContext: EGL_BAD_MATCH error: In eglCreateContext: Failure in argument parsing

       922.603100909 [4560-4621] ../src/export-buf.c:  60                    debug [EGL] eglCreateContext: EGL_BAD_ATTRIBUTE error: In eglCreateContext: Failure in argument parsing

      1055.603714164 [4560-4621] ../src/export-buf.c:  60                    debug [EGL] eglCreateContext: EGL_BAD_MATCH error: In eglCreateContext: Failure in argument parsing

      1055.603782225 [4560-4621] ../src/export-buf.c:  60                    debug [EGL] eglCreateContext: EGL_BAD_ATTRIBUTE error: In eglCreateContext: Failure in argument parsing
```

And `vainfo`:

```
$ NVD_LOG=1 vainfo
      5515.277509953 [20222-20222] ../src/vabackend.c: 100                     init cuda error 'unknown error' (999)
```

I tried logging out and back in, the problem persisted.
What did work was **rebooting** completely.

So apparently something happens between sleep and resume that prevents the driver from working. Do you have any ideas? How can I help to fix it?

### Logs

Gist with all logs (firefox_about_support, firefox.log, nvd.log, nvidia_vaapi_coredump.log, vainfo.log): https://gist.github.com/Roman2K/c16a44d37bfbcebf652fec2f3439648e

### Versions used:

Firefox Developer Edition: 103.0b5
nvidia-vaapi-driver: 0.0.6.r13.ga4fb2d6-1
Kernel: 5.18.9-arch1-1
Relevant cmdline: `nvidia-drm.modeset=1 loglevel=3 audit=0 lsm=landlock,lockdown,yama,integrity,apparmor,bpf`
Relevant env:
```
export LIBVA_DRIVER_NAME=nvidia
export MOZ_DISABLE_RDD_SANDBOX=1
```

KnightMurloc, 2022-07-10:
> init cuda error 'unknown error' (999)

looks like a known issue with CUDA not working after sleep. try unloading and loading the nvidia_uvm module.
sudo rmmod nvidia_uvm
sudo modprobe nvidia_uvm

Roman2K, 2022-07-10:
Ah yes! Looks like a dup of https://github.com/elFarto/nvidia-vaapi-driver/issues/42 - will try the suggestions from over there, closing this for now.

Thank you!

closed 2022-07-10 by Roman2K