elFarto/nvidia-vaapi-driver #101: chrome-support branch displays wrong colors for vp9

marioroy, 2022-07-24:
I tried the chrome-support branch.
```text
OS: Ubuntu 22.04 LTS
Browser: Brave Version 1.41.100 Chromium: 103.0.5060.134 (Official Build) (64-bit)
Graphics Processor: NVIDIA GeForce RTX 3070
NVIDIA Driver Version: 515.48.07
XDG_SESSION_TYPE: x11
nvidia-vaapi-driver branch: chrome-support
```
YouTube video: [Big Buck Bunny 60fps 4K](https://www.youtube.com/watch?v=aqz-KE-bpKQ)

I use the enhanced-h264ify extension to toggle AV1 off / on.

VP9: wrong colors, hardware accelerated

[image]

AV1: correct colors, not hardware accelerated

[image]

Brave arguments:
```bash
brave-browser-stable \
    --disable-features=UseChromeOSDirectVideoDecoder \
    --enable-features=VaapiVideoDecoder \
    --use-gl=desktop $* &> /dev/null &
```

elFarto, 2022-07-24:
That branch is using quite the hack to get it to work. I'm having to do the YUV -> RGB conversion myself, and I doubt I got the right magic numbers for the colour space. I'm having to copy the image to the CPU, do the conversion then copy it to the destination window, which is hideously inefficient. I'm currently investigating if I can do this in one go, but trying to convince the NVIDIA X driver to give up the address of the output is not easy.

This branch isn't intended for general use, I was hoping to work on the VPP functions that the Chromium EGL path uses, but that's not used at all for the X11 path.

elFarto, 2022-08-06:
Ok, all the colours should be sorted now. However for the moment you'll need to run it with `NVD_BACKEND=direct` as I've completely reworked the copying code to work entirely on the GPU, rather than copying it back to RAM first, and I'm yet to reinstate the old copy path (or figure out a way to do a GPU->GPU copy without talking directly to the driver).

marioroy, 2022-08-06:
I will try. Unfortunately, building `chrome-support` emits errors.
```text
$ git clone https://github.com/elFarto/nvidia-vaapi-driver
$ cd nvidia-vaapi-driver
$ git checkout chrome-support
```
```text
$ export CFLAGS="-O2 -march=native"
$ export PATH="$PATH:/usr/local/cuda/bin"
$ export PKG_CONFIG_PATH="/usr/local/lib/pkgconfig"
$ meson setup build
The Meson build system
Version: 0.63.0
Source dir: /home/mario/nvidia-vaapi-driver
Build dir: /home/mario/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: gcc (gcc 12.1.1 "gcc (Clear Linux OS for Intel Architecture) 12.1.1 20220803 releases/gcc-12.1.0-322-g3df2f03587")
C linker for the host machine: gcc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Found pkg-config: /usr/bin/pkg-config (0.29.2)
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.20.3
Run-time dependency ffnvcodec found: YES 11.1.5.1
Run-time dependency libva found: YES 1.15.0
Run-time dependency threads found: YES
Library X11 found: YES
Run-time dependency xcb found: YES 1.15
Program nvcc found: YES (/usr/local/cuda/bin/nvcc)
Program ld found: YES (/usr/bin/ld)
Build targets in project: 3

Found ninja-1.11.0 at /usr/bin/ninja
```
```text
$ meson install -C build
ninja: Entering directory `/home/mario/nvidia-vaapi-driver/build'
[2/19] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
FAILED: nvidia_drv_video.so.p/src_h264.c.o 
gcc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I/usr/local/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -O2 -march=native -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_h264.c.o -MF nvidia_drv_video.so.p/src_h264.c.o.d -o nvidia_drv_video.so.p/src_h264.c.o -c ../src/h264.c
In file included from /usr/include/EGL/eglplatform.h:134,
                 from /usr/include/EGL/egl.h:20,
                 from ../src/vabackend.h:6,
                 from ../src/h264.c:1:
../nvidia-include/nv-ioctl.h:74:14: error: two or more data types in declaration specifiers
   74 |     NvU32    Status;
      |              ^~~~~~
In file included from ../nvidia-include/nvidia.h:8,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:13:
../nvidia-include/nv-ioctl.h:83:14: error: two or more data types in declaration specifiers
   83 |     NvU32    Status;
      |              ^~~~~~
```

marioroy, 2022-08-06:
The above was on Clear Linux [36750](https://cdn.download.clearlinux.org/releases/36750/clear/) using this [guide](https://github.com/marioroy/nvidia-driver-on-clear-linux) to install the OS, NVIDIA proprietary driver, and the CUDA Toolkit.

I will try building on Ubuntu 20.04 next.

elFarto, 2022-08-06:
I've pushed an attempt at a fix for that issue. There's some system header that's defining Status as a macro which breaks the NVIDIA headers.

marioroy, 2022-08-06:
The fix works.
```text
[19/19] Linking target nvidia_drv_video.so
Installing nvidia_drv_video.so to /usr/lib64/dri
```

DatCrazyDelphox, 2022-08-06:
The build also fails on Arch Linux when using either the build instructions in README or the build instructions inside the PKGBUILD from AUR

`meson setup build` produces this:
```
+ exec meson setup --prefix /usr --libexecdir lib --sbindir bin --buildtype plain --auto-features enabled --wrap-mode nodownload -D b_lto=true -D b_pie=true . build
The Meson build system
Version: 0.63.0
Source dir: /mnt/ram/nvidia-vaapi-driver
Build dir: /mnt/ram/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 12.1.1 "cc (GCC) 12.1.1 20220730")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Found pkg-config: /usr/bin/pkg-config (1.8.0)
Run-time dependency gstreamer-codecparsers-1.0 found: YES 1.20.3
Run-time dependency ffnvcodec found: YES 11.1.5.1
Run-time dependency libva found: YES 1.15.0
Run-time dependency threads found: YES
Library X11 found: YES
Run-time dependency xcb found: YES 1.15
Program nvcc found: YES (/opt/cuda/bin/nvcc)
Program ld found: YES (/usr/bin/ld)
Build targets in project: 3

nvidia-vaapi-driver 0.1

  User defined options
    auto_features: enabled
    buildtype    : plain
    libexecdir   : lib
    prefix       : /usr
    sbindir      : bin
    wrap_mode    : nodownload
    b_lto        : true
    b_pie        : true

Found ninja-1.11.0 at /usr/bin/ninja
```

And it fails with this message:
```
ninja: Entering directory `/mnt/ram/nvidia-vaapi-driver/build'
[1/19] Compiling C object nvidia_drv_video.so.p/src_export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_export-buf.c.o -MF nvidia_drv_video.so.p/src_export-buf.c.o.d -o nvidia_drv_video.so.p/src_export-buf.c.o -c ../src/export-buf.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/export-buf.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: Arquivo ou diretório inexistente
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[2/19] Compiling C object nvidia_drv_video.so.p/src_hevc.c.o
FAILED: nvidia_drv_video.so.p/src_hevc.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_hevc.c.o -MF nvidia_drv_video.so.p/src_hevc.c.o.d -o nvidia_drv_video.so.p/src_hevc.c.o -c ../src/hevc.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/hevc.c:3:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: Arquivo ou diretório inexistente
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[... 98 lines cut ...]
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/vp9.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: Arquivo ou diretório inexistente
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[13/19] Compiling C object nvidia_drv_video.so.p/src_vp8.c.o
FAILED: nvidia_drv_video.so.p/src_vp8.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_vp8.c.o -MF nvidia_drv_video.so.p/src_vp8.c.o.d -o nvidia_drv_video.so.p/src_vp8.c.o -c ../src/vp8.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/vp8.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: Arquivo ou diretório inexistente
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[14/19] Generating nvcc-ptx with a custom command
ninja: build stopped: subcommand failed.
Could not rebuild /mnt/ram/nvidia-vaapi-driver/build
```

Technically it can be fixed by renaming all occurrences of "drm/drm.h" to "libdrm/drm.h"

marioroy, 2022-08-06:
@DatCrazyDelphox, did you try the fix `git pull`. @elFarto committed a fix.

DatCrazyDelphox, 2022-08-06:
I just happened to pull right after the fix was submited
Same error nonetheless

elFarto, 2022-08-06:
It looks like the NVIDIA headers are requiring the kernel headers to be install to build against it. I'm hesitant to modify those headers as they're taken directly from NVIDIA's open source driver, and would make keeping up to date with them much more difficult if I have to continually patch them up.

I'll investigate if there's a way of supporting this cleanly, or if we'll have to just require the kernel headers be available.

marioroy, 2022-08-06:
@elFarto The colors look okay. However zero utilization on the video engine. The VDPAU VA driver works fine and consumes the video engine.

I'm using the following script to launch Brave. Also, installed the `Not, AV1` extension.
```bash
#! /bin/bash
## Launch script for Brave stable.

EXECCMD=/opt/3rd-party/bundles/clearfraction/opt/brave.com/brave/brave-browser

if [[ ! -x "$EXECCMD" ]]; then
    echo "ERROR: cannot access '$EXECCMD': No such executable"
    exit 1
fi

# Launch browser.
export FONTCONFIG_PATH=/usr/share/defaults/fonts

# Using NVIDIA graphics?
if [[ -d /opt/nvidia && -f /usr/lib64/dri/vdpau_drv_video.so ]]; then
    # export LIBVA_DRIVER_NAME=vdpau

    # nvidia-vaapi-driver chrome-support branch
    # https://github.com/elFarto/nvidia-vaapi-driver/issues/101
    export LIBVA_DRIVER_NAME=nvidia
    export NVD_BACKEND=direct
fi

# Otherwise, uncomment one of the following matching your configuration.
# export LIBVA_DRIVER_NAME=iHD
# export LIBVA_DRIVER_NAME=i965
# export LIBVA_DRIVER_NAME=r600
# export LIBVA_DRIVER_NAME=radeonsi

[[ $XDG_SESSION_TYPE == wayland ]] && GL=egl || GL=desktop

SCALE_FACTOR=$(
    xrdb -query Xft.dpi | awk '/^Xft.dpi:/ {
        dpi = $2; if (dpi % 2) dpi += 1
        printf("%.9f", dpi / 96)
    }'
)

# Do not enable #ignore-gpu-blocklist for NVIDIA graphics.
# It may cause extra CPU utilization in Wayland, during video playback.

exec "$EXECCMD" --window-size=1214,1000 \
    --high-dpi-support=1 --force-device-scale-factor=$SCALE_FACTOR \
    --disable-direct-composition \
    --disable-features=UseChromeOSDirectVideoDecoder \
    --disable-font-subpixel-positioning --disable-gpu-vsync \
    --disable-gpu-driver-bug-workarounds --enable-zero-copy \
    --enable-accelerated-2d-canvas --enable-smooth-scrolling \
    --enable-features=VaapiVideoDecoder,CanvasOopRasterization \
    --enable-gpu-rasterization --use-gl=$GL $* &> /dev/null &
```

marioroy, 2022-08-06:
```bash
$ vainfo
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
```

marioroy, 2022-08-06:
This time with `NVD_LOG=1` set.
```bash
$ NVD_LOG=1 vainfo
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      5464.675819666 [10343-10343] ../src/vabackend.c:1919       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5597f8e0a050 10
      5464.675824986 [10343-10343] ../src/vabackend.c:1923       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
      5464.675826866 [10343-10343] ../src/vabackend.c:1940       __vaDriverInit_1_0 Now have 0 (0 max) instances
      5464.675828476 [10343-10343] ../src/vabackend.c:1962       __vaDriverInit_1_0 Selecting EGL backend
      5464.677992637 [10343-10343] ../src/export-buf.c: 160       findGPUIndexFromFd Looking for GPU index: 0
      5464.678645163 [10343-10343] ../src/export-buf.c: 172       findGPUIndexFromFd Found 1 EGL devices
      5464.678693604 [10343-10343] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      5464.678696564 [10343-10343] ../src/export-buf.c: 133 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
      5464.678698164 [10343-10343] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
      5464.678699444 [10343-10343] ../src/export-buf.c: 215       findGPUIndexFromFd No match found, falling back to default device
      5464.679456642 [10343-10343] ../src/export-buf.c: 269         egl_initExporter Driver doesn't support 16-bit surfaces
      5464.747996954 [10343-10343] ../src/vabackend.c:1998       __vaDriverInit_1_0 cuda error 'the provided PTX was compiled with an unsupported toolchain.' (222)
```

marioroy, 2022-08-06:
I have `nvidia_drm modeset` enabled. In fact, Firefox works with `nvdec` and consumes the video engine.
```bash
$ cat /etc/modprobe.d/enable-nvidia-modeset.conf 
options nvidia-drm modeset=1
```
Below, `nvdec` is from the master branch whereas `nvidia` is from the chrome-support branch.
```bash
$ ls -ltr /usr/lib64/dri
...
-rwxr-xr-x  1 root root   516800 Aug  5 13:04 nvdec_drv_video.so
-rwxr-xr-x  1 root root   611296 Aug  5 13:05 vdpau_drv_video.so
-rwxr-xr-x  1 root root   691336 Aug  6 10:06 nvidia_drv_video.so
```

elFarto, 2022-08-06:
Can you run chrome with `NVD_LOG=1`?

marioroy, 2022-08-06:
Okay. Working... :) I committed a [fix](https://github.com/marioroy/nvidia-driver-on-clear-linux/commit/f964918ba8ab4a95e90fff77036ea329cb75b00f) to the `install-cuda` installer script.

Running `sudo install-cuda auto` will fetch the CUDA toolkit matching the display driver. I had tried display driver 510 and needed to remove the existing CUDA Toolkit installation before installation.

```bash
$ LIBVA_DRIVER_NAME=nvidia vainfo
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
```

marioroy, 2022-08-06:
```bash
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      8239.587300039 [11848-11848] ../src/vabackend.c:1919       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55e3c8b02050 10
      8239.587304929 [11848-11848] ../src/vabackend.c:1923       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
      8239.587306899 [11848-11848] ../src/vabackend.c:1940       __vaDriverInit_1_0 Now have 0 (0 max) instances
      8239.587308579 [11848-11848] ../src/vabackend.c:1962       __vaDriverInit_1_0 Selecting EGL backend
      8239.589510006 [11848-11848] ../src/export-buf.c: 160       findGPUIndexFromFd Looking for GPU index: 0
      8239.590172426 [11848-11848] ../src/export-buf.c: 172       findGPUIndexFromFd Found 1 EGL devices
      8239.590220647 [11848-11848] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      8239.590223747 [11848-11848] ../src/export-buf.c: 133 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
      8239.590225197 [11848-11848] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
      8239.590226548 [11848-11848] ../src/export-buf.c: 215       findGPUIndexFromFd No match found, falling back to default device
      8239.591083263 [11848-11848] ../src/export-buf.c: 269         egl_initExporter Driver doesn't support 16-bit surfaces
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      8239.647503778 [11848-11848] ../src/vabackend.c:1852              nvTerminate Terminating 0x55e3c8b02050
      8239.647509068 [11848-11848] ../src/export-buf.c:  65      egl_releaseExporter Releasing exporter, 0 outstanding frames
      8239.647511779 [11848-11848] ../src/export-buf.c:  82      egl_releaseExporter Done releasing frames
      8239.668216174 [11848-11848] ../src/vabackend.c:1868              nvTerminate Now have 0 (0 max) instances
```

marioroy, 2022-08-06:
The video flickers. I verified the following in the Brave launch script.
```text
export LIBVA_DRIVER_NAME=nvidia
export NVD_BACKEND=direct
```
[image]

marioroy, 2022-08-06:
Working including consuming the video engine.  I needed to close out of Brave completely. There was a Brave window hiding behind a Terminal window.

[image]

marioroy, 2022-08-06:
Congrats @elFarto. Here, confirming the chrome-support branch working thus far for VP9 acceleration (e.g. Not, AV1 extension enabled).  I disabled the Not, AV1 extension. AV1 plays, but not accelerated or using the video engine.

marioroy, 2022-08-06:
Please, when the time comes, can you make a release (or tag) for the current master branch before merging in the chrome-support branch. The reason is chrome-support requires one to have the CUDA Toolkit installation. In my build script, I can check if the CUDA Toolkit is installed and if not fetch the latest release prior to the chrome-support feature.

marioroy, 2022-08-06:
This is the 2nd time I experienced the requirement for the CUDA Toolkit to match the Display Driver or the PTX may fail. The other time was with [NUMBA for CUDA GPUs](http://numba.pydata.org/numba-doc/latest/cuda/). Do you think worthy to append a note to the README file? E.g. "provided PTX was compiled with an unsupported toolchain" error.

elFarto, 2022-08-06:
I'm not planning on releasing this branch until I've done more testing, and got it working (however well) with the EGL backend aswell. The direct backend is a dubious idea at best, as I'm effectively poking around inside the NVIDIA kernel/Xorg driver, both of which don't have to keep compatibility between versions.

I did attempt to make nvcc optional, I've included the PTX file from my machine, which it should use if it's not available. How exactly is the build failing without it?

Could you attach a log of it not working with AV1 (with NVD_LOG=1 set), there shouldn't be an issue with this, but I can't test AV1.

marioroy, 2022-08-06:
Ah, I saw `Program nvcc found: NO` and mistakenly thought nvcc was required and yes building without it. Thank you for making nvcc (e.g. CUDA Toolkit) optional.

AV1 video playback:
```text
     12986.690383533 [16066-16066] ../src/vabackend.c:1919       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x26ec0034a800 10
     12986.690391504 [16066-16066] ../src/vabackend.c:1923       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
     12986.690392884 [16066-16066] ../src/vabackend.c:1940       __vaDriverInit_1_0 Now have 0 (0 max) instances
     12986.690394044 [16066-16066] ../src/vabackend.c:1965       __vaDriverInit_1_0 Selecting Direct backend
     12986.693667377 [16066-16066] ../src/direct/direct-export-buf.c:  55      direct_initExporter Manually opened DRM device
     12986.693672708 [16066-16066] ../src/direct/nv-driver.c: 208            init_nvdriver Initing nvdriver...
     12986.693675848 [16066-16066] ../src/direct/nv-driver.c: 213            init_nvdriver Got dev info: 100 1 2 6
     12986.693748979 [16066-16066] ../src/direct/nv-driver.c: 274            init_nvdriver NVIDIA kernel driver version: 510.85.02
     12986.769316983 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 6 (1) with 1 attributes
     12986.769323723 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.769325553 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 4 (nil) -1431655766
     12986.769327613 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 4 0x26ec010e7300 5
     12986.770444119 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 6 (1) with 0 attributes
     12986.770455199 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 7 (1) with 1 attributes
     12986.770456939 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.770458269 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 4 (nil) -1431655766
     12986.770459399 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 4 0x26ec010e7300 5
     12986.771591464 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 7 (1) with 0 attributes
     12986.771598145 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 13 (1) with 1 attributes
     12986.771599385 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.771600595 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 4 (nil) -1431655766
     12986.771602155 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 4 0x26ec010e7300 5
     12986.772737350 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 13 (1) with 0 attributes
     12986.772745670 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 14 (1) with 1 attributes
     12986.772747680 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.772749251 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 9 (nil) -1431655766
     12986.772750401 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 9 0x26ec010e7300 5
     12986.773874136 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 14 (1) with 0 attributes
     12986.773880996 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 19 (1) with 1 attributes
     12986.773882216 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.773883436 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 10 (nil) -1431655766
     12986.773884876 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 10 0x26ec010e7300 5
     12986.774992381 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 19 (1) with 0 attributes
     12986.775003421 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 32 (1) with 1 attributes
     12986.775005111 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.775006411 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 11 (nil) -1431655766
     12986.775007521 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with 11 0x26ec010e7300 5
     12986.776131717 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: 32 (1) with 0 attributes
     12986.776141817 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: -1 (10) with 1 attributes
     12986.776143727 [16066-16066] ../src/vabackend.c: 642           nvCreateConfig got config attrib: 0 0 1
     12986.776145017 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with -1 (nil) -1431655766
     12986.776146077 [16066-16066] ../src/vabackend.c:1597 nvQuerySurfaceAttributes with -1 0x26ec010e7300 5
     12986.776147127 [16066-16066] ../src/vabackend.c: 618           nvCreateConfig got profile: -1 (10) with 0 attributes
     12986.776150787 [16066-16066] ../src/vabackend.c:1852              nvTerminate Terminating 0x26ec0034a800
     12986.800846024 [16066-16066] ../src/vabackend.c:1868              nvTerminate Now have 0 (0 max) instances
[16066:16066:0806/124513.306174:ERROR:gpu_init.cc(486)] Passthrough is not supported, GL is desktop, ANGLE is
[16029:16029:0806/124513.411243:ERROR:CONSOLE(1)] "[Shields]: Can't request shields panel data for tabId: 11.  Error: No tab url specified", source: chrome-extension://mnojpmjdmbbfmejpflffifhffcmidifd/out/brave_extension_background.bundle.js (1)
```

elFarto, 2022-08-06:
I can't see anything obviously wrong with that log. It's possible VA-API use isn't enabled for AV1 yet within Chrome, although it is probing it (profile 32), it's not even attempting to use it. I'll investigate it further

marioroy, 2022-08-07:
Previously, I tested with Display Driver 510.85.02 and CUDA 11.6.2.

Next, I tried Display Driver 470.141.03 and CUDA 11.4.4. The build emits errors.

```text
$ sudo meson install -C build
ninja: Entering directory `/home/mario/nvidia-vaapi-driver/build'
[8/19] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
../src/h264.c: In function ‘computeH264CudaCodec’:
../src/h264.c:116:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  116 |     if (profile == VAProfileH264Baseline || profile == VAProfileH264ConstrainedBaseline || profile == VAProfileH264Main || profile == VAProfileH264High) {
      |     ^~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:7,
                 from ../src/h264.c:1:
/usr/include/va/va.h:502:5: note: declared here
  502 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
../src/h264.c: At top level:
../src/h264.c:128:5: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  128 |     VAProfileH264Baseline,
      |     ^~~~~~~~~~~~~~~~~~~~~
/usr/include/va/va.h:502:5: note: declared here
  502 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
[9/19] Compiling C object nvidia_drv_video.so.p/src_x11.c.o
../src/x11.c: In function ‘nvPutSurface’:
../src/x11.c:250:73: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  250 |     xcb_nv_glx_export_pixmap_response *xcb_ret = pixmapToNvHandle(conn, (int) draw, newXid, drv->driverContext.clientObject);
      |                                                                         ^
[16/19] Generating nvcc-ptx with a custom command
FAILED: YUVtoRGB.ptx 
/opt/cuda/bin/nvcc -allow-unsupported-compiler --ptx -o YUVtoRGB.ptx ../src/YUVtoRGB.cu
/usr/include/stdio.h(189): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(201): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(223): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(260): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(285): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(294): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(303): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(309): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(315): error: attribute "__malloc__" does not take arguments

/usr/include/stdio.h(830): error: attribute "__malloc__" does not take arguments

/usr/include/stdlib.h(579): error: attribute "__malloc__" does not take arguments

/usr/include/stdlib.h(583): error: attribute "__malloc__" does not take arguments

/usr/include/stdlib.h(812): error: attribute "__malloc__" does not take arguments

13 errors detected in the compilation of "../src/YUVtoRGB.cu".
[17/19] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
../src/vabackend.c: In function ‘nvQueryConfigProfiles’:
../src/vabackend.c:461:9: warning: ‘VAProfileH264Baseline’ is deprecated [-Wdeprecated-declarations]
  461 |         profile_list[profiles++] = VAProfileH264Baseline;
      |         ^~~~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32,
                 from ../src/vabackend.h:7,
                 from ../src/vabackend.c:3:
/usr/include/va/va.h:502:5: note: declared here
  502 |     VAProfileH264Baseline va_deprecated_enum = 5,
      |     ^~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
Could not rebuild /home/mario/nvidia-vaapi-driver/build
```

elFarto, 2022-08-07:
Looks like that's a known issue with the version of glibc/system header files. I think in this case we should just fall back to using the prebuilt PTX. Infact, we should use that by default unless an option is specified to rebuild it from source. I can invision a lot of nvcc/gcc/glibc compatibility issues if we don't, and it's just not necessary for everyone to rebuild it.

marioroy, 2022-08-07:
I will try. Installation succeeded using Display Driver 470.141.03 without nvcc/CUDA Toolkit. Next I tried the Brave browser, but zero video engine utilization.
```text
       289.597619344 [3015-3015] ../src/vabackend.c:1919       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x2e240034aa00 10
       289.597628015 [3015-3015] ../src/vabackend.c:1923       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
       289.597629415 [3015-3015] ../src/vabackend.c:1940       __vaDriverInit_1_0 Now have 0 (0 max) instances
       289.597630545 [3015-3015] ../src/vabackend.c:1965       __vaDriverInit_1_0 Selecting Direct backend
       289.600749685 [3015-3015] ../src/direct/direct-export-buf.c:  55      direct_initExporter Manually opened DRM device
       289.600754645 [3015-3015] ../src/direct/nv-driver.c: 208            init_nvdriver Initing nvdriver...
       289.600758065 [3015-3015] ../src/direct/nv-driver.c: 213            init_nvdriver Got dev info: 100 1 2 6
       289.600857739 [3015-3015] ../src/direct/nv-driver.c: 274            init_nvdriver NVIDIA kernel driver version: 470.141.03
       289.759704133 [3015-3015] ../src/vabackend.c:1998       __vaDriverInit_1_0 cuda error 'the provided PTX was compiled with an unsupported toolchain.' (222)
```

elFarto, 2022-12-07:
Closing issue as I don't think I'll be releasing that chrome-support branch. The amount of code required to make that work is excessive, requiring knowing how the GPU lays out textures in memory. There does seem to be some work to get Chrome to use DMA-BUF, so it should be easier to support once that's in.

closed 2022-12-07 by elFarto