elFarto/nvidia-vaapi-driver #102: [470.129.06 drivers] Not working on Firefox 103

guihkx, 2022-07-27:
On Firefox 103.0, I get the following output, testing with [this mp4 video](https://docs.evostream.com/sample_content/assets/bun33s.mp4):

```
$ NVD_MAX_INSTANCES=3 NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia firefox
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
Missing chrome or resource URL: resource://gre/modules/UpdateListener.jsm
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
     29801.507165713 [17633-17656] ../src/vabackend.c:1780       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f653d2225c0 31
     29801.507182618 [17633-17656] ../src/vabackend.c:1801       __vaDriverInit_1_0 Now have 0 (3 max) instances
     29801.515643700 [17633-17656] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 0
     29801.517501676 [17633-17656] ../src/export-buf.c: 172       findGPUIndexFromFd Found 3 EGL devices
     29801.517864945 [17633-17656] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     29801.517874837 [17633-17656] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 0
     29801.517908553 [17633-17656] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
     29801.534404428 [17633-17656] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
     29801.534417835 [17633-17656] ../src/export-buf.c:  99                reconnect Reconnecting to stream
     29801.534505303 [17633-17656] ../src/vabackend.c:1129      nvQueryImageFormats In nvQueryImageFormats
     29801.571613489 [17633-17656] ../src/vabackend.c: 557           nvCreateConfig got profile: 6 with 0 attributes
     29801.571631236 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 0
     29801.571635046 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c6780 5
     29801.574606846 [17633-17656] ../src/vabackend.c: 557           nvCreateConfig got profile: 7 with 0 attributes
     29801.574624150 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 0
     29801.574627718 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c6b80 5
     29801.577077605 [17633-17656] ../src/vabackend.c: 557           nvCreateConfig got profile: 13 with 0 attributes
     29801.577089088 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 0
     29801.577092903 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c6f80 5
     29801.616137469 [17633-17656] ../src/vabackend.c: 557           nvCreateConfig got profile: 13 with 0 attributes
     29801.616155233 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 0
     29801.616158980 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c6300 5
     29801.618347346 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 32613
     29801.618354962 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c6500 5
     29801.620519816 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66bc00)
     29801.620528615 [17633-17656] ../src/vabackend.c:1221            nvDeriveImage In nvDeriveImage
     29801.620534804 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c420)
     29801.620538772 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c4c0)
     29801.620542442 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c560)
     29801.620561625 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c920)
     29801.620575053 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c600)
     29801.620587154 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cba0)
     29801.620591908 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c6a0)
     29801.620598362 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c740)
     29801.620610580 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cf60)
     29801.620622658 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c7e0)
     29801.620629189 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c880)
     29801.620642711 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66ce20)
     29801.620648785 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66c9c0)
     29801.620653857 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66ca60)
     29801.620658633 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cb00)
     29801.620663963 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cc40)
     29801.620677149 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cce0)
     29801.620681811 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cd80)
     29801.620686417 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a66cec0)
     29801.620693030 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a7dcac0)
     29801.620708933 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a7dd100)
     29801.620715544 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a7dc340)
     29801.620720639 [17633-17656] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 480x272, format 1 (0x7f651a7dc020)
     29801.652895374 [17633-17656] ../src/vabackend.c: 557           nvCreateConfig got profile: 13 with 0 attributes
     29801.652915995 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 0
     29801.652920293 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c7e80 5
     29801.655327241 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 (nil) 32613
     29801.655337177 [17633-17656] ../src/vabackend.c:1497 nvQuerySurfaceAttributes with 4 0x7f651a8c7900 5
     29801.657963994 [17633-17656] ../src/vabackend.c: 748          nvCreateContext with 24 render targets, 24 surfaces, at 480x272
     29801.667874692 [17633-17669] ../src/vabackend.c: 343          resolveSurfaces [RT] Resolve thread for 0x7f651a529000 started
     29801.668009026 [17633-17656] ../src/vabackend.c:1032             nvEndPicture Decoded frame successfully to idx: 0 (0x7f651a7dc020)
     29801.668039509 [17633-17656] ../src/vabackend.c:1708    nvExportSurfaceHandle Exporting surface: 0 (0x7f651a7dc020)
     29801.668044598 [17633-17656] ../src/export-buf.c: 527           realiseSurface No free surfaces found
     29801.668057682 [17633-17669] ../src/vabackend.c: 358          resolveSurfaces Reading from queue: 0 1
     29801.668076321 [17633-17669] ../src/vabackend.c: 373          resolveSurfaces Mapping surface 0
[... 74 lines cut ...]
     29801.745571033 [17633-17668] ../src/vabackend.c: 720        nvDestroySurfaces Destroying surface -1 (0x7f651a66c6a0)
     29801.745574747 [17633-17668] ../src/export-buf.c: 341 detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     29801.745578771 [17633-17668] ../src/vabackend.c: 720        nvDestroySurfaces Destroying surface -1 (0x7f651a66cba0)
[... cut ...]
     29801.745639625 [17633-17668] ../src/export-buf.c: 318      destroyBackingImage Destroying BackingImage: 0x7f651a8c7880
     29801.745848543 [17633-17668] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
     29801.746923984 [17633-17668] ../src/export-buf.c:  82          releaseExporter Done releasing frames
     29801.759359816 [17633-17668] ../src/vabackend.c:1771              nvTerminate Now have 0 (3 max) instances
```

- OS: Arch Linux
- GPU: GTX 660

guihkx, 2022-07-27:
Additional log with  `MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5"`

```
[Parent 18483: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 18483: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 18483: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
[Parent 18483: Main Thread]: D/Dmabuf DMABuf is enabled
Missing chrome or resource URL: resource://gre/modules/UpdateListener.jsm
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
[Parent 18483: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Chrome process
[Parent 18483: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Parent 18483: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Parent 18483: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Child 18716: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 18716: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Child 18716: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD128 error Permission denied
[Child 18716: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 18716: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 0
[Child 18716: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[Child 18716: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type
[RDD 18739: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 18739: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[RDD 18739: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[RDD 18739: Main Thread]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[Child 18716: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility decoder supports requested type
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 18739: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 18739: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 18739: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/vorbis'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 18739: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/x-wav'
[RDD 18739: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/x-wav'
[RDD 18739: MediaSupervisor #1]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 18739: MediaSupervisor #1]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 18739: MediaSupervisor #1]: D/PlatformDecoderModule FFMPEG: FFmpegVideoDecoder::FFmpegVideoDecoder MIME video/avc Codec ID 27
[RDD 18739: MediaSupervisor #1]: D/Dmabuf nsDMABufDevice::IsDMABufVAAPIEnabled: EGL 1 media_ffmpeg_vaapi_enabled 1 CanUseHardwareVideoDecoding 1 XRE_IsRDDProcess 1
[... cut ...]
     30031.286859484 [18739-18758] ../src/vabackend.c:1757              nvTerminate Terminating 0x7f4bfa423880
     30031.286862899 [18739-18758] ../src/export-buf.c: 318      destroyBackingImage Destroying BackingImage: 0x7f4bd7ac7800
     30031.287078482 [18739-18758] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
     30031.288154706 [18739-18758] ../src/export-buf.c:  82          releaseExporter Done releasing frames
     30031.295040547 [18739-18758] ../src/vabackend.c:1771              nvTerminate Now have 0 (3 max) instances
```

elFarto, 2022-07-27:
Looks like it's another case of picking the wrong EGL backend. I'll see if I can figure out what it's doing.

guihkx, 2022-07-27:
>You may be able to work around the issue by removing/renaming `/usr/share/glvnd/egl_vendor.d/50_mesa.json`

This actually works, thanks!

eglinfo: 

```
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base EGL_EXT_device_enumeration
    EGL_EXT_device_query EGL_KHR_client_get_all_proc_addresses
    EGL_EXT_client_extensions EGL_KHR_debug EGL_KHR_platform_x11
    EGL_EXT_platform_x11 EGL_EXT_platform_device EGL_KHR_platform_wayland
    EGL_EXT_platform_wayland EGL_MESA_platform_xcb EGL_MESA_platform_gbm
    EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless

GBM platform:
EGL API version: 1.4
EGL vendor string: Mesa Project
EGL version string: 1.4
EGL client APIs: OpenGL OpenGL_ES 
EGL driver name: kms_swrast
EGL extensions string:
    EGL_ANDROID_blob_cache EGL_EXT_buffer_age
    EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers
    EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context
    EGL_KHR_create_context_no_error EGL_KHR_fence_sync
    EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace
    EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image
    EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image
    EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap
    EGL_KHR_no_config_context EGL_KHR_reusable_sync
    EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float
    EGL_KHR_wait_sync EGL_MESA_configless_context
    EGL_MESA_image_dma_buf_export EGL_MESA_query_driver
Configurations:
     bf lv colorbuffer dp st  ms    vis   cav bi  renderable  supported
  id sz  l  r  g  b  a th cl ns b    id   eat nd gl es es2 vg surfaces 
---------------------------------------------------------------------
0x01 32  0 10 10 10  2  0  0  0 0 0x30335241--         y  y  y     win
0x02 32  0 10 10 10  2 16  0  0 0 0x30335241--         y  y  y     win
0x03 32  0 10 10 10  2 24  0  0 0 0x30335241--         y  y  y     win
0x04 32  0 10 10 10  2 24  8  0 0 0x30335241--         y  y  y     win
0x05 32  0 10 10 10  2 32  0  0 0 0x30335241--         y  y  y     win
0x06 32  0 10 10 10  2  0  0  4 1 0x30335241--         y  y  y     win
0x07 32  0 10 10 10  2 16  0  4 1 0x30335241--         y  y  y     win
0x08 32  0 10 10 10  2 24  0  4 1 0x30335241--         y  y  y     win
0x09 32  0 10 10 10  2 24  8  4 1 0x30335241--         y  y  y     win
0x0a 32  0 10 10 10  2 32  0  4 1 0x30335241--         y  y  y     win
0x0b 30  0 10 10 10  0  0  0  0 0 0x30335258--         y  y  y     win
0x0c 30  0 10 10 10  0 16  0  0 0 0x30335258--         y  y  y     win
0x0d 30  0 10 10 10  0 24  0  0 0 0x30335258--         y  y  y     win
0x0e 30  0 10 10 10  0 24  8  0 0 0x30335258--         y  y  y     win
0x0f 30  0 10 10 10  0 32  0  0 0 0x30335258--         y  y  y     win
0x10 30  0 10 10 10  0  0  0  4 1 0x30335258--         y  y  y     win
0x11 30  0 10 10 10  0 16  0  4 1 0x30335258--         y  y  y     win
0x12 30  0 10 10 10  0 24  0  4 1 0x30335258--         y  y  y     win
0x13 30  0 10 10 10  0 24  8  4 1 0x30335258--         y  y  y     win
0x14 30  0 10 10 10  0 32  0  4 1 0x30335258--         y  y  y     win
0x15 32  0  8  8  8  8  0  0  0 0 0x34325241--         y  y  y     win
0x16 32  0  8  8  8  8 16  0  0 0 0x34325241--         y  y  y     win
0x17 32  0  8  8  8  8 24  0  0 0 0x34325241--         y  y  y     win
0x18 32  0  8  8  8  8 24  8  0 0 0x34325241--         y  y  y     win
0x19 32  0  8  8  8  8 32  0  0 0 0x34325241--         y  y  y     win
0x1a 32  0  8  8  8  8  0  0  4 1 0x34325241--         y  y  y     win
0x1b 32  0  8  8  8  8 16  0  4 1 0x34325241--         y  y  y     win
0x1c 32  0  8  8  8  8 24  0  4 1 0x34325241--         y  y  y     win
0x1d 32  0  8  8  8  8 24  8  4 1 0x34325241--         y  y  y     win
0x1e 32  0  8  8  8  8 32  0  4 1 0x34325241--         y  y  y     win
0x1f 24  0  8  8  8  0  0  0  0 0 0x34325258--         y  y  y     win
0x20 24  0  8  8  8  0 16  0  0 0 0x34325258--         y  y  y     win
[... 450 lines cut ...]
0x1b 32  0  8  8  8  8 16  0  4 1 0x00--      a  y  y  y     pb
0x1c 32  0  8  8  8  8 24  0  4 1 0x00--      a  y  y  y     pb
0x1d 32  0  8  8  8  8 24  8  4 1 0x00--      a  y  y  y     pb
0x1e 32  0  8  8  8  8 32  0  4 1 0x00--      a  y  y  y     pb
0x1f 24  0  8  8  8  0  0  0  0 0 0x00--      y  y  y  y     pb
0x20 24  0  8  8  8  0 16  0  0 0 0x00--      y  y  y  y     pb
0x21 24  0  8  8  8  0 24  0  0 0 0x00--      y  y  y  y     pb
0x22 24  0  8  8  8  0 24  8  0 0 0x00--      y  y  y  y     pb
0x23 24  0  8  8  8  0 32  0  0 0 0x00--      y  y  y  y     pb
0x24 24  0  8  8  8  0  0  0  4 1 0x00--      y  y  y  y     pb
0x25 24  0  8  8  8  0 16  0  4 1 0x00--      y  y  y  y     pb
0x26 24  0  8  8  8  0 24  0  4 1 0x00--      y  y  y  y     pb
0x27 24  0  8  8  8  0 24  8  4 1 0x00--      y  y  y  y     pb
0x28 24  0  8  8  8  0 32  0  4 1 0x00--      y  y  y  y     pb
0x29 16  0  5  6  5  0  0  0  0 0 0x00--      y  y  y  y     pb
0x2a 16  0  5  6  5  0 16  0  0 0 0x00--      y  y  y  y     pb
0x2b 16  0  5  6  5  0 24  0  0 0 0x00--      y  y  y  y     pb
0x2c 16  0  5  6  5  0 24  8  0 0 0x00--      y  y  y  y     pb
0x2d 16  0  5  6  5  0 32  0  0 0 0x00--      y  y  y  y     pb
0x2e 16  0  5  6  5  0  0  0  4 1 0x00--      y  y  y  y     pb
0x2f 16  0  5  6  5  0 16  0  4 1 0x00--      y  y  y  y     pb
0x30 16  0  5  6  5  0 24  0  4 1 0x00--      y  y  y  y     pb
0x31 16  0  5  6  5  0 24  8  4 1 0x00--      y  y  y  y     pb
0x32 16  0  5  6  5  0 32  0  4 1 0x00--      y  y  y  y     pb
```

elFarto, 2022-07-28:
That's what I expected, MESA surfaceless being selected before the NVIDIA one. Firefox searches for a display in the following order: EGLDevice that has a render node equal to the one Firefox opened (requires `EGL_EXT_device_drm_render_node`), a surfaceless context, then if those fail, the default context.

I think we'd need a patch to Firefox to support EGL devices that don't implement `EGL_EXT_device_drm_render_node`, which should be possible as it supports `EGL_EXT_device_drm` which lets you get the master device file which could be used instead.

closed 2022-07-28 by elFarto