elFarto/nvidia-vaapi-driver #105: Q: VA-API driver documentation

mar0x, 2022-08-03:
Hello.
I'm unable to find the documentation for VA-API user space driver developers. Is there is a something like OpenMAX IL Specification? Could you please share the documentation you are using.

elFarto, 2022-08-03:
I've not found any documentation for VA-API driver developers, as far as I can tell none exists. I worked out what to do by looking at the other open source drivers and lots of trial and error.

mar0x, 2022-08-03:
Thanks for reply!

> I worked out what to do by looking at the other open source drivers and lots of trial and error.

Yes, it seems to be the only source. But someone have created those open source drivers before! :)

Best regard and happy sources digging,
Max

closed 2022-08-03 by mar0x