elFarto/nvidia-vaapi-driver #107: Lag when playing VR videos

weter11, 2022-08-14:
Hi! There are some problems with playing VR180, VR360 videos: no any problems, when resolution low like 240p or 480p, but starting with 720p fps dropping and bigger the resolution, bigger lags. First I thought that the problem in my CPU power {5800H}, so I install driver and now I can play any video on YT like 8k60p in AV1 codec smoothly, `Video Engine Utilization` report 20-50% utilization, and VR videos also report some usage of hardware decoder, but videos very laggy at 2160p.
So Its a problem with firefox WebGL implementation or there are some additional settings needed for VR videos?

elFarto, 2022-08-15:
I've noticed this too, but I'm not sure what's going on. I don't think it's this driver causing it. It's possible it's related to the previous issue we've had around Firefox's slowness decoding large videos (see issue #13).

There are no settings I know of that help the performance.

elFarto, 2023-01-07:
I did do a bit more investigation into this and it appears that for each frame Firefox needs to read it back from the GPU to Javascript, so it can be uploaded to WebGL for it to be distorted, which is hideously inefficient. This combined with Firefox's one frame at a time decoding style means we end up with a lot of lag.

I'll see about getting an issue raised for this in Firefox's bug tracker.

elFarto, 2023-01-07:
Issue has been raised here: https://bugzilla.mozilla.org/show_bug.cgi?id=1809026

weter11, 2023-10-22:
Looks like YT360 fixed, also videos on ZDF and interactive videos on WDR also working now good. At the same time looks like some features still absent (one of websites on which videos sometimes stop to work report that my browser don't support playback), because videos on DeoVR still laggy even when you decrease quality to minimum they looks like 5fps show on any resolution. Thanks for support.

closed 2023-10-22 by weter11