elFarto/nvidia-vaapi-driver #108: Firefox acceleration not working

scamdotnet, 2022-08-15:
Hi, I just installed this driver, and setup everything in about:config as well as the environment variables, but I am getting these errors when launching Firefox from the command line:
```
[GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection.
Missing chrome or resource URL: resource://gre/modules/UpdateListener.jsm
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
```
When I open the YouTube home page, I additionally get these errors:
```
[2022-08-15T20:22:40Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-15T20:22:43Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-15T20:22:43Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-15T20:22:43Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
```
Attempting to play back any video spams the log with the following errors repeatedly, and the Video Engine Utilization meter in the NVIDIA Settings app shows no usage, even on 8K videos:
```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
```
The `vainfo` command returns the following:
```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
```

Any help would be appreciated :)

Thanks,
             Scam.net

elFarto, 2022-08-15:
Are you running with EGL? Take a look in the `about:support` page.

scamdotnet, 2022-08-15:
I wasn't sure where in the about:support page you wanted me to look, but I just did a search for EGL and took screenshots of the results.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Hope this helps!

weter11, 2022-08-16:
`gfx.x11-egl.force-enabled` need to be set true.
Also update vaapi to 1.15.0, that version have support for av1.

scamdotnet, 2022-08-16:
> `gfx.x11-egl.force-enabled` need to be set true.
I already have this option enabled.

> Also is it nvidia-vaapi-driver version 0.0.5 or 0.0.6?
How can I check this with the installed driver file?

scamdotnet, 2022-08-17:
> `gfx.x11-egl.force-enabled` need to be set true.

Yep, I have that set in `about:config`

> Also is it nvidia-vaapi-driver version 0.0.5 or 0.0.6?

Is there a way to check this? I just did a git clone from the master branch.

weter11, 2022-08-17:
Yeah, its the latest. And /etc/environment have options
```
LIBVA_DRIVER_NAME=nvidia
MOZ_DISABLE_RDD_SANDBOX=1
```
But I still recommend to update libva, libva2 and vainfo to the latest versions 2.15.0.

scamdotnet, 2022-08-17:
> Yeah, its the latest. And /etc/environment have options
> 
> ```
> LIBVA_DRIVER_NAME=nvidia
> MOZ_DISABLE_RDD_SANDBOX=1
> ```

Hi, yes I set this up before in /etc/profile.d, but I changed also applied this in /etc/environment as you said as well.

> But I still recommend to update libva, libva2 and vainfo to the latest versions 2.15.0.

I tried doing this by setting up the ubuntu kinetic repositories, and upgrading libva-dev, libva-drm2, libva-glx2, libva-wayland2, libva-x11-2, libva2, and vainfo, then saw that the latest version of vainfo offered in these repositories is still 2.12.0. I tried negating this by downloading the source for libva-utils, the source package for vainfo and other utilities, and building and installing that. This caused further issues when running the command, either with just `vainfo` or `./vainfo/vainfo` in the source directory after building, with the following output from those commands:
```
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/local/lib/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
Why is it saying VA-API version 1.15.0? I wonder if this is a bug with libva-utils where it isn't picking up the version 2.x.x part correctly. Anyways, as for the further output, this seems to be an issue with vainfo trying to look for the files in /usr/local/lib instead of /usr/lib. I decided to fix this by making a sim-link to the right file installed by this repository. Now running va-info appears to work, although still giving strange version numbers.

Attempting to run firefox with these changes seems to work a little better, with nvidia-settings showing no usage on video encoding, but the firefox binary only spitting out the following messages when playing a 4k video on youtube:
```
[GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection.
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
[2022-08-17T19:21:19Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-17T19:21:19Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-17T19:21:19Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-17T19:21:19Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
```
Previously, I would see issues stating that the driver installed by this repo could not be loaded being spammed repeatedly in the output of the firefox command. Additonally, the Nvidia-smi command only shows G next to the firefox executable, instead of the C that this repository said it would show if it was using acceleration. The video lag seems to be gone with 4k videos, but I seem to be missing the 8k option in the same videos that i tried last time.

I suspect this means I am closer to getting acceleration working in firefox, but I am not there yet. Any further assistance would be greatly appreciated.

elFarto, 2022-08-17:
It's not required to have libva 2.15.0, it should work just as well with prior versions. VA-API does print 1.15.0 due to some weird stuff with they way they version that library, it is correct.

As for the `failed to initialise VAAPI connection` error, I'm not entirely sure. The error seems to indicate that the vaInitalise method failed. Can you try running Firefox with `NVD_LOG=1` set, and see what it prints out.

scamdotnet, 2022-08-17:
> Can you try running Firefox with `NVD_LOG=1` set, and see what it prints out.

I did this, and it seems to output the same exact thing as before: (full log of `NVD_LOG=1 firefox`, opening a 4k youtube video, and closing firefox):
```
[GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection.
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
[2022-08-17T20:28:57Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-17T20:28:57Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-17T20:28:57Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
[2022-08-17T20:28:57Z ERROR mp4parse] Found 2 nul bytes in "\u{0}\u{0}"
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
[2022-08-17T20:29:19Z ERROR viaduct::backend::ffi] Missing HTTP status
[2022-08-17T20:29:19Z ERROR viaduct::backend::ffi] Missing HTTP status
```

elFarto, 2022-08-19:
Ok, that indicates that libva can't even load the driver, since there's none of our logging being printed. I'm a little stumped at this. Normally this sort of failure happens when it just can't find the file, but you've said that vainfo is giving the correct output.

At this point I would reinstall the distro provided libva and vainfo. Then rerun vainfo/Firefox with NVD_LOG=1 set.

weter11, 2022-08-19:
Is installed version of firefox 100% from ppa, not a snap? Ubuntu by default if you don't change that in settings update ppa version to snap.

scamdotnet, 2022-08-19:
> Is installed version of firefox 100% from ppa, not a snap? Ubuntu by default if you don't change that in settings update ppa version to snap.

Yes, I made sure to follow a guide to install it as a ppa, and the guide included steps to set the ppa as a higher priority for updates than the snap.

scamdotnet, 2022-08-19:
> I will make sure to double check this however.

Update: I checked this just now, and it appears that I am running firefox from a ppa

> At this point I would reinstall the distro provided libva and vainfo.

This is going to be fun because the last time I did this apt decided that it didn't need my entire de lol, but I'll make sure to keep note of the packages that it will remove.

scamdotnet, 2022-08-19:
Well it appears I may have broken things even more. When I run `vainfo` with or without `NVD_LOG=1`, it now returns the following:
`vainfo : symbol lookup error: /lib/x86_64-linux-gnu/libva-x11.so.2: undefined symbol: va_fool postp`
I wonder if this is due to me building a newer version of va info and using that with the equivalent version of the va libraries, which I just downgraded to the ubuntu jammy version. Let me see if I can fix this...

multiflexi, 2022-10-11:
I have the same issue. I tried mpv with VAAPI and this is the output:
```
841.812679856 [21492-21506] ../src/vabackend.c:1781       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f78ec1ec4b0 10
       841.812701897 [21492-21506] ../src/vabackend.c:1785       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
       841.812710282 [21492-21506] ../src/vabackend.c:1802       __vaDriverInit_1_0 Now have 0 (0 max) instances
       841.812731644 [21492-21506] ../src/export-buf.c: 161       findGPUIndexFromFd Looking for GPU index: 0
       841.813604331 [21492-21506] ../src/export-buf.c: 172       findGPUIndexFromFd Found 3 EGL devices
       841.813778480 [21492-21506] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       841.813791409 [21492-21506] ../src/export-buf.c: 133 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
       841.813798409 [21492-21506] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
       841.903740842 [21492-21506] ../src/export-buf.c: 267             initExporter Driver doesn't support 16-bit surfaces
       841.903754680 [21492-21506] ../src/export-buf.c:  99                reconnect Reconnecting to stream
       841.903883573 [21492-21506] ../src/vabackend.c:1130      nvQueryImageFormats In nvQueryImageFormats
       841.927306292 [21492-21506] ../src/vabackend.c: 557           nvCreateConfig got profile: 0 with 0 attributes
       841.927322369 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 1 (nil) 0
       841.927335035 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 1 0x7f78ec80fe80 5
       841.928925640 [21492-21506] ../src/vabackend.c: 686        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7f78ec810ee0)
       841.928936830 [21492-21506] ../src/vabackend.c:1222            nvDeriveImage In nvDeriveImage
       841.929006245 [21492-21506] ../src/export-buf.c:  60                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x2)

       841.929017371 [21492-21506] ../src/export-buf.c:  60                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7f78ec65bed0)

       841.929025733 [21492-21506] ../src/export-buf.c:  60                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7f78ec452eb0)

       841.929033885 [21492-21506] ../src/vabackend.c:1709    nvExportSurfaceHandle Exporting surface: -1 (0x7f78ec810ee0)
       841.929038657 [21492-21506] ../src/export-buf.c: 527           realiseSurface No free surfaces found
       841.929194347 [21492-21506] ../src/export-buf.c: 443     allocateBackingImage Presenting frame -1 128x128 (0x7f78ec810ee0, 0x7f78ec811d80, 0x7f78ec8156e0)
       841.929272460 [21492-21506] ../src/export-buf.c: 461     allocateBackingImage Adding frame from EGLStream: 0x7f78ec8105c1
       841.929284325 [21492-21506] ../src/export-buf.c: 472     allocateBackingImage Acquired image from EGLStream: 0x7f78ec8105c1
       841.929294448 [21492-21506] ../src/export-buf.c: 283       exportBackingImage eglExportDMABUFImageQueryMESA: 0x7f78ec8162a0 NV12 (3231564e) planes:2 mods:300000000606014 300000000606013
       841.929644993 [21492-21506] ../src/vabackend.c:1750    nvExportSurfaceHandle Exporting with 128 128 0 128 0 128
       841.929653380 [21492-21506] ../src/vabackend.c:1063            nvSyncSurface Syncing on surface: -1 (0x7f78ec810ee0)
       841.929658255 [21492-21506] ../src/vabackend.c:1078            nvSyncSurface Surface -1 resolved (0x7f78ec810ee0)
       841.930033517 [21492-21506] ../src/vabackend.c: 720        nvDestroySurfaces Destroying surface -1 (0x7f78ec810ee0)
       841.930042646 [21492-21506] ../src/export-buf.c: 353 detachBackingImageFromSurface Detaching BackingImage 0x7f78ec8162a0 from Surface 0x7f78ec810ee0
       841.930049360 [21492-21506] ../src/vabackend.c: 557           nvCreateConfig got profile: 1 with 0 attributes
       841.930055956 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 1 (nil) 0
       841.930061009 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 1 0x7f78ec8145c0 5
       841.931761242 [21492-21506] ../src/vabackend.c: 557           nvCreateConfig got profile: 8 with 0 attributes
       841.931772713 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 3 (nil) 0
       841.931778311 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 3 0x7f78ec814700 5
       841.933548564 [21492-21506] ../src/vabackend.c: 557           nvCreateConfig got profile: 9 with 0 attributes
       841.933559842 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 3 (nil) 0
       841.933564594 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 3 0x7f78ec80cb40 5
       841.935636511 [21492-21506] ../src/vabackend.c: 557           nvCreateConfig got profile: 10 with 0 attributes
       841.935647107 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 3 (nil) 0
       841.935659571 [21492-21506] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 3 0x7f78ec80cb40 5
       841.937588278 [21492-21506] ../src/vabackend.c: 557           nvCreateConfig got profile: 5 with 0 attributes
[... cut ...]
       842.442113101 [21492-21506] ../src/vabackend.c:1750    nvExportSurfaceHandle Exporting with 3840 2026 0 3840 0 3840
       842.442117852 [21492-21506] ../src/vabackend.c:1063            nvSyncSurface Syncing on surface: 6 (0x55b8408ed480)
       842.442121798 [21492-21506] ../src/vabackend.c:1078            nvSyncSurface Surface 6 resolved (0x55b8408ed480)

[vo/gpu/opengl] after creating texture: OpenGL error INVALID_OPERATION.

```
and of course, it continues. I am on the master branch. Firefox behaves the same, vainfo works fine.

mashu, 2022-10-28:
I have the same issue on GNU/Linux Debian (sid).
```
LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 vainfo
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

Also many apps like discord, firefox fail to load nvidia_drv_video.so usually indicated by
```
libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
```

mashu, 2022-10-28:
Update, problem solved.

I run ` NVD_LOG=1 vainfo`

Which indicated that loading nvidia_drv_video.so requires nvidia-drm.modeset=1

I edited /**etc/defaut/grub** to contain
```
GRUB_CMDLINE_LINUX="nvidia-drm.modeset=1"
```
and rebuild grub
```
grub-mkconfig -o /boot/grub/grub.cfg
update-initramfs -k all # Just to be extra sure
```

Problem solved for me
```
vainfo 
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
```

Thanks !!