elFarto/nvidia-vaapi-driver #109: Cannot get Kodi to play along with it

jmarcet, 2022-09-02:
I use the driver with Firefox with no issues. With Kodi, which as far as I can read in the issues should also work, all I get is audio and a stalled picture which is updated every few seconds.

[Here](https://openwrt.marcet.info/u7d/nvidia-vaapi-kodi.log) are some logs.

This is all on an Optimus laptop (Dell XPS 9570) with Arch linux and only the nvidia driver, so everything is running on the nvidia card.

As a workaround I can use Kodi by forcing VDPAU, setting `LIBVA_DRIVER_NAME=mesa`. I miss hevc though.

elFarto, 2022-09-03:
I'm seeing the same issue here, but I can't see a cause for it. There's nothing in the driver that's causing it to run slowly, and there's nothing showing up in a profile of Kodi to suggest that it's getting stuck anywhere.

When you set the driver name to mesa, is it actually using VA-API, or does it print something like this if you run Kodi from the command line?:
```
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'mesa'
libva info: Trying to open /usr/lib64/dri/mesa_drv_video.so
libva info: va_openDriver() returns -1
```

jmarcet, 2022-09-03:
I don't get any output from the libva driver, and in Kodi's logs:

```
2022-09-03 13:56:13.771 T:135437 WARNING <general>: Pulseaudio module module-allow-passthrough not loaded - opening PT devices might fail
2022-09-03 13:56:13.771 T:135437    INFO <general>: Found 1 Lists of Devices
2022-09-03 13:56:13.771 T:135437    INFO <general>: Enumerated PULSE devices:
2022-09-03 13:56:13.771 T:135437    INFO <general>:     Device 1
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_deviceName      : Default
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_displayName     : Default
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_displayNameExtra: Default Output Device (PULSEAUDIO)
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_deviceType      : AE_DEVTYPE_PCM
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_channels        : FL, FR
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_streamTypes     : No passthrough capabilities
2022-09-03 13:56:13.771 T:135437    INFO <general>:     Device 2
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_deviceName      : alsa_output.pci-0000_00_1f.3.hdmi-stereo
2022-09-03 13:56:13.771 T:135437    INFO <general>:         m_displayName     : Built-in Audio Digital Stereo (HDMI)
2022-09-03 13:56:13.772 T:135437    INFO <general>:         m_displayNameExtra: HDMI / DisplayPort (PULSEAUDIO)
2022-09-03 13:56:13.772 T:135437    INFO <general>:         m_deviceType      : AE_DEVTYPE_PCM
2022-09-03 13:56:13.772 T:135437    INFO <general>:         m_channels        : FL, FR
2022-09-03 13:56:13.772 T:135437    INFO <general>:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
2022-09-03 13:56:13.772 T:135437    INFO <general>:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
2022-09-03 13:56:13.772 T:135437    INFO <general>:         m_streamTypes     : No passthrough capabilities
2022-09-03 13:56:13.772 T:135438    INFO <general>: CActiveAESink::OpenSink - initialize sink
2022-09-03 13:56:13.786 T:135438    INFO <general>: PulseAudio: Opened device Default in pcm mode with Buffersize 150 ms Periodsize 50 ms
2022-09-03 13:56:13.788 T:135422    INFO <general>: CKeyboardLayoutManager: loading keyboard layouts from special://xbmc/system/keyboardlayouts...
2022-09-03 13:56:14.087 T:135422 WARNING <general>: CXRandR::Query - output eDP-1-1 has no current mode, assuming disconnected
2022-09-03 13:56:14.087 T:135422    INFO <general>: Available videomodes (xrandr):
2022-09-03 13:56:14.087 T:135422    INFO <general>: Output 'DP-1-1' has 15 modes
2022-09-03 13:56:14.087 T:135422    INFO <general>: ID:0x22f Name:3840x2160 Refresh:59.996620 Width:3840 Height:2160
2022-09-03 13:56:14.087 T:135422    INFO <general>: Pixel Ratio: 1.000160
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x230 Name:3840x2160 Refresh:30.000000 Width:3840 Height:2160
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000160
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x231 Name:3840x2160 Refresh:29.969999 Width:3840 Height:2160
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000160
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x232 Name:2560x1440 Refresh:59.950550 Width:2560 Height:1440
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000159
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x233 Name:1920x1080 Refresh:60.000000 Width:1920 Height:1080
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000160
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x234 Name:1920x1080 Refresh:59.940201 Width:1920 Height:1080
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000160
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x235 Name:1920x1080 Refresh:59.933880 Width:1920 Height:1080
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000160
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x236 Name:1280x720 Refresh:60.000000 Width:1280 Height:720
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000159
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x237 Name:1280x720 Refresh:59.940201 Width:1280 Height:720
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.000159
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x1fa Name:1024x768 Refresh:60.003841 Width:1024 Height:768
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.333546
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x209 Name:800x600 Refresh:60.316540 Width:800 Height:600
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.333546
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x238 Name:720x480 Refresh:60.000000 Width:720 Height:480
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.185374
2022-09-03 13:56:14.088 T:135422    INFO <general>: ID:0x239 Name:720x480 Refresh:59.940060 Width:720 Height:480
2022-09-03 13:56:14.088 T:135422    INFO <general>: Pixel Ratio: 1.185374
[... cut ...]
2022-09-03 13:56:14.228 T:135422    INFO <general>: GL_EXTENSIONS = GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_viewport_index GL_AMD_vertex_shader_layer GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compatibility GL_ARB_compressed_texture_pixel_storage GL_ARB_cons>
2022-09-03 13:56:14.233 T:135422    INFO <general>: GL: Maximum texture width: 32768
2022-09-03 13:56:14.332 T:135422    INFO <general>: GL: Enabling VSYNC
```

jmarcet, 2022-09-03:
When playing a video with VDPAU:

```
2022-09-03 14:07:39.151 T:135422    INFO <general>: VideoPlayer::OpenFile: nfs://some_video.mkv
2022-09-03 14:07:39.151 T:142086    INFO <general>: Creating InputStream
2022-09-03 14:07:39.275 T:142088    INFO <general>: initializing python engine.
2022-09-03 14:07:39.309 T:142086    INFO <general>: Skipped 2 duplicate messages..
2022-09-03 14:07:39.309 T:142086    INFO <general>: Creating Demuxer
2022-09-03 14:07:39.312 T:135526    INFO <general>: JELLYFIN.jellyfin_kodi.objects.actions -> INFO::jellyfin_kodi/objects/actions.py:762  [ play ] kodi_id: 9621 media: episode
2022-09-03 14:07:39.384 T:142086    INFO <general>: Opening stream: 0 source: 256
2022-09-03 14:07:39.384 T:142086    INFO <general>: Creating video codec with codec id: 27
2022-09-03 14:07:39.384 T:142086    INFO <general>: CDVDVideoCodecFFmpeg::Open() Using codec: H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
2022-09-03 14:07:39.384 T:142086    INFO <general>: Creating video thread
2022-09-03 14:07:39.384 T:142092    INFO <general>: running thread: video_thread
2022-09-03 14:07:39.384 T:142086    INFO <general>: Opening stream: 1 source: 256
2022-09-03 14:07:39.384 T:142086    INFO <general>: Finding audio codec for: 86019
2022-09-03 14:07:39.384 T:142086    INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2022-09-03 14:07:39.384 T:142086    INFO <general>: Creating audio thread
2022-09-03 14:07:39.384 T:142093    INFO <general>: running thread: CVideoPlayerAudio::Process()
2022-09-03 14:07:39.384 T:142086    INFO <general>: Opening stream: 3 source: 256
2022-09-03 14:07:39.470 T:142092    INFO <general>: VDPAU::CreateContext - creating decoder context
2022-09-03 14:07:39.475 T:142093    INFO <general>: Creating audio stream (codec id: 86019, channels: 6, sample rate: 48000, no pass-through)
2022-09-03 14:07:39.475 T:135438    INFO <general>: CActiveAESink::OpenSink - initialize sink
2022-09-03 14:07:39.483 T:142092    INFO <general>: vdp_device = 0x00000001 vdp_st = 0x0
2022-09-03 14:07:39.483 T:142092    INFO <general>: VDPAU Decoder capabilities:
2022-09-03 14:07:39.483 T:142092    INFO <general>: name          level macbs width height
2022-09-03 14:07:39.483 T:142092    INFO <general>: ------------------------------------
2022-09-03 14:07:39.483 T:142092    INFO <general>: MPEG1             0 65536  4096  4096
2022-09-03 14:07:39.483 T:142092    INFO <general>: MPEG2_SIMPLE      3 65536  4096  4096
2022-09-03 14:07:39.483 T:142092    INFO <general>: MPEG2_MAIN        3 65536  4096  4096
2022-09-03 14:07:39.483 T:142092    INFO <general>: H264_BASELINE    51 65536  4096  4096
2022-09-03 14:07:39.483 T:142092    INFO <general>: H264_MAIN        51 65536  4096  4096
2022-09-03 14:07:39.483 T:142092    INFO <general>: H264_HIGH        51 65536  4096  4096
2022-09-03 14:07:39.483 T:142092    INFO <general>: VC1_SIMPLE        1  8190  2048  2048
2022-09-03 14:07:39.483 T:142092    INFO <general>: VC1_MAIN          2  8190  2048  2048
2022-09-03 14:07:39.483 T:142092    INFO <general>: VC1_ADVANCED      4  8190  2048  2048
2022-09-03 14:07:39.483 T:142092    INFO <general>: MPEG4_PART2_ASP   5  8192  2048  2048
2022-09-03 14:07:39.483 T:142092    INFO <general>: HEVC_MAIN        153 262144  8192  8192
2022-09-03 14:07:39.483 T:142092    INFO <general>: VP9_PROFILE_0     1 262144  8192  8192
2022-09-03 14:07:39.483 T:142092    INFO <general>: ------------------------------------
2022-09-03 14:07:39.483 T:142092    INFO <general>: Mixer feature: VDP_VIDEO_MIXER_FEATURE_NOISE_REDUCTION
2022-09-03 14:07:39.483 T:142092    INFO <general>: Mixer feature: VDP_VIDEO_MIXER_FEATURE_SHARPNESS
2022-09-03 14:07:39.483 T:142092    INFO <general>: Mixer feature: VDP_VIDEO_MIXER_FEATURE_DEINTERLACE_TEMPORAL
2022-09-03 14:07:39.483 T:142092    INFO <general>: Mixer feature: VDP_VIDEO_MIXER_FEATURE_DEINTERLACE_TEMPORAL_SPATIAL
2022-09-03 14:07:39.483 T:142092    INFO <general>: Mixer feature: VDP_VIDEO_MIXER_FEATURE_INVERSE_TELECINE
2022-09-03 14:07:39.483 T:142092    INFO <general>: Mixer feature: VDP_VIDEO_MIXER_FEATURE_HIGH_QUALITY_SCALING_L1
2022-09-03 14:07:39.492 T:142092    INFO <general>:  (VDPAU) screenWidth:1920 vidWidth:1920 surfaceWidth:1920
2022-09-03 14:07:39.492 T:142092    INFO <general>:  (VDPAU) screenHeight:1080 vidHeight:1080 surfaceHeight:1088
2022-09-03 14:07:39.498 T:142097    INFO <general>: COutput::OnStartup: Output Thread created
2022-09-03 14:07:39.499 T:142098    INFO <general>: CMixer::OnStartup: Output Thread created
2022-09-03 14:07:39.499 T:142098    INFO <general>:  (VDPAU) Creating the video mixer
2022-09-03 14:07:39.501 T:142097    INFO <general>: VDPAU::COutput::InitBufferPool - Output Surface created
2022-09-03 14:07:39.504 T:142098    INFO <general>: Skipped 3 duplicate messages..
2022-09-03 14:07:39.504 T:142098    INFO <general>: Setting Noise Reduction to 1.000000
2022-09-03 14:07:39.504 T:142098    INFO <general>: Setting Sharpness to 1.000000
2022-09-03 14:07:39.504 T:142098    INFO <general>: Setting Noise Reduction to 1.000000
2022-09-03 14:07:39.504 T:142098    INFO <general>: Setting Sharpness to 1.000000
2022-09-03 14:07:39.510 T:142097    INFO <general>: VDPAU::COutput::InitBufferPool - Output Surface created
2022-09-03 14:07:39.524 T:135422    INFO <general>: Skipped 2 duplicate messages..
2022-09-03 14:07:39.524 T:135422    INFO <general>: CInteropState::Init: vdpau gl interop initialized
[... cut ...]
2022-09-03 14:07:39.533 T:135422    INFO <general>: GL: Using VDPAU render method
2022-09-03 14:07:39.533 T:135422    INFO <general>: GL: Using GL_ARB_pixel_buffer_object
2022-09-03 14:07:39.533 T:135422    INFO <general>: Using GL_TEXTURE_2D
2022-09-03 14:07:39.533 T:135422 WARNING <general>: CLinuxRendererGL::UpdateVideoFilter - chosen scaling method 8, is not supported by renderer
2022-09-03 14:07:39.549 T:135422    INFO <general>: Loading skin file: VideoFullScreen.xml, load type: KEEP_IN_MEMORY
```

jmarcet, 2022-09-03:
Meanwhile, with vaapi:

```
2022-09-03 14:11:21.599 T:144225    INFO <general>: VideoPlayer::OpenFile: nfs://some_video.mkv
2022-09-03 14:11:21.599 T:144627    INFO <general>: Creating InputStream
2022-09-03 14:11:21.713 T:144627    INFO <general>: Creating Demuxer
2022-09-03 14:11:21.720 T:144629    INFO <general>: initializing python engine.
2022-09-03 14:11:21.746 T:144336    INFO <general>: Skipped 2 duplicate messages..
2022-09-03 14:11:21.746 T:144336    INFO <general>: JELLYFIN.jellyfin_kodi.objects.actions -> INFO::jellyfin_kodi/objects/actions.py:762  [ play ] kodi_id: 9621 media: episode
2022-09-03 14:11:21.776 T:144627    INFO <general>: Opening stream: 0 source: 256
2022-09-03 14:11:21.776 T:144627    INFO <general>: Creating video codec with codec id: 27
2022-09-03 14:11:21.776 T:144627    INFO <general>: CDVDVideoCodecFFmpeg::Open() Using codec: H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
2022-09-03 14:11:21.776 T:144627    INFO <general>: Creating video thread
2022-09-03 14:11:21.776 T:144633    INFO <general>: running thread: video_thread
2022-09-03 14:11:21.776 T:144627    INFO <general>: Opening stream: 1 source: 256
2022-09-03 14:11:21.776 T:144627    INFO <general>: Finding audio codec for: 86019
2022-09-03 14:11:21.776 T:144627    INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2022-09-03 14:11:21.776 T:144627    INFO <general>: Creating audio thread
2022-09-03 14:11:21.776 T:144634    INFO <general>: running thread: CVideoPlayerAudio::Process()
2022-09-03 14:11:21.776 T:144627    INFO <general>: Opening stream: 3 source: 256
2022-09-03 14:11:21.897 T:144634    INFO <general>: Creating audio stream (codec id: 86019, channels: 6, sample rate: 48000, no pass-through)
2022-09-03 14:11:21.956 T:144239    INFO <general>: CActiveAESink::OpenSink - initialize sink
2022-09-03 14:11:21.994 T:144639    INFO <general>: COutput::OnStartup: Output Thread created
2022-09-03 14:11:21.994 T:144639    INFO <general>: VAAPI::SupportsFilter vaDeriveImage not supported by driver - ffmpeg postprocessing and CPU-copy rendering will not be available
2022-09-03 14:11:21.994 T:144639   ERROR <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
2022-09-03 14:11:22.065 T:144629    INFO <general>: CPythonInvoker(47, /home/jmarcet/.kodi/addons/script.embuary.helper/plugin.py): script successfully run
2022-09-03 14:11:22.066 T:144225    INFO <general>: GL: Selecting YUV 2 RGB shader
2022-09-03 14:11:22.066 T:144225    INFO <general>: GL: Using GL_ARB_pixel_buffer_object
2022-09-03 14:11:22.066 T:144225    INFO <general>: Using GL_TEXTURE_2D
2022-09-03 14:11:22.078 T:144633    INFO <general>: CDVDVideoCodecFFmpeg::CDropControl: calculated diff time: 40000
2022-09-03 14:11:22.080 T:144629    INFO <general>: Python interpreter stopped
2022-09-03 14:11:22.162 T:144239    INFO <general>: PulseAudio: Opened device Default in pcm mode with Buffersize 150 ms Periodsize 50 ms
2022-09-03 14:11:22.170 T:144225 WARNING <general>: CLinuxRendererGL::UpdateVideoFilter - chosen scaling method 8, is not supported by renderer
2022-09-03 14:11:22.170 T:144225    INFO <general>: GL: Selecting YUV 2 RGB shader
2022-09-03 14:11:22.170 T:144225    INFO <general>: GL: Using GL_ARB_pixel_buffer_object
```

With these lines repeating every few seconds, when I see a new frame:

```
2022-09-03 14:15:32.727 T:144634    INFO <general>: Skipped 113 duplicate messages..
2022-09-03 14:15:32.727 T:144634    INFO <general>: CVideoPlayerAudio::Process - stream stalled
2022-09-03 14:15:32.747 T:144633 WARNING <general>: OutputPicture - timeout waiting for buffer
2022-09-03 14:15:32.816 T:146927    INFO <general>: COutput::OnStartup: Output Thread created
2022-09-03 14:15:32.816 T:146927    INFO <general>: VAAPI::SupportsFilter vaDeriveImage not supported by driver - ffmpeg postprocessing and CPU-copy rendering will not be available
2022-09-03 14:15:32.816 T:146927   ERROR <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
2022-09-03 14:15:32.882 T:144225    INFO <general>: VAAPI::Close
2022-09-03 14:15:32.882 T:146852    INFO <general>: COutput::OnExit: Output Thread terminated
2022-09-03 14:15:33.398 T:144633 WARNING <general>: OutputPicture - timeout waiting for buffer
```

philipl, 2022-09-03:
The key line is
```
2022-09-03 14:15:32.816 T:146927   ERROR <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
```

Kodi is using the VPP (Video Post Processor) to do the colourspace conversion to RGB for final display. This is an optional VAAPI feature that nvidia-vaapi-driver doesn't implement - and couldn't do without incorporating an entire rendering pipeline into itself (like all of libplacebo). nvidia don't provide access to the hardware that does this work.

Ideally the media player would have a GPU (GL/Vulkan) based pipeline for this, and surely Kodi has one but it's possible that they have assumed any vaapi driver will provide vpp. No idea how much work is required on the Kodi side to lift that requirement.

elFarto, 2022-09-03:
I'm not sure that's the case. VAProfile 12 is JPEG. And the two lines about vaDeriveImage and vaCreateConfig for VPP are only meant to happen on init. So it seems something is causing Kodi to destroy and recreate the decoder continuously. This normally happens if one of the threads crashes, but that doesn't seem to be happening here (both threads show up again after the long pause).

I'll keep digging.

elFarto, 2022-09-03:
I think it might be a Kodi bug. I turned on frame timing debugging and noticed this difference:
With VA-API
```
2022-09-03 18:53:33.805 T:61545   DEBUG <general>: PrepareNextRender: frameOnScreen: 5662565.000000 renderPts: -1065101659.645470 nextFramePts: 33333.000000 -> diff: -1065134992.645470  render: 0 forceNext: 0
2022-09-03 18:53:33.823 T:61545   DEBUG <general>: PrepareNextRender: frameOnScreen: 5679924.623000 renderPts: -1065101675.022470 nextFramePts: 33333.000000 -> diff: -1065135008.022470  render: 0 forceNext: 0
2022-09-03 18:53:33.839 T:61545   DEBUG <general>: PrepareNextRender: frameOnScreen: 5696613.455000 renderPts: -1065101736.190470 nextFramePts: 33333.000000 -> diff: -1065135069.190470  render: 0 forceNext: 0
2022-09-03 18:53:33.855 T:61545   DEBUG <general>: PrepareNextRender: frameOnScreen: 5712586.096000 renderPts: -1065101638.549470 nextFramePts: 33333.000000 -> diff: -1065134971.549470  render: 0 forceNext: 0
```
Without VA-API:
```
2022-09-03 18:54:51.364 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5554119.212000 renderPts: 5642849.660158 nextFramePts: 5666666.000000 -> diff: -23816.339842  render: 0 forceNext: 0
2022-09-03 18:54:51.380 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5569886.153000 renderPts: 5658085.943369 nextFramePts: 5666666.000000 -> diff: -8580.056631  render: 0 forceNext: 0
2022-09-03 18:54:51.397 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5587489.676000 renderPts: 5675689.466369 nextFramePts: 5700000.000000 -> diff: -24310.533631  render: 0 forceNext: 0
2022-09-03 18:54:51.414 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5604138.548000 renderPts: 5692338.338369 nextFramePts: 5700000.000000 -> diff: -7661.661631  render: 0 forceNext: 0
2022-09-03 18:54:51.430 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5619942.770000 renderPts: 5708142.560369 nextFramePts: 5733333.000000 -> diff: -25190.439631  render: 0 forceNext: 0
2022-09-03 18:54:51.447 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5637556.632000 renderPts: 5725756.422369 nextFramePts: 5733333.000000 -> diff: -7576.577631  render: 0 forceNext: 0
2022-09-03 18:54:51.463 T:61727   DEBUG <general>: PrepareNextRender: frameOnScreen: 5653285.614000 renderPts: 5741485.404369 nextFramePts: 5766666.000000 -> diff: -25180.595631  render: 0 forceNext: 0
```
So it looks like Kodi isn't bothering to decode new frames because it doesn't think it needs them yet. It's possible it's a bug with this driver, but Kodi is using VA-API through ffmpeg, same as Firefox so that seems unlikely.

cgutman, 2022-09-03:
> VAProfile 12 is JPEG.

FWIW, I'm pretty sure that's a `VAStatus`, not a `VAProfile` value. 12 is `VA_STATUS_ERROR_UNSUPPORTED_PROFILE` which matches the debug message.

philipl, 2022-09-03:
Fair enough on VPP. In the initial logs it's not treated as a fatal error and we see it load the colourspace conversion shader. But in the repeated logs, it does close immediately after VPP fails to load, FWIW.

elFarto, 2022-09-04:
@cgutman I think you might be right actually, that would make more sense.

@philipl I'm less sure about what's going on here. All I do know is that it was working with Kodi, and even with all the errors playing videos now, I'm occasionally getting a valid frame displayed which wouldn't happen unless Kodi was doing the YUV -> RGB conversion. In addition, Kodi is exporting the original frame via DMA-BUF, which it wouldn't do if it was planning to use VPP.

philipl, 2022-09-04:
Very strange, but I agree, the VPP thing is a red herring.

elFarto, 2023-01-07:
I've just retested this with Kodi 19.4 and it seems to be working mostly fine. It didn't seem to want to use it for one of my interlaced test files, but everything else was good. Is this still an issue?

erkrali, 2023-01-21:
I'm getting this issue with Kodi 20.0 (also using nvidia-vaapi-driver with firefox without any problem so far):

```
2023-01-21 16:09:23.588 T:31804   error <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
2023-01-21 16:09:23.693 T:27917    info <general>: VAAPI::Close
2023-01-21 16:09:23.693 T:31171    info <general>: COutput::OnExit: Output Thread terminated
2023-01-21 16:09:24.210 T:28022 warning <general>: OutputPicture - timeout waiting for buffer
2023-01-21 16:09:33.582 T:28023    info <general>: Skipped 101 duplicate messages..
2023-01-21 16:09:33.582 T:28023    info <general>: CVideoPlayerAudio::Process - stream stalled
2023-01-21 16:09:33.593 T:28022 warning <general>: OutputPicture - timeout waiting for buffer
2023-01-21 16:09:33.685 T:32606    info <general>: COutput::OnStartup: Output Thread created
2023-01-21 16:09:33.685 T:32606    info <general>: VAAPI::SupportsFilter vaDeriveImage not supported by driver - ffmpeg postprocessing and CPU-copy rendering will not be available
2023-01-21 16:09:33.685 T:32606   error <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
2023-01-21 16:09:33.793 T:27917    info <general>: VAAPI::Close
2023-01-21 16:09:33.793 T:31804    info <general>: COutput::OnExit: Output Thread terminated
2023-01-21 16:09:34.310 T:28022 warning <general>: OutputPicture - timeout waiting for buffer
2023-01-21 16:09:44.254 T:28023    info <general>: Skipped 102 duplicate messages..
2023-01-21 16:09:44.254 T:28023    info <general>: CVideoPlayerAudio::Process - stream stalled
2023-01-21 16:09:44.260 T:28022 warning <general>: OutputPicture - timeout waiting for buffer
2023-01-21 16:09:44.271 T:1047     info <general>: COutput::OnStartup: Output Thread created
2023-01-21 16:09:44.271 T:1047     info <general>: VAAPI::SupportsFilter vaDeriveImage not supported by driver - ffmpeg postprocessing and CPU-copy rendering will not be available
2023-01-21 16:09:44.271 T:1047    error <general>: VAAPI/vpp vaCreateConfig error: the requested VAProfile is not supported (12)
2023-01-21 16:09:44.376 T:27917    info <general>: VAAPI::Close
2023-01-21 16:09:44.376 T:32606    info <general>: COutput::OnExit: Output Thread terminated
2023-01-21 16:09:44.893 T:28022 warning <general>: OutputPicture - timeout waiting for buffer
```

The workaround of forcibly disabling VA-API by choosing a different profile also works.

morfeus77, 2023-01-28:
> I'm getting this issue with Kodi 20.0 (also using nvidia-vaapi-driver with firefox without any problem so far):
>
> ```
> 2023-01-21 16:09:23.588 T:31804   error <general>: VAAPI/vpp vaCreateConfig [...]

I had a similar issue (working with firefox but not playing along with Kodi).  I fixed this by installing [nv-codec-headers](https://git.videolan.org/?p=ffmpeg/nv-codec-headers.git)

erkrali, 2023-01-28:
> I had a similar issue (working with firefox but not playing along with Kodi). I fixed this by installing [nv-codec-headers](https://git.videolan.org/?p=ffmpeg/nv-codec-headers.git). You might want to double check if you have those installed. I also recompiled ffmpeg to make use of these headers (though I'm not sure that's a necessary step). I basically followed this guide https://www.cyberciti.biz/faq/how-to-install-ffmpeg-with-nvidia-gpu-acceleration-on-linux/ since I'm on Linux Mint.

Thanks for the tip!

Given that these are just headers, I'd expect that you use them to compile something with them to get any effect.

Also, I'm looking into that now, but how did you tell kodi to use your version of ffmpeg? AIUI kodi uses its own internal ffmpeg library, so did you recompile it as well?

morfeus77, 2023-01-28:
Please ignore my comment seems all that did was disable the vdpau driver.

closed 2023-06-16 by elFarto