elFarto/nvidia-vaapi-driver #110: Content process occasionally crashes

remyabel2, 2022-09-05:
Occasionally, I get a crash report that says `firefox killed by SIGSEGV` for the following commandline:

```
/usr/lib64/firefox/firefox -contentproc -parentBuildID 20220830142553 -prefsLen 39613 -prefMapSize 243035 -appDir /usr/lib64/firefox/browser 30 true rdd
```
I've attached a GDB backtrace, but wasn't sure what information to include.

```
#0  0x00007f98ed533fec in __run_exit_handlers
    (status=1, listp=0x7f98ed6eb838 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true) at exit.c:71
#1  0x00007f98ed534200 in __GI_exit (status=<optimized out>) at exit.c:143
#2  0x00007f98cde86c44 in checkCudaErrors
    (err=2, file=0x7f98cde8f3cc "../src/vabackend.c", function=0x7f98cde8fe00 <__func__.14> "nvQuerySurfaceAttributes", line=1508)
    at ../src/vabackend.c:138
#3  0x00007f98cde8a180 in nvQuerySurfaceAttributes (ctx=0x7f98aecd0100, config=20, attrib_list=0x7f97eac61900, num_attribs=0x7f98ae4fe084)
    at ../src/vabackend.c:1508
#4  0x00007f98d2ecd97a in vaQuerySurfaceAttributes (dpy=0x7f98c0c0e760, config=20, attrib_list=0x7f97eac61900, num_attribs=0x7f98ae4fe084)
    at /usr/src/debug/libva-2.14.0-1.fc36.x86_64/va/va.c:1152
#5  0x00007f98d2a8278a in vaapi_decode_find_best_format (device=<optimized out>, frames=0x7f97eac61c00, config_id=20, avctx=0x7f988134e100)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/vaapi_decode.c:305
#6  vaapi_decode_make_config
    (avctx=avctx@entry=0x7f988134e100, device_ref=<optimized out>, va_config=va_config@entry=0x7f97eac61580, frames_ref=0x7f97eae7e2c0)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/vaapi_decode.c:570
#7  0x00007f98d2a83135 in ff_vaapi_decode_init (avctx=0x7f988134e100)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/vaapi_decode.c:692
#8  0x00007f98d2a634bf in hwaccel_init (hw_config=0x7f98d2d84600 <__compound_literal.0>, avctx=0x7f988134e100)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/decode.c:1300
#9  ff_get_format (avctx=0x7f988134e100, fmt=<optimized out>) at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/decode.c:1440
#10 0x00007f98d2a99ebf in update_size (h=713, w=956, avctx=0x7f988134e100)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/vp9.c:244
#11 decode_frame_header
    (avctx=avctx@entry=0x7f988134e100, data=data@entry=0x7f9891b1c000 "\202I\203B", size=size@entry=43247, ref=ref@entry=0x7f98ae4fe324)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/vp9.c:770
#12 0x00007f98d2a9c0ee in vp9_decode_frame (avctx=0x7f988134e100, frame=0x7f987bc4fc00, got_frame=0x7f98ae4fe390, pkt=0x7f987ba70980)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/vp9.c:1565
#13 0x00007f98d2a61cae in decode_simple_internal (discarded_samples=<synthetic pointer>, frame=<optimized out>, avctx=0x7f988134e100)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/decode.c:329
#14 decode_simple_receive_frame (frame=<optimized out>, avctx=<optimized out>)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/decode.c:530
#15 decode_receive_frame_internal (avctx=avctx@entry=0x7f988134e100, frame=0x7f987bc4fc00)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/decode.c:550
#16 0x00007f98d2a62604 in avcodec_send_packet (avctx=0x7f988134e100, avpkt=0x7f98ae4fe500)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/media/ffvpx/libavcodec/decode.c:617
#17 0x00007f98e250b8da in mozilla::FFmpegVideoDecoder<46465650>::DoDecode(mozilla::MediaRawData*, unsigned char*, int, bool*, nsTArray<RefPtr<mozilla::MediaData> >&) (this=0x7f98aee67400, aSample=0x7f98865e7380, aData=0x7f987bc69000 "\202I\203B", aSize=43247, aGotFrame=0x7f98ae4fe787, aResults=...)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/dom/media/platforms/ffmpeg/FFmpegVideoDecoder.cpp:834
#18 0x00007f98e250713f in mozilla::FFmpegDataDecoder<46465650>::DoDecode(mozilla::MediaRawData*, bool*, nsTArray<RefPtr<mozilla::MediaData> >&)
    (this=0x7f98aee67400, aSample=aSample@entry=0x7f98865e7380, aGotFrame=aGotFrame@entry=0x7f98ae4fe787, aResults=...)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/dom/media/platforms/ffmpeg/FFmpegDataDecoder.cpp:193
#19 0x00007f98e250b062 in mozilla::FFmpegDataDecoder<46465650>::ProcessDecode(mozilla::MediaRawData*) (this=<optimized out>, aSample=0x7f98865e7380)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/dom/media/platforms/ffmpeg/FFmpegDataDecoder.cpp:147
#20 0x00007f98e403703d in mozilla::detail::RunnableMethodArguments<mozilla::MediaRawData*>::applyImpl<mozilla::FFmpegDataDecoder<46465650>, RefPtr<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true> > (mozilla::FFmpegDataDecoder<46465650>::*)(mozilla::MediaRawData*), StoreRefPtrPassByPtr<mozilla::MediaRawData>, 0ul>(mozilla::FFmpegDataDecoder<46465650>*, RefPtr<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true> > (mozilla::FFmpegDataDecoder<46465650>::*)(mozilla::MediaRawData*), mozilla::Tuple<StoreRefPtrPassByPtr<mozilla::MediaRawData> >&, std::integer_sequence<unsigned long, 0ul>) (args=<optimized out>, m=<optimized out>, o=<optimized out>)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/objdir/dist/include/nsThreadUtils.h:1147
#21 mozilla::detail::RunnableMethodArguments<mozilla::MediaRawData*>::apply<mozilla::FFmpegDataDecoder<46465650>, RefPtr<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true> > (mozilla::FFmpegDataDecoder<46465650>::*)(mozilla::MediaRawData*)>(mozilla::FFmpegDataDecoder<46465650>*, RefPtr<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true> > (mozilla::FFmpegDataDecoder<46465650>::*)(mozilla::MediaRawData*)) (m=<optimized out>, o=<optimized out>, this=<optimized out>)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/objdir/dist/include/nsThreadUtils.h:1153
#22 mozilla::detail::MethodCall<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true>, RefPtr<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true> > (mozilla::FFmpegDataDecoder<46465650>::*)(mozilla::MediaRawData*), mozilla::FFmpegDataDecoder<46465650>, mozilla::MediaRawData*>::Invoke() (this=<optimized out>)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/objdir/dist/include/mozilla/MozPromise.h:1518
#23 mozilla::detail::ProxyRunnable<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true>, RefPtr<mozilla::MozPromise<nsTArray<RefPtr<mozilla::MediaData> >, mozilla::MediaResult, true> > (mozilla::FFmpegDataDecoder<46465650>::*)(mozilla::MediaRawData*), mozilla::FFmpegDataDecoder<46465650>, mozilla::MediaRawData*>::Run() (this=0x7f9811dd0fd0)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/objdir/dist/include/mozilla/MozPromise.h:1538
#24 0x00007f98e38722c7 in mozilla::TaskQueue::Runner::Run() (this=0x7f9811eaee60)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/xpcom/threads/TaskQueue.cpp:259
#25 0x00007f98e3871faf in nsThreadPool::Run() (this=0x7f98aecd1130) at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/xpcom/threads/nsThreadPool.cpp:310
#26 0x00007f98e3779c7c in nsThread::ProcessNextEvent(bool, bool*) (this=0x7f98c18f3100, aMayWait=<optimized out>, aResult=0x7f98ae4fea57)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/objdir/dist/include/nsCOMPtr.h:851
#27 0x00007f98e3779a13 in NS_ProcessNextEvent(nsIThread*, bool) (aThread=<optimized out>, aMayWait=<optimized out>)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/xpcom/threads/nsThreadUtils.cpp:465
#28 0x00007f98e3a4cfee in mozilla::ipc::MessagePumpForNonMainThreads::Run(base::MessagePump::Delegate*)
    (this=0x7f98ada879c0, aDelegate=0x7f98ae4feb00) at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/ipc/glue/MessagePump.cpp:300
#29 0x00007f98e3a26eeb in MessageLoop::RunInternal() (this=<optimized out>)
    at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/objdir/dist/include/mozilla/RefPtr.h:280
#30 MessageLoop::RunHandler() (this=<optimized out>) at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/ipc/chromium/src/base/message_loop.cc:373
#31 MessageLoop::Run() (this=<optimized out>) at /usr/src/debug/firefox-104.0.1-1.fc36.x86_64/ipc/chromium/src/base/message_loop.cc:355
#32 0x00007f98e3871898 in nsThread::ThreadFunc(void*) (aArg=0x7f98ada4ab60)
[... cut ...]
#2  0x00007f98cde86c44 in checkCudaErrors (err=2, file=0x7f98cde8f3cc "../src/vabackend.c", 
    function=0x7f98cde8fe00 <__func__.14> "nvQuerySurfaceAttributes", line=1508) at ../src/vabackend.c:138
138	        exit(EXIT_FAILURE);
errStr = 0x7f98cca8dd5d "out of memory"
``

However, I don't think I ran out of video memory or RAM. It did not kill the entire browser, only the content process and the tab still seemed to play video fine. I think it's killed the entire browser before, but I didn't log the error when that happened.

elFarto, 2022-09-10:
What GPU do you have? And what site(s) are you viewing when this happens? We've seen this issue (the "out of memory" one) before on sites that contain lots of videos. You can trying using the `NVD_MAX_INSTANCES` environment variable to limit how many concurrent instances the driver will allow, which can lower the amount of memory that's used.

The driver is forcing the app to exit when we hit a CUDA error as that should never happen in the normal case, so that part is expected.

elFarto, 2022-12-19:
The latest version has revised the error handling, so we will never exit the process on a CUDA error now.

closed 2022-12-19 by elFarto