elFarto/nvidia-vaapi-driver #111: chrome-support: meson build install issue

pc00per, 2022-09-10:
Not sure how to debug this. Can anyone help this out?
Linux: Arch
Driver package: nvidia

```
$[/tmp/nvidia-vaapi-driver]$ meson install -C build
ninja: Entering directory `/tmp/nvidia-vaapi-driver/build'
[1/17] Compiling C object nvidia_drv_video.so.p/src_export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_export-buf.c.o -MF nvidia_drv_video.so.p/src_export-buf.c.o.d -o nvidia_drv_video.so.p/src_export-buf.c.o -c ../src/export-buf.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/export-buf.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[2/17] Compiling C object nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o -MF nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o.d -o nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o -c ../src/direct/direct-export-buf.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/../direct/nv-driver.h:7,
                 from ../src/direct/../vabackend.h:4,
                 from ../src/direct/direct-export-buf.c:3:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[3/17] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
FAILED: nvidia_drv_video.so.p/src_h264.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_h264.c.o -MF nvidia_drv_video.so.p/src_h264.c.o.d -o nvidia_drv_video.so.p/src_h264.c.o -c ../src/h264.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/h264.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[4/17] Compiling C object nvidia_drv_video.so.p/src_av1.c.o
FAILED: nvidia_drv_video.so.p/src_av1.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_av1.c.o -MF nvidia_drv_video.so.p/src_av1.c.o.d -o nvidia_drv_video.so.p/src_av1.c.o -c ../src/av1.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/av1.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[5/17] Compiling C object nvidia_drv_video.so.p/src_hevc.c.o
FAILED: nvidia_drv_video.so.p/src_hevc.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_hevc.c.o -MF nvidia_drv_video.so.p/src_hevc.c.o.d -o nvidia_drv_video.so.p/src_hevc.c.o -c ../src/hevc.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/hevc.c:3:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[6/17] Compiling C object nvidia_drv_video.so.p/src_jpeg.c.o
FAILED: nvidia_drv_video.so.p/src_jpeg.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_jpeg.c.o -MF nvidia_drv_video.so.p/src_jpeg.c.o.d -o nvidia_drv_video.so.p/src_jpeg.c.o -c ../src/jpeg.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/jpeg.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[7/17] Compiling C object nvidia_drv_video.so.p/src_direct_nv-driver.c.o
FAILED: nvidia_drv_video.so.p/src_direct_nv-driver.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_nv-driver.c.o -MF nvidia_drv_video.so.p/src_direct_nv-driver.c.o.d -o nvidia_drv_video.so.p/src_direct_nv-driver.c.o -c ../src/direct/nv-driver.c
../src/direct/nv-driver.c:11:10: fatal error: drm/drm_fourcc.h: No such file or directory
   11 | #include <drm/drm_fourcc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[8/17] Compiling C object nvidia_drv_video.so.p/src_mpeg2.c.o
FAILED: nvidia_drv_video.so.p/src_mpeg2.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_mpeg2.c.o -MF nvidia_drv_video.so.p/src_mpeg2.c.o.d -o nvidia_drv_video.so.p/src_mpeg2.c.o -c ../src/mpeg2.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:4,
                 from ../src/mpeg2.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Could not rebuild /tmp/nvidia-vaapi-driver/build
$[/tmp/nvidia-vaapi-driver]$
```

elFarto, 2022-09-10:
Either change the `drm/` bit to `libdrm/` or installl the kernel-headers to get the DRM header files. The new chrome code doesn't have the extra logic to switch between them at the moment.

pc00per, 2022-09-10:
> Either change the `drm/` bit to `libdrm/` or installl the kernel-headers to get the DRM header files. The new chrome code doesn't have the extra logic to switch between them at the moment.

Okay tq. Fixed it with `ln -s /usr/include/libdrm /usr/include/drm`

So, is it done? That's it? Or anything needed for manual build of chrome-support branch?

elFarto, 2022-09-10:
Currently Chome support requires `NVD_BACKEND=direct` to be set, and that option is very experimental.

pc00per, 2022-09-10:
> Currently Chome support requires `NVD_BACKEND=direct` to be set, and that option is very experimental.

Tq. But uhh, you're right. Stuttering exist.

closed 2022-09-10 by pc00per

elFarto, 2023-06-17:
There is/was a chrome-support branch that implemented a different method to get it working with Chrome. However that method was an absurd amount of work, and the code path it made use of has been removed from Chrome now so it will no longer work.