elFarto/nvidia-vaapi-driver #112: VA-API works with everything except native firefox

wael444, 2022-09-17:
There is no VA-API/GPU logs when firefox is ran:
`NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia /usr/lib/firefox/firefox`
Using Flatpak firefox recognizes VA-API but isn't usable, i am currently figuring out why Firefox ignores the va-api driver.
```
[Child 323: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type audio/mp4a-latm
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type audio/mp4a-latm
[RDD 354: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/av1'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/av1'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type 'video/av1'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/mp4a-latm'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/vorbis'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/vorbis'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/x-wav'
[RDD 354: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/x-wav'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/x-wav'
[RDD 354: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/x-wav'
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox Utility decoder supports requested type audio/mp4a-latm
[Child 323: Main Thread]: D/PlatformDecoderModule Sandbox Utility decoder supports requested type audio/mp4a-latm
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Utility process
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/av1'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type 'video/av1'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule Agnostic decoder rejects requested type 'video/theora'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[Utility AudioDecoder 369: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[... 70 lines cut ...]

firefox build flags:
```
ac_add_options --prefix=/usr
ac_add_options --libdir=/usr/lib
ac_add_options --with-ccache=sccache
ac_add_options --enable-strip
ac_add_options --enable-release
ac_add_options --enable-rust-simd
ac_add_options --enable-audio-backends=alsa
ac_add_options --enable-audio-backends=pulseaudio
ac_add_options --enable-official-branding
ac_add_options --enable-application=browser
ac_add_options --enable-optimize="${CFLAGS:-} -w"
ac_add_options --with-system-nss
ac_add_options --with-system-nspr
ac_add_options --without-wasm-sandboxed-libraries
ac_add_options --disable-dbus
ac_add_options --enable-linker="$ld"
ac_add_options --disable-tests
ac_add_options --disable-updater
ac_add_options --disable-necko-wifi
ac_add_options --disable-debug-symbols
ac_add_options --disable-parental-controls
```

elFarto, 2022-09-17:
Can you run firefox with this set `MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5"`, and paste the output here?

wael444, 2022-09-17:
`NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5" /us
r/lib/firefox/firefox 2>&1 | tee firefox.log`
https://termbin.com/xm66
I am also on nvidia drivers 515.65.01

elFarto, 2022-09-17:
The first few lines should have been something like:
```
[Parent 11381: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 11381: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 11381: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 11381: Main Thread]: D/Dmabuf DMABuf is enabled
```
Which leads me to believe that the settings in about:config haven't been set correctly.

wael444, 2022-09-17:
```
widget.dmabuf.force-enabled            null
gfx.x11-egl.force-enabled              true 
media.av1.enabled                      false 
media.rdd-ffmpeg.enabled               false
media.ffmpeg.vaapi.enabled             true
gfx.webrender.all                      true
gfx.webrender.compositor.force-enabled true
gfx.webrender.enabled                  true 
```
`widget.dmabuf.force-enabled`  does not exist.
```
$ /bin/ls -la /dev/dri/renderD128
crw-rw----    1 root     video     226, 128 Sep 17 14:40 /dev/dri/renderD128
$ groups
wael tty audio video input
```

elFarto, 2022-09-17:
`media.rdd-ffmpeg.enabled` should be `true`

wael444, 2022-09-17:
it is already true, just made a mistake while formatting the list

i use a custom kernel, so i wonder if the dmabuf thing is a problem with my kernel configuration, that or that i do not have GBM libraries, where are they located so i can check?

elFarto, 2022-09-17:
Can you have a look in `about:support` in Firefox,  and see what it says under DMABUF and HARDWARE_VIDEO_DECODING.

wael444, 2022-09-17:
[image]
so,, i need to build firefox with wayland support now..?

elFarto, 2022-09-17:
Ah, yes, Wayland support is needed for VA-API decoding to work. I'm not sure why Firefox has those options are linked like that, VA-API works fine on X11.

wael444, 2022-09-17:
i will do that and see how it goes.

wael444, 2022-09-17:
it seems to work!
```
|    0   N/A  N/A      8185      C   /usr/lib/firefox/firefox           79MiB |
```
however Video Engine Utilization is 0% for some reason.
but it does work i believe...
cpu usage is still quite the same.
NVD_LOG=1 works and does output logs

wael444, 2022-09-17:
https://termbin.com/5l50

elFarto, 2022-09-17:
It's not working:
```
[RDD 10853: MediaSupervisor #1]: D/PlatformDecoderModule FFVPX: Hardware WebRender is off, VAAPI is disabled
```
Take a look in `about:support` again.

wael444, 2022-09-17:
[image]
```
        --setenv LIBVA_DRIVER_NAME nvidia \
        --setenv VDPAU_DRIVER nvidia \
        --setenv MOZ_DISABLE_RDD_SANDBOX 1 \
        --setenv NVD_LOG 1 \
```

elFarto, 2022-09-17:
Firefox seems to think that you don't have hardware compositing enabled, but those values seem to suggest it is enabled. What does the 'Compositing' value say, under the 'Features' section?

wael444, 2022-09-17:
[image]

elFarto, 2022-09-17:
Not sure what's going on here. You're output looks the same as mine, which is working. But you're hitting [this](https://searchfox.org/mozilla-central/source/dom/media/platforms/ffmpeg/FFmpegVideoDecoder.cpp#372) line of code, which indicates something is not working as expected.

wael444, 2022-09-17:
well idk tbh
```
ac_add_options --prefix=/usr
ac_add_options --libdir=/usr/lib
ac_add_options --enable-default-toolkit=cairo-gtk3-x11-wayland
ac_add_options --with-ccache=sccache
ac_add_options --enable-strip
ac_add_options --enable-release
ac_add_options --enable-rust-simd
ac_add_options --enable-audio-backends=alsa
ac_add_options --enable-audio-backends=pulseaudio
ac_add_options --enable-install-strip
ac_add_options --enable-official-branding
ac_add_options --enable-application=browser
ac_add_options --enable-optimize="${CFLAGS:-} -w"
ac_add_options --with-system-libvpx
ac_add_options --with-system-ffi
ac_add_options --with-system-png
ac_add_options --with-system-jpeg
ac_add_options --with-system-zlib
ac_add_options --with-system-pixman
ac_add_options --without-system-nss
ac_add_options --without-system-nspr
ac_add_options --without-wasm-sandboxed-libraries
ac_add_options --disable-dbus
ac_add_options --enable-linker="$ld"
ac_add_options --disable-tests
ac_add_options --disable-vtune
ac_add_options --disable-updater
ac_add_options --disable-jemalloc
ac_add_options --disable-elf-hack
ac_add_options --disable-callgrind
ac_add_options --disable-profiling
ac_add_options --disable-necko-wifi
ac_add_options --disable-crashreporter
ac_add_options --disable-accessibility
ac_add_options --disable-debug
ac_add_options --disable-debug-symbols
ac_add_options --disable-parental-controls
ac_add_options --disable-system-extension-dirs
```
```
Name 	        Firefox
Version 	105.0b9
Build ID 	20220917182220
```

wael444, 2022-09-20:
`
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: glxtest: VA-API test failed: process crashed. Please check your VA-API drivers. (t=0.211121) [GFX1-]: glxtest: VA-API test failed: process crashed. Please check your VA-API drivers.
`
[image]
```
Name 	        Firefox
Version 	105.0
Build ID 	20220920150712
```

elFarto, 2022-09-20:
Can you check if `media.ffmpeg.enabled` is set to `true` in about:config. If it is, can you try setting `media.ffvpx.enabled` to `false`. There's something odd in that last log output I didn't notice at first. It's not checking if the VA-API library supports VP9, only VP8, and your 2060 can do VP9.
```
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: FFmpegVideoDecoder::GetAcceleratedFormats()
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Profile H264Main:
      6470.306065161 [10853-10896] ../src/vabackend.c: 557           nvCreateConfig got profile: 6 with 0 attributes
      6470.306073781 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 4 (nil) 0
      6470.306076670 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 4 0x7ff8ec718000 5
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:     codec h264 format nv12              3         12
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Profile H264High:
      6470.314923853 [10853-10896] ../src/vabackend.c: 557           nvCreateConfig got profile: 7 with 0 attributes
      6470.314939466 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 4 (nil) 0
      6470.314942579 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 4 0x7ff8ec622b40 5
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:     codec h264 format nv12              3         12
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Profile H264ConstrainedBaseline:
      6470.324045537 [10853-10896] ../src/vabackend.c: 557           nvCreateConfig got profile: 13 with 0 attributes
      6470.324051607 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 4 (nil) 0
      6470.324054560 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 4 0x7ff8ec622bc0 5
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:     codec h264 format nv12              3         12
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Profile VP8Version0_3:
      6470.333026186 [10853-10896] ../src/vabackend.c: 557           nvCreateConfig got profile: 14 with 0 attributes
      6470.333032377 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 9 (nil) 0
      6470.333035337 [10853-10896] ../src/vabackend.c:1498 nvQuerySurfaceAttributes with 9 0x7ff8ec71a6c0 5
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:     codec vp8 format nv12              3         12
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Supported accelerated formats:
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:       h264
[RDD 10853: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:       vp8
```

wael444, 2022-09-20:
> Can you check if media.ffmpeg.enabled is set to true in about:config. If it is, can you try setting media.ffvpx.enabled to false.

no change

wael444, 2022-09-29:
fresh reinstall, unable to test VA-API with mpv:
```
Unsupported hwdec: vaapi
```
libva is installed on my system and mpv does not seem to want to compile with support for vaapi.

mpv: `HEAD is now at b2aaf72 ao_pipewire: don't try to lock nonexistent loop`

wael444, 2022-09-29:
i would also like to point out that firefox:
`MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5" LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 NVD_LOG=1 /usr/lib/firefox/firefox`

```
[Parent 26932: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 26932: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 26932: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 26932: Main Thread]: D/Dmabuf DMABuf is enabled
```
unfortunately in `about:support` HARDWARE_VIDEO_DECODING is still 'block'listed by gfxInfo. i wonder if there is a way to log that and see what causes it to get blacklisted.

maybe a userjs thing?

wael444, 2022-09-29:
ffmpeg had to be recompiled for libva support, mpv the same, works for both.

firefox still says that HARDWARE_VIDEO_DECODING is ''''block''''listed by gfxInfo
maybe recompilation of firefox to detect hte new ffmpeg?

elFarto, 2022-09-30:
I'm not sure Firefox actually uses the system ffmpeg libraries. I believe it has it's own copies that it uses. You might be better served asking in the Firefox support channels (bugzilla or matrix) about Firefox issues, I still don't have a clear understanding of all the steps FF jumps through to decide on using VA-API, and I'm they've been improving/changing these since this library was initially released.

I will add though, that my HARDWARE_VIDEO_DECODING field also says blocked by gfxInfo, but it still works for me.

wael444, 2022-09-30:
> I will add though, that my HARDWARE_VIDEO_DECODING field also says blocked by gfxInfo, but it still works for me.

!!! how can I test if va-api is working then

elFarto, 2022-09-30:
There's a section at the bottom of the README describing how to verify it's working.

wael444, 2022-09-30:
NVD_LOG takes affect and i see C+G in nvidia-smi, does this mean its working?

elFarto, 2022-09-30:
Yes, that is working.

wael444, 2022-09-30:
yayyyy I will keep this closed and try to fix Firefox running under bwrap, I usually run Firefox sandboxed and native VA-API works. however sandboxed va-api test 'fails'. I have given it access to `/usr/lib/DRI`; is there anything more special I need?

closed 2022-09-30 by wael444

elFarto, 2022-09-30:
I'm not sure about the sandbox stuff, I've never run anything under it. But it'll likely need access to all the NVIDIA driver and CUDA libraries for a start.

wael444, 2022-10-01:
```
execve("/usr/bin/vainfo", ["vainfo"], 0x7ffce44b0e60 /* 45 vars */) = 0
brk(NULL)                               = 0x55d1b2d87000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f7760437000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/glibc-hwcaps/x86-64-v3/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/glibc-hwcaps/x86-64-v3", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/glibc-hwcaps/x86-64-v2/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/glibc-hwcaps/x86-64-v2", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/haswell/x86_64/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/tls/haswell/x86_64", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/haswell/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/tls/haswell", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/x86_64/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/tls/x86_64", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/tls", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/haswell/x86_64/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/haswell/x86_64", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/haswell/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/haswell", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64/libva.so.2", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/usr/lib/x86_64", 0x7ffd7497ffb0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libva.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=191008, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 195200, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f7760407000
mmap(0x7f776040a000, 106496, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f776040a000
mmap(0x7f7760424000, 69632, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1d000) = 0x7f7760424000
mmap(0x7f7760435000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2d000) = 0x7f7760435000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libX11.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=1403576, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 1407464, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f77602af000
mmap(0x7f77602cd000, 647168, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e000) = 0x7f77602cd000
mmap(0x7f776036b000, 610304, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xbc000) = 0x7f776036b000
mmap(0x7f7760400000, 28672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x150000) = 0x7f7760400000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libva-x11.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=27320, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 29480, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f77602a7000
mmap(0x7f77602a9000, 12288, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f77602a9000
mmap(0x7f77602ac000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5000) = 0x7f77602ac000
mmap(0x7f77602ad000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5000) = 0x7f77602ad000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libva-drm.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=14544, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 16608, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f77602a2000
mmap(0x7f77602a3000, 4096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f77602a3000
mmap(0x7f77602a4000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f77602a4000
mmap(0x7f77602a5000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f77602a5000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libdrm.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=92936, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 97552, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f776028a000
[... cut ...]
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f7760041510) = 31
wait4(31, [{WIFEXITED(s) && WEXITSTATUS(s) == 1}], 0, NULL) = 31
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=31, si_uid=1000, si_status=1, si_utime=0, si_stime=0} ---
ioctl(-1, _IOC(_IOC_NONE, 0, 0x2, 0x3000), 0) = -1 EBADF (Bad file descriptor)
ioctl(4, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x29, 0x10), 0x7ffd74980230) = 0
close(4)                                = 0
getpid()                                = 28
exit_group(1)                           = ?
+++ exited with 1 +++

wael444, 2022-10-01:
fixed the strace.
on host vainfo, it looks for /dev/nvidia-uvm, which isnt mounted.