elFarto/nvidia-vaapi-driver #113: mp4parse error appearing in logs

peterlobster, 2022-09-19:
I setup the driver, and everything seems like it's working correctly. I was able to verify the use in `nvidia-settings`. However, I see this when I run it from command line...

(FWIW I'm using Gentoo with @tatsh's ebuild: https://github.com/Tatsh/tatsh-overlay/tree/2ec677d0c5b040a7f3cc00d930a93590316109c8/media-plugins/nvidia-vaapi-driver. This is significant since the driver has been renamed to `nvdec`.)

```bash
$ LIBVA_DRIVER_NAME="nvdec" MOZ_DISABLE_RDD_SANDBOX=1 firefox
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[2022-09-19T01:39:29Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-09-19T01:39:29Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-09-19T01:39:29Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-09-19T01:39:29Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[2022-09-19T01:39:30Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-09-19T01:39:30Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-09-19T01:39:30Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-09-19T01:39:30Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvdec'
libva info: Trying to open /usr/lib64/va/drivers/nvdec_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
Missing chrome or resource URL: resource://gre/modules/UpdateListener.sys.mjs
```

elFarto, 2022-09-20:
That error is being printed by Firefox, not this driver.

ran-dall, 2022-09-20:
I've seen this in the Arch support forums before. It's related to using Pipewire and Wireplummber. https://bugs.archlinux.org/task/74741

elFarto, 2022-10-16:
Closing issue as it's nothing to do with the driver.

closed 2022-10-16 by elFarto