elFarto/nvidia-vaapi-driver #114: Possible Firefox crashing due to nvidia-vaapi

Fxzzi, 2022-09-20:
https://crash-stats.mozilla.org/report/index/b2cca642-3f5e-4696-92f8-8d9910220920

Firefox version listed in there. I am running the latest (0.0.6) version of the driver. Not sure when the error started happening but firefox would crash almost every other time I opened it. I've tried on a new profile and everything (only with vaapi enabled) and it still seems to occur. Should I try the latest git version? Or should the stable release be working completely fine?

elFarto, 2022-09-20:
That's an odd one. This driver isn't in the stack trace for that error, and it seems to be occurring in the Renderer thread, which VA-API isn't run on (it's on the RDD threads), so it's difficult to say if it is this driver causing the issue.

Does it crash if you disable VA-API?

Fxzzi, 2022-09-21:
> Does it crash if you disable VA-API?

I've tried updating to the latest -git release from the AUR and I haven't experienced a crash yet. Only used it for a little bit though so I'll have to do a bit more testing to see if this is solved.

Fxzzi, 2022-09-22:
Okay after further testing I am still experiencing the crash. I will try disabling the vaapi flag and try to see if it crashes again.

Another problem I am having is sometimes vaapi just stops working in everything but firefox (vlc, mpv, etc)

Here is the `mpv <video> -v log`

```
[vo/gpu/x11] X11 opening display: :0
[vo/gpu/x11] Display 0 (DP-0): [0, 0, 1920, 1080] @ 74.972503 FPS
[vo/gpu/x11] Display 1 (DP-2): [1920, 0, 4480, 1440] @ 169.830584 FPS
[vo/gpu/x11] Current display FPS: 74.972503
[vo/gpu/opengl] EGL_VERSION=1.5
[vo/gpu/opengl] EGL_VENDOR=NVIDIA
[vo/gpu/opengl] EGL_CLIENT_APIS=OpenGL_ES OpenGL
[vo/gpu/opengl] Trying to create Desktop OpenGL context.
[vo/gpu/opengl] Choosing visual EGL config 0x28, visual ID 0x2b
[vo/gpu/opengl] GL_VERSION='4.4.0 NVIDIA 515.65.01'
[vo/gpu/opengl] Detected desktop OpenGL 4.4.
[vo/gpu/opengl] GL_VENDOR='NVIDIA Corporation'
[vo/gpu/opengl] GL_RENDERER='NVIDIA GeForce RTX 3070/PCIe/SSE2'
[vo/gpu/opengl] GL_SHADING_LANGUAGE_VERSION='4.40 NVIDIA via Cg compiler'
[vo/gpu] Testing FBO format rgba16f
[vo/gpu] Using FBO format rgba16f.
[vo/gpu] Assuming 60.000000 FPS for display sync.
[vd] Container reported FPS: 23.976024
[vd] Codec list:
[vd]     h264 - H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
[vd]     h264_v4l2m2m (h264) - V4L2 mem2mem H.264 decoder wrapper
[vd]     h264_qsv (h264) - H264 video (Intel Quick Sync Video acceleration)
[vd]     h264_cuvid (h264) - Nvidia CUVID H264 decoder
[vd] Opening decoder h264
[vd] Looking at hwdec h264-nvdec...
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
```

uninstalling the nvidia-vaapi-driver, mpv and vlc begin working again.

elFarto, 2022-09-22:
When it stops responding, can you try running it with `NVD_LOG=1` set?

Fxzzi, 2022-09-24:
> When it stops responding, can you try running it with `NVD_LOG=1` set?

Hi again. I am stupid and the `nvidia.NVreg_PreserveVideoMemoryAllocations=1` had a typo in it. fixing the typo seems to have fixed the mpv and vlc issues. However firefox still crashes when opening it. I'll try with the flag set and seeing if I can catch it.

closed 2022-10-03 by Fxzzi