elFarto/nvidia-vaapi-driver #117: How to find out that it works?

rubyFeedback, 2022-10-01:
I just compiled the latest git source of nvidia-vaapi-driver.

No error was reported.

But how do I know it works?

Trying to use "nvidia-settings" from the commandline does not
show anything. There is also no example for mpv. Could you
perhaps add a bit more documentation here? Ideally also some
way to test that things may work now?

I found this project by a recommendation from reddit, also due
to my nvidia card having problems so I was hoping perhaps
this can help. But I am a bit lost what to do ... Making this
more explicit in the main README would be super-helpful.

elFarto, 2022-10-01:
There's a section at the bottom of the README that details how to verify if it's working.

pc00per, 2022-10-12:
Tbh. You should open a discussion thread on this repo. QnA shouldn't be asked in the issues.

closed 2022-10-16 by elFarto