elFarto/nvidia-vaapi-driver #122: Might be neccessary to disable media.rdd-ffvpx.enabled in Firefox 106(?)

philipl, 2022-10-21:
This isn't a problem with this driver, but it seems relevant to us.

I noticed that while I could play back other formats, I wasn't getting accelerated av1 so I took a look.

```
[RDD 15307: MediaSupervisor #1]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 15307: MediaSupervisor #1]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 15307: MediaSupervisor #1]: D/PlatformDecoderModule FFVPX: FFmpegVideoDecoder::FFmpegVideoDecoder MIME video/av1 Codec ID 226
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: Initialising VA-API FFmpeg decoder
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   codec av1 : Alliance for Open Media AV1
[RDD 15307: MediaPDecoder #1]: D/Dmabuf Using DRM device /dev/dri/renderD128
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      1882.908233921 [15307-15543] ../src/vabackend.c:1856       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f06c8523c40 31
      1882.908239626 [15307-15543] ../src/vabackend.c:1877       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1882.910883063 [15307-15543] ../src/export-buf.c: 159       findGPUIndexFromFd Looking for DRM device index: 0
      1882.911825332 [15307-15543] ../src/export-buf.c: 172       findGPUIndexFromFd Found 4 EGL devices
      1882.911920084 [15307-15543] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      1882.911922754 [15307-15543] ../src/export-buf.c: 187       findGPUIndexFromFd Found drmDeviceIndex: 0
      1882.911925815 [15307-15543] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
      1882.962963960 [15307-15543] ../src/export-buf.c: 267             initExporter Driver supports 16-bit surfaces
      1882.962965434 [15307-15543] ../src/export-buf.c:  99                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
      1882.963041921 [15307-15543] ../src/vabackend.c:1191      nvQueryImageFormats In nvQueryImageFormats
[AVHWDeviceContext @ 0x7f06bcc836c0] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x7f06bcc836c0] Format 0x30313050 -> p010le.
[AVHWDeviceContext @ 0x7f06bcc836c0] Format 0x32313050 -> unknown.
[AVHWDeviceContext @ 0x7f06bcc836c0] VAAPI driver: VA-API NVDEC driver.
[AVHWDeviceContext @ 0x7f06bcc836c0] Driver not found in known nonstandard list, using standard behaviour.
[av1 @ 0x7f06c855c800] Error parsing decoder bitstream filters 'av1_frame_split': Bitstream filter not found
      1882.963074442 [15307-15543] ../src/vabackend.c:1833              nvTerminate Terminating 0x7f06c8523c40
      1882.963076556 [15307-15543] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
      1882.964134289 [15307-15543] ../src/export-buf.c:  82          releaseExporter Done releasing frames
      1882.984254677 [15307-15543] ../src/vabackend.c:1847              nvTerminate Now have 0 (0 max) instances
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Couldn't initialise VA-API decoder
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: Initialising FFmpeg decoder
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   codec libdav1d : dav1d AV1 decoder by VideoLAN
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: Set get_buffer2 for customized buffer allocation
[libdav1d @ 0x7f06b01c7c00] libdav1d cd5e415270285a58f48c1e9ec1a2dd024b9acf9f
[libdav1d @ 0x7f06b01c7c00] Using 8 threads, 0 max_frame_delay
[RDD 15307: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   FFmpeg decoder init successful.
```

What we see is that the `av1dec` decoder that the hwaccel plugs into fails to initialise, so it falls back to libdav1d software decoding.

But why is it using ffvpx? I have rdd-ffmpeg enabled, but it appears to be preferring ffvpx regardless. I found `media.rdd-ffvpx.enabled` and disabled it. This made it use my system ffmpeg and then the hw decoding goes just fine. Is this something new in Firefox 106? Do we need to document this?

elFarto, 2022-10-21:
Looks like this has been raised as bug [1793507](https://bugzilla.mozilla.org/show_bug.cgi?id=1793507) in Bugzilla. I'll leave this issue open until it gets fixed.

philipl, 2022-11-11:
The mozilla bug is closed and marked as 107, but I've just tested with 107 and the bug is still present (and it wasn't closed yesterday so how could it have made it into 107 anyway?). So hopefully fixed in 108

elFarto, 2022-11-11:
The fix is going to be in 108. There is a build available with the fix in now: https://archive.mozilla.org/pub/firefox/nightly/2022/11/2022-11-11-09-37-17-mozilla-central/

philipl, 2022-12-14:
Confirmed that the workaround isn't needed in 108. I'll let you decide whether to close it now or leave this open as a sign for people to upgrade.

elFarto, 2022-12-14:
I think we can close it now, it didn't seem to effect everyone, myself included. Thanks for retesting it.

closed 2022-12-14 by elFarto

goodboydheeraj, 2024-09-22:
Not sure my comment relates to elFarto. But I had to set `media.ffvpx.enabled=false` on my 770 UHD to make youtube 4K/premium bitrate videos playback stable and made my firefox v130 work crash free! Was getting too many `EXCEPTION_BREAKPOINT` `EXCEPTION_ACCESS_VIOLATION_WRITE` and some other errors. Now its running smooth as ever and plays higher resolution 4K youtube videos without a hiccup. The issue started to occur a year ago and I had to RMA my memory sticks and also my 13900k(Intel gave me a 14900k and the issue still persisted) trying to figure out what the issue could be!