elFarto/nvidia-vaapi-driver #124: More than 1 GPU vendor detected via PCI, cannot deduce vendor

jayala, 2022-10-27:
I'm unable to run Firefox with my RTX 2060, it only works with the Intel HD.
GPUs:
```
00:02.0 VGA compatible controller: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
01:00.0 VGA compatible controller: NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
```
NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       782.619429857 [8995-8995] ../src/vabackend.c:1856       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x558070fe5170 40
       782.619437872 [8995-8995] ../src/vabackend.c:1860       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
       782.619441924 [8995-8995] ../src/vabackend.c:1877       __vaDriverInit_1_0 Now have 0 (0 max) instances
       782.625257696 [8995-8995] ../src/export-buf.c: 161       findGPUIndexFromFd Looking for GPU index: 0
       782.627316934 [8995-8995] ../src/export-buf.c: 172       findGPUIndexFromFd Found 4 EGL devices
       782.627405593 [8995-8995] ../src/export-buf.c: 181       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       782.627414033 [8995-8995] ../src/export-buf.c: 205       findGPUIndexFromFd Selecting EGLDevice 0
       782.709776969 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: AB24
       782.709786308 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: XB24
       782.709789126 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: R8  
       782.709791691 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: RG88
       782.709794354 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: UYVY
       782.709796846 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: AB4H
       782.709799334 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: AR24
       782.709801814 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: AS24
       782.709804349 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: XS24
       782.709806840 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: XR24
       782.709809324 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: RG16
       782.709811807 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: AR30
       782.709814412 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: R101
       782.709816891 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: R100
       782.709819368 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: AB30
       782.709822775 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: B101
       782.709826055 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: B100
       782.709829018 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: YU12
       782.709831529 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: NV12
       782.709834025 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: NV21
       782.709837146 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: NV16
       782.709840260 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: NV24
       782.709843458 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: P010
       782.709846530 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: P210
       782.709849577 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: P012
       782.709852590 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: HD01
       782.709855646 [8995-8995] ../src/export-buf.c: 258             initExporter Found format: UHD0
       782.709858634 [8995-8995] ../src/export-buf.c: 269             initExporter Driver doesn't support 16-bit surfaces
       782.709861373 [8995-8995] ../src/export-buf.c:  99                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
       782.726304943 [8995-8995] ../src/vabackend.c:1833              nvTerminate Terminating 0x558070fe5170
       782.726310955 [8995-8995] ../src/export-buf.c:  65          releaseExporter Releasing exporter, 0 outstanding frames
       782.727378832 [8995-8995] ../src/export-buf.c:  82          releaseExporter Done releasing frames
       782.751243920 [8995-8995] ../src/vabackend.c:1847              nvTerminate Now have 0 (0 max) instances
```
Running Firefox (106.0.2) from CLI with env,
```
NVD_MAX_INSTANCES=3 NVD_LOG=1 MOZ_ENABLE_WAYLAND=1 DRI_PRIME=1 LIBVA_DRIVER_NAME=nvidia EGL_PLATFORM=wayland __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json MOZ_DISABLE_RDD_SANDBOX=1 firefox
```
Output
```
[GFX1-]: glxtest: libEGL no display
[GFX1-]: glxtest: EGL test failed
[GFX1-]: More than 1 GPU vendor detected via PCI, cannot deduce vendor
[GFX1-]: PCI candidate 0x8086/0x3e9b
[GFX1-]: PCI candidate 0x10de/0x1f11
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed GL context creation for hardware WebRender: true
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed GL context creation for hardware WebRender: true
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed GL context creation for hardware WebRender: true
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed GL context creation for hardware WebRender: true
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed[3] to create EGL library  display: FEATURE_FAILURE_NO_DISPLAY
[GFX1-]: Failed GL context creation for WebRender: 0
[GFX1-]: FEATURE_FAILURE_WEBRENDER_INITIALIZE_UNSPECIFIED(Create)
[GFX1-]: Failed to connect WebRenderBridgeChild. isParent=true
[GFX1-]: Fallback WR to SW-WR
```
about:support

[image]

elFarto, 2022-10-28:
Unfortunately this is a Firefox issue, and not something I can help with.

jayala, 2022-10-29:
Ok, I try some more stuff and let you know what I find

closed 2022-10-29 by jayala