elFarto/nvidia-vaapi-driver #126: Issue after upgrading to 525.53 driver

qumaciel, 2022-11-11:
Hi there,

After updating to 525.53 (proprietary) driver, I am unable to get hardware acceleration.

<details>
<summary>Console output</summary>

```
>env NVD_LOG=va.log MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox-developer-edition
[GFX1-]: glxtest: VA-API test failed: process crashed. Please check your VA-API drivers.
[Parent 13936: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Parent 13936: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 13936: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 13936: Main Thread]: D/Dmabuf DMABuf is enabled
[Child 14158: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[2022-11-11T22:03:28Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[2022-11-11T22:03:28Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 14158: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[RDD 14297: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 14297: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 14297: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/mp4a-latm'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 14297: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 14297: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 14297: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[... 416 lines cut ...]
     15980.887996137 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: YU12
     15980.887998600 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: NV12
     15980.888001245 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: NV21
     15980.888004258 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: NV16
     15980.888007054 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: NV24
     15980.888009711 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: P010
     15980.888014766 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: P210
     15980.888017165 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: P012
     15980.888019506 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: HD01
     15980.888021734 [13948-13948] ../src/export-buf.c: 258             initExporter Found format: UHD0
     15980.888024280 [13948-13948] ../src/export-buf.c: 267             initExporter Driver supports 16-bit surfaces
     15980.888026900 [13948-13948] ../src/export-buf.c:  99                reconnect Reconnecting to stream
     15980.888089490 [13948-13948] ../src/export-buf.c: 118                reconnect cuda error 'operation not supported' (801)

```
</details>

It works in case a revert the update.
(side note, choosing media.rdd-ffvpx.enabled=true/false makes no difference)

OS: CachyOS x86_64
Kernel: 6.0.8-1-cachyos-pds
DE: Plasma 5.26.3
GPU: NVIDIA GeForce RTX 2060 Mobile
GPU: Intel CoffeeLake-H GT2 [UHD Graphics 630]

qumaciel, 2022-11-14:
There is another thing, running `vainfo` I get a blank output.
```
vainfo -a
Trying display: wayland
Trying display: x11
```

elFarto, 2022-11-14:
Sorry, forgot to comment on this the other day. I've raised this issue on the NVIDIA forums, as I don't think it's an issue on our side.
Also, vainfo doesn't show anything as the driver kills the process if it detects a CUDA error.

nirfse, 2022-11-15:
@elFarto Could you please share the link of your issue on the NVIDIA forums?

elFarto, 2022-11-15:
Sure, it's [here](https://forums.developer.nvidia.com/t/cueglstreamproducerconnect-returns-error-801-on-525-53-driver/233610).

Wish9051, 2022-11-28:
Seems like same issue persists in 525.60.11 version as well.

thesword53, 2022-11-29:
As a workaround, the `direct-backend` branch seems to work with `NVD_BACKEND=direct` environment variable.

qumaciel, 2022-11-30:
Nice! The `direct-backend` branch is working for me as well.

Wish9051, 2022-12-01:
Unfortunately that didn't do it for me. Perhaps I'm missing out on something?

CoelacanthusHex, 2022-12-01:
Unfortunately when I try direct backend, I got 
```
libva error: /usr/lib/dri//nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```.
In strace, I see 
```
openat(AT_FDCWD, "/dev/nvidiactl", O_RDWR|O_CLOEXEC) = 13
openat(AT_FDCWD, "/dev/nvidia0", O_RDWR|O_CLOEXEC) = 14
ioctl(13, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x2b, 0x28), 0x7ffef6f201d0) = 0
ioctl(13, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0xd4, 0x4), 0x7ffef6f201d0) = -1 EINVAL (Invalid argument)
close(13)                               = 0
close(14)                               = 0
ioctl(0, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x2a, 0x20), 0x7ffef6f201d0) = -1 ENOTTY (Inappropriate ioctl for device)
```

kndgs, 2022-12-01:
Not sure if related, but encoding works in OBS but Decoding is broken:
```
[AVHWDeviceContext @ 0x7f2368003c40] libva: /usr/lib/dri/nvidia_drv_video.so init failed
[AVHWDeviceContext @ 0x7f2368003c40] Failed to initialise VAAPI connection: 1 (operation failed).
```

elFarto, 2022-12-01:
The direct backend is very experimental, and hasn't seen a lot of testing, so your mileage may vary.

I'm going to have a look into if there's a way around the original issue, but I'm not holding out much hope.

elFarto, 2022-12-01:
I can't seem to find anyway to get it to work. I have push a small fix to master to let the driver fail gracefully, rather than just exiting the process when it encounters this error. This should let applications like Firefox or MPV not crash or log error messages when it happens and gracefully failover.

Fxzzi, 2022-12-01:
I am having loads of issues with the latest 525 drivers. hardware acceleration in electron apps are broken (discord, pomotroid, etc) and on top of these issues too. Might just rollback to 520 and stay on there until these issues are fixed.

thesword53, 2022-12-02:
> The direct backend is very experimental, and hasn't seen a lot of testing, so your mileage may vary.
> 
> I'm going to have a look into if there's a way around the original issue, but I'm not holding out much hope.

Since Nvidia implemented GBM, I think they deprecated EGLStream.

philipl, 2022-12-02:
Is it worth merging the direct-backend into master as a separate mode? Might be dealing with this for a while.

elFarto, 2022-12-02:
> Is it worth merging the direct-backend into master as a separate mode? Might be dealing with this for a while.

Potentially, but I was still undecided if releasing it is a good idea. It feels like it's going to be a pain to support long term.

thesword53, 2022-12-02:
> Is it worth merging the direct-backend into master as a separate mode? Might be dealing with this for a while.

I think it's worth implementing a GBM backend since `direct-backend` uses directly the Nvidia driver which is not ABI/API stable and EGLStream isn't working with 525 drivers.

elFarto, 2022-12-02:
> I think it's worth implementing a GBM backend since `direct-backend` uses directly the Nvidia driver which is not ABI/API stable and EGLStream isn't working with 525 drivers.

As far as I know, that's not possible, as there's no GBM <> CUDA/NVDEC interoperability. We use the EGL/direct backend to be able to get frames out via a DMA-BUF. With EGL we can export from CUDA -> EGL -> DMA-BUF. With the direct backend, we can talk to the driver to create the image and then import it into CUDA and export to a DMA-BUF.

Lastique, 2022-12-02:
Is it possible to use Vulkan instead of EGL?

zbvk, 2022-12-02:
i can't tell you how happy i am i found this issue i've been pulling my hair out trying to understand what i did wrong turns out it was just a problem with the nvidia driver, sincerely go fuck yourself nvidia.

philipl, 2022-12-02:
> Is it possible to use Vulkan instead of EGL?

Maybe. Newer drivers support dma-buf import/export for buffers in vulkan. The complexity is that the buffer has to export to cuda as opaque fd and to the vaapi client as dma buf. I'm not sure you can export a buffer twice in two formats. There's also a bunch of work you need to do with semaphores to ensure consistent access. If I was going to experiment with this, I'd use libplacebo as it wraps a lot of the vulkan overhead. And of course, one could equally use full OpenGL as the buffer allocator to replace eglstreams.

elFarto, 2022-12-07:
> Is it worth merging the direct-backend into master as a separate mode? Might be dealing with this for a while.

The direct backend has now been merged into the master branch. If everyone could give it a test, that would be much appreciated.

EriksRemess, 2022-12-07:
Still same init failed in mpv.

EriksRemess, 2022-12-07:
Tried with and without NVD_BACKEND=direct, same issue. In Firefox it looks like it tries to use the decoding (I see brief activitiy in  GreenWithEnvy decoder bar) but after playback starts, it plays without gpu decode.

elFarto, 2022-12-07:
Can you run Firefox with NVD_LOG=1 set aswell, and open a new issue with the output?

crimist, 2022-12-07:
Direct backend fixed thsi issue for me.

```
a79b9a@nvidia-vaapi-driver
local/nvidia 525.60.11-2
1080Ti
```

reliant8307, 2022-12-07:
> > Is it worth merging the direct-backend into master as a separate mode? Might be dealing with this for a while.
> 
> The direct backend has now been merged into the master branch. If everyone could give it a test, that would be much appreciated.

Direct backend worked for me too:
cd1f6a9ebf3c94d9f955ff309a04b934c9da3c9b@nvidia-vaapi-driver
Nvidia driver 525.60.11
RTX 2070 Mobile

Owenlaw222, 2022-12-07:
> The direct backend has now been merged into the master branch. If everyone could give it a test, that would be much appreciated.

Works for me:

cd1f6a9e@master
Driver Version: 525.60.11
GTX 1660 SUPER

anonish, 2022-12-07:
does using direct backend work with electron/chrome-based apps like discord?

elFarto, 2022-12-07:
> does using direct backend work with electron/chrome-based apps like discord?

No, Chrome doesn't support using DMA-BUF based VA-API yet, so is not compatible with this library.

anonish, 2022-12-07:
> No, Chrome doesn't support using DMA-BUF based VA-API yet, so is not compatible with this library.

oh sorry, i meant discord (electron based) as per an eariler comment on this issue.   I just upgraded to 525.60.11 and latest git of this package and discord/telegram both seem to be working well, so mark me as another "it works!"

thesword53, 2022-12-08:
> does using direct backend work with electron/chrome-based apps like discord?

VLC also doesn't work because `vaDeriveImage` isn't implemented (black screen).
VLC also supports VDPAU but it doesn't work on Wayland sessions.

```
[00007f3c78c1a9c0] avcodec decoder: Using VA-API NVDEC driver [direct backend] for hardware decoding
[00007f3c58c875b0] vaapi_filters filter error: vaDeriveImage: operation failed
```

elFarto, 2022-12-08:
> oh sorry, i meant discord (electron based) as per an eariler comment on this issue. I just upgraded to 525.60.11 and latest git of this package and discord/telegram both seem to be working well, so mark me as another "it works!"

Yes, my apologies. It is now fixed for other applications that were crashing if they happen to load this library.

anonish, 2022-12-08:
> Yes, my apologies. It is now fixed for other applications that were crashing if they happen to load this library.
that's alright, I had a hard crash overnight with 525 driver so back to 520 for me until 525 stabilizes a bit more.  on 520 i have no issues.

NoXPhasma, 2022-12-09:
It works with `NVD_BACKEND=direct` on driver version 525.60.11 for me on a 1080. But the VRAM fills up over time and never gets freed, until I restart Firefox.

elFarto, 2022-12-09:
> It works with `NVD_BACKEND=direct` on driver version 525.60.11 for me on a 1080. But the VRAM fills up over time and never gets freed, until I restart Firefox.

There was an issue where the cuCtxDestroy call went missing from the nvTerminate function. Can you check if your's has it there?

NoXPhasma, 2022-12-09:
@elFarto I've built it from the AUR yesterday with commit https://github.com/elFarto/nvidia-vaapi-driver/commit/cd1f6a9ebf3c94d9f955ff309a04b934c9da3c9b. Rebuilt it just now and will check if it is fixed.

NoXPhasma, 2022-12-09:
With latest commit, it frees memory once I stop a video or close the tab, but not entirely. It keeps growing the VRAM load over time when I watch several videos. Interestingly enough, I can not see that in `nvidia-smi`. According to the api, FF consumes less than 200 MB. Which comes not even close to how much gets freed once I close Firefox.

elFarto, 2022-12-09:
I'm seeing the same memory issue here. There's a weird artifact of using the NVIDIA API directly, allocated memory doesn't show up against the process in nvidia-smi. It does show up in the total used memory though.

This is going to take a while to track down.

elFarto, 2022-12-09:
Ok, give master a go. I think I found the issue, but I'm slightly confused as to what's happening. I thought when you imported the allocated memory into DRM it closed the fd, hence we had to duplicate the fd prior so we could give that one to CUDA. But now it seems not to be doing that, and the second fd must be closed at the same time the first one is.

NoXPhasma, 2022-12-09:
Can confirm, it now frees the memory.

YidaozhanYa, 2022-12-10:
This direct backend will not work under Optimus system.
It used to work for me when using the legacy EGLStream backend with 520xx drivers. While it doesn't work in Firefox and VLC, it works fine in MPV and SMPlayer. But when switching to this direct backend, I got an `init failed` error.

Anyway, I'm looking forward to a Vulkan backend that works on any system.

elFarto, 2022-12-10:
> This direct backend will not work under Optimus system. It used to work for me when using the legacy EGLStream backend with 520 drivers. While it doesn't work in Firefox and VLC, it works fine in MPV and SMPlayer. But after switching to this direct backend and latest 525 drivers, I got an `init failed` error.

That's likely related to the hard coded device path that's still in the direct-backend.c file. If you change that to you NVIDIA card it should work.

crimist, 2022-12-10:
@YidaozhanYa Try #137

YidaozhanYa, 2022-12-10:
I tried #137 or changing renderD128 to renderD129 in the master branch, neither of which worked for me.  
It's still trying to render with integrated graphics.

```
$ vainfo
Trying display: wayland
Trying display: x11
       129.496002602 [4992-4992] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x562a259ed830 10
       129.496019414 [4992-4992] ../src/vabackend.c:1889       __vaDriverInit_1_0 Invalid driver for DRM device: i915
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

YidaozhanYa, 2022-12-10:
btw I'm on a laptop with MUXless scheme and not using any optimus-manager or external monitor.

YidaozhanYa, 2022-12-10:
The following command correctly gets vainfo to run, but I still can't get it to work in mpv or any media player.

```
$ vainfo --display drm --device /dev/dri/renderD129
Trying display: drm
       584.898899514 [8758-8758] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x557fdc88a8e0 31
       584.898911096 [8758-8758] ../src/vabackend.c:1895       __vaDriverInit_1_0 Now have 0 (0 max) instances
       584.898913875 [8758-8758] ../src/vabackend.c:1920       __vaDriverInit_1_0 Selecting Direct backend
       584.903399679 [8758-8758] ../src/direct/nv-driver.c: 194            init_nvdriver Initing nvdriver...
       584.903420033 [8758-8758] ../src/direct/nv-driver.c: 199            init_nvdriver Got dev info: 100 0 0 0
       584.903472167 [8758-8758] ../src/direct/nv-driver.c:  40          nv_alloc_object Allocating NVObject: C1D00038
       584.903498235 [8758-8758] ../src/direct/nv-driver.c:  40          nv_alloc_object Allocating NVObject: CAF00000
       584.903516717 [8758-8758] ../src/direct/nv-driver.c:  40          nv_alloc_object Allocating NVObject: CAF00001
       584.903529248 [8758-8758] ../src/direct/nv-driver.c: 260            init_nvdriver NVIDIA kernel driver version: 525.60.11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
       584.983866700 [8758-8758] ../src/vabackend.c:1843              nvTerminate Terminating 0x557fdc88a8e0
       584.983871909 [8758-8758] ../src/direct/nv-driver.c:  58           nv_free_object Freeing NVObject: CAF00001
       584.983884302 [8758-8758] ../src/direct/nv-driver.c:  58           nv_free_object Freeing NVObject: CAF00000
       584.983894771 [8758-8758] ../src/direct/nv-driver.c:  58           nv_free_object Freeing NVObject: C1D00038
       584.983919440 [8758-8758] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

YidaozhanYa, 2022-12-10:
I may have completely corrupted something. I rolled back to driver 520.56.06, but it still tries to use direct backend and gives DRM error.
This may be caused by changes brought by the direct backend.

```
$ unset NVD_BACKEND; export NVD_BACKEND=egl; export NVD_LOG=1; export LIBVA_DRIVER_NAME=nvidia

$ nvidia-smi
Sat Dec 10 23:22:48 2022
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 520.56.06    Driver Version: 520.56.06    CUDA Version: 11.8     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   47C    P8     3W /  N/A |      5MiB /  4096MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A       764      G   /usr/lib/Xorg                       4MiB |
+-----------------------------------------------------------------------------+

$ vainfo
Trying display: wayland
Trying display: x11
        98.891975494 [3109-3109] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55fd636ca830 10
        98.892001639 [3109-3109] ../src/vabackend.c:1889       __vaDriverInit_1_0 Invalid driver for DRM device: i915
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

elFarto, 2022-12-10:
> $ vainfo
> Trying display: wayland
> Trying display: x11
>         98.891975494 [3109-3109] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55fd636ca830 10 [...]

Ok, looks like it is possible for an X11 display to pass in a DRM fd. It's using DRI3 to open an fd and pass it into us. I've pushed an update to the DRM fd check to really lock down when we handle it.

Bad3r, 2022-12-11:
@YidaozhanYa I have the same issue on ThinkPad X1 Extreme gen 3. I do though use optimus-manager. I replaced this package with `libva-vdpau-driver-chromium`  from the AUR. This solved the issue for electron applications. 

```
❯ vainfo                                           
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Splitted-Desktop Systems VDPAU backend for VA-API - 0.7.4
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG4Simple            :	VAEntrypointVLD
      VAProfileMPEG4AdvancedSimple    :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
```

```
❯ vainfo --display drm --device /dev/dri/renderD129
Trying display: drm
zsh: segmentation fault (core dumped)  vainfo --display drm --device /dev/dri/renderD129
```

```
❯ nvidia-smi                                       
Sun Dec 11 00:48:17 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.11    Driver Version: 525.60.11    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   58C    P8     6W /  N/A |   1247MiB /  4096MiB |      4%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A     27715      G   /usr/lib/Xorg                     566MiB |
|    0   N/A  N/A     75839      G   ...RendererForSitePerProcess       41MiB |
|    0   N/A  N/A    627499      G   ...AAAAAAAAA= --shared-files       19MiB |
|    0   N/A  N/A    770042      G   ...972527913914867483,131072       16MiB |
|    0   N/A  N/A    777365      G   /usr/lib/firefox/firefox          304MiB |
|    0   N/A  N/A   1129613      G   kitty                               2MiB |
|    0   N/A  N/A   1604086      G   ...RendererForSitePerProcess       75MiB |
|    0   N/A  N/A   1758244      G   ...RendererForSitePerProcess       95MiB |
|    0   N/A  N/A   3338699      G   ...RendererForSitePerProcess       98MiB |
|    0   N/A  N/A   3374581      G   kitty                               2MiB |
|    0   N/A  N/A   3375211      G   ...RendererForSitePerProcess       16MiB |
+-----------------------------------------------------------------------------+

````

YidaozhanYa, 2022-12-11:
> Ok, looks like it is possible for an X11 display to pass in a DRM fd. It's using DRI3 to open an fd and pass it into us. I've pushed an update to the DRM fd check to really lock down when we handle [...]

In the latest commit, vainfo works, but media players still don't work properly with hardware decoding.

I tried mpv, and the 3D usage of the integrated graphics and CPU usage are both high, which means mpv is using software decoding. This works just like Firefox with 520xx drivers and EGL backend.

According to my shallow understanding of Optimus system, it is not possible to achieve _direct_ rendering with discrete graphics on a laptop with a MUXless scheme. Video output always goes through the integrated graphics. Of course, there may still be some kind of "magic" making it work.

```
       176.554230476 [2009-2024] ../src/direct/direct-export-buf.c: 272    direct_realiseSurface Unable to realise surface: 0x7f00ecd25f30 (-1)
       176.554232572 [2009-2024] ../src/vabackend.c:1825    nvExportSurfaceHandle Unable to export surface
       176.554239490 [2009-2024] ../src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f00ecd25f30)
       176.554242566 [2009-2024] ../src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f00ecd21600)
       176.554246208 [2009-2024] ../src/vabackend.c: 578           nvCreateConfig got profile: 21 with 0 attributes
       176.554250130 [2009-2024] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) (nil) 0
       176.554252566 [2009-2024] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) 0x7f00ecd24000 7
       176.555528257 [2009-2024] ../src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7f00ecd21600)
       176.555537171 [2009-2024] ../src/vabackend.c:1314            nvDeriveImage In nvDeriveImage
       176.555544194 [2009-2024] ../src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7f00ecd24830)
       176.555546750 [2009-2024] ../src/vabackend.c:1314            nvDeriveImage In nvDeriveImage
       176.555570775 [2009-2024] ../src/vabackend.c:1820    nvExportSurfaceHandle Exporting surface: -1 (0x7f00ecd24830)
       176.555573689 [2009-2024] ../src/direct/direct-export-buf.c: 124 direct_allocateBackingImage Allocating BackingImages: 0x7f00ecd23870 128x128
       176.555576317 [2009-2024] ../src/direct/nv-driver.c: 412              alloc_image Calculated GOB size: 64x128
       176.555579240 [2009-2024] ../src/direct/nv-driver.c: 424              alloc_image Aligned image size: 128x128 = 16384 (16384)
       176.555689150 [2009-2024] ../src/direct/nv-driver.c: 476              alloc_image DRM_IOCTL_PRIME_HANDLE_TO_FD failed: -1
       176.555692591 [2009-2024] ../src/direct/nv-driver.c: 412              alloc_image Calculated GOB size: 64x64
       176.555695462 [2009-2024] ../src/direct/nv-driver.c: 424              alloc_image Aligned image size: 128x64 = 8192 (8192)
       176.555793855 [2009-2024] ../src/direct/nv-driver.c: 476              alloc_image DRM_IOCTL_PRIME_HANDLE_TO_FD failed: -1
       176.555797029 [2009-2024] ../src/direct/direct-export-buf.c:  87           import_to_cuda importing memory size: 0x0 = 0
       176.555801392 [2009-2024] ../src/direct/direct-export-buf.c:  89           import_to_cuda CUDA ERROR 'invalid argument' (1)
```

YidaozhanYa, 2022-12-11:
> Ok, looks like it is possible for an X11 display to pass in a DRM fd. It's using DRI3 to open an fd and pass it into us. I've pushed an update to the DRM fd check to really lock down when we handle [...]

Latest commit works. Thanks!

YidaozhanYa, 2022-12-11:
The latest commit works for me, I see mpv in nvidia-smi and DEC in nvtop when using `mpv --hwdec=nvdec-copy`. This means that this issue has been resolved, and in 525 and future drivers, direct should be set as the default backend.
However, after many rounds of comparison, I found that nvidia-vaapi-driver is not available for Optimus systems at all. I also get high CPU and integrated graphics 3D usage while the NVIDIA graphics card is decoding the video. This is true even for previous 520xx drivers where I drew the wrong conclusions. On Windows with DXVA2 hwdec I only got 6% integrated graphics 3D usage, while on Linux with NVIDIA vaapi driver I got 98%.
It seems that whether it is with EGL or direct backend enabled, it is very performance-intensive when displaying frames decoded by NVIDIA graphics to the laptop screen. The resulting performance has a lot of dropped frames and is not even as good as decoding video directly with integrated graphics.

I'm also sorry for the multiple sloppy replies I just posted, I've now deleted them. I may need to create a new issue, or continue this topic in #11.

[525 logs.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10201720/525.logs.txt)
[520 logs.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10201722/520.logs.txt)

2shrestha22, 2022-12-11:
Direct backend works for me with 1650Ti mobile GPU.

2shrestha22, 2022-12-11:
After a reboot I got:
```
❯ vainfo
Trying display: wayland
Trying display: x11
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

reliant8307, 2022-12-11:
I can confirm the error on vainfo, using 4029d456846f9e1e13763466c9d95824e29ceefe (but video playing on Firefox is perfect):

```
$ NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/xorg/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     88772.279380810 [13332-13332] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x1b216d0 10
     88772.279384514 [13332-13332] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
     88772.279386321 [13332-13332] ../src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
     88772.282063294 [13332-13332] ../src/direct/direct-export-buf.c:  59      direct_initExporter Manually opened DRM device
     88772.282067662 [13332-13332] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
     88772.282069723 [13332-13332] ../src/direct/nv-driver.c: 165          get_device_info DRM_IOCTL_NVIDIA_GET_DEV_INFO failed: -1
     88772.282071654 [13332-13332] ../src/direct/nv-driver.c:  79            nv_rm_control nv_rm_control failed: -1 0
libva error: /usr/xorg/lib/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

elFarto, 2022-12-11:
I've merged in #137, which should do better when manually having to open the DRM device.

reliant8307, 2022-12-11:
> I've merged in #137, which should do better when manually having to open the DRM device.

#137 fixed the issue with vainfo. Thanks!

2shrestha22, 2022-12-12:
> > After a reboot I got:
> > ```
> > ❯ vainfo
> > Trying display: wayland
> > Trying display: x11
> > libva error: /usr/lib/dri/nvidia_drv_video.so init failed
> > vaInitialize failed with error code 1 (operation failed),exit
> > ```
> 
> Please post the output with `NVD_LOG=1` set.

Will post the output if I got this issue again. Today morning it is working. I am not sure what is causing this. Thank you.

Korothi, 2022-12-13:
Download latest git
set NVD_BACKEND=direct
profit?
works on ampere (tested on 3080, arch)

[15 more replies]

closed 2022-12-17 by elFarto