elFarto/nvidia-vaapi-driver #129: Direct backend errors with operation failed

elFarto, 2022-12-03:
Unfortunately when I try direct backend, I got 
```
libva error: /usr/lib/dri//nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
In strace, I see 
```
openat(AT_FDCWD, "/dev/nvidiactl", O_RDWR|O_CLOEXEC) = 13
openat(AT_FDCWD, "/dev/nvidia0", O_RDWR|O_CLOEXEC) = 14
ioctl(13, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x2b, 0x28), 0x7ffef6f201d0) = 0
ioctl(13, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0xd4, 0x4), 0x7ffef6f201d0) = -1 EINVAL (Invalid argument)
close(13)                               = 0
close(14)                               = 0
ioctl(0, _IOC(_IOC_READ|_IOC_WRITE, 0x46, 0x2a, 0x20), 0x7ffef6f201d0) = -1 ENOTTY (Inappropriate ioctl for device)
```

_Originally posted by @CoelacanthusHex in https://github.com/elFarto/nvidia-vaapi-driver/issues/126#issuecomment-1333498661_

elFarto, 2022-12-03:
@CoelacanthusHex Could you possibly run the direct backend again with `NVD_LOG=1` set as well, and post the log here?

CoelacanthusHex, 2022-12-03:
```
>>> NVD_BACKEND=direct LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 vainfo
Trying display: wayland
      1154.651901020 [21020-21020] ../src/vabackend.c:1871       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5a8a81de5b60 40
      1154.651911007 [21020-21020] ../src/vabackend.c:1876       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
      1154.651914289 [21020-21020] ../src/vabackend.c:1893       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1154.651917153 [21020-21020] ../src/vabackend.c:1918       __vaDriverInit_1_0 Selecting Direct backend
      1154.654812847 [21020-21020] ../src/direct/nv-driver.c: 187            init_nvdriver Initing nvdriver...
      1154.654824720 [21020-21020] ../src/direct/nv-driver.c: 192            init_nvdriver Got dev info: 1 0 0 0
      1154.654858104 [21020-21020] ../src/direct/nv-driver.c: 220            init_nvdriver nv_attach_gpu failed
      1154.654861945 [21020-21020] ../src/direct/nv-driver.c: 275            init_nvdriver Got error initing
      1154.654923266 [21020-21020] ../src/direct/nv-driver.c:  76            nv_rm_control nv_rm_control failed: -1 0
libva error: /usr/lib/dri//nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

The environment changed slightly (I downgraded the Nvidia driver to 520), but the error is still the same (including strace results).

elFarto, 2022-12-03:
Do you have multiple GPUs? Potentially an iGPU? 

Can you paste the output of this command: `ls -l /dev/dri/renderD*`?

CoelacanthusHex, 2022-12-03:
> Do you have multiple GPUs? Potentially an iGPU?
> 
> Can you paste the output of this command: `ls -l /dev/dri/renderD*`?

Yes, I have two GPUs, iGPU is AMD Cezanne, dGPU is RTX 3050 Mobile.

```
Permissions    Size User Group  Date Modified    Name
crw-rw-rw-  226,128 root render 2022-12-03 19:26 /dev/dri/renderD128
crw-rw-rw-  226,129 root render 2022-12-03 19:25 /dev/dri/renderD129
```

elFarto, 2022-12-03:
Can you see if this works:
```
NVD_BACKEND=direct LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129
```

CoelacanthusHex, 2022-12-03:
> Can you see if this works:
> NVD_BACKEND=direct LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129

Yes, it works.

elFarto, 2022-12-03:
The first issue is a bug in the driver due to you using Wayland, and the library not checking the passed into DRM fd to see if it's compatible. That fd should have been rejected upfront with an error. This is partly caused by setting LIBVA_DRIVER_NAME, which isn't needed in the most recent version of libva (you're forcing it to use this driver for the AMD iGPU).

Does the direct (or egl for that matter) backend work within Firefox for you? As far as I know, this library doesn't work if Firefox is running on the iGPU, as the DMA-BUFs we export can't be imported there. If it does work, can you look at what GPU Firefox is using in `about:support`.

CoelacanthusHex, 2022-12-03:
No, it doesn't work. I tried four methods:
1. `LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 NVD_BACKEND=direct MOZ_LOG="PlatformDecoderModule:5;Dmabuf:5" firefox-nightly`
2. `LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 NVD_BACKEND=direct MOZ_LOG="PlatformDecoderModule:5;Dmabuf:5" __NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidia firefox-nightly`
3. `LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 MOZ_LOG="PlatformDecoderModule:5;Dmabuf:5" __NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidiafirefox-nightly`
4. `LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 MOZ_LOG="PlatformDecoderModule:5;Dmabuf:5" firefox-nightly`

They all ware blocked by gfxtest failed:
1. `[GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection.`
2. 
```
[GFX1-]: No GPUs detected via PCI
[GFX1-]: glxtest: process failed (received signal 11)
```
3.
```
[GFX1-]: No GPUs detected via PCI
[GFX1-]: glxtest: process failed (received signal 11)
```
4. `[GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection.`

elFarto, 2022-12-03:
You could also try setting `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json`

CoelacanthusHex, 2022-12-03:
> You could also try setting `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json`

with `LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 MOZ_LOG="PlatformDecoderModule:5;Dmabuf:5" __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json __NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidia firefox-nightly` w/ and w/o `NVD_BACKEND=direct`, I still got
```
[GFX1-]: No GPUs detected via PCI
[GFX1-]: glxtest: process failed (received signal 11)
```

elFarto, 2022-12-03:
I'm not sure I can fix that. I believe the issue is with Firefox on Wayland. When it goes to run it's 'gfxtest' tool it grabs the default EGLDisplay based on what the Wayland compositor is running on, and I assume in your case that's running on the iGPU.
Normally on X11, setting the __EGL_VENDOR_LIBRARY_FILENAMES variable restricts what devices are available to force it onto the right GPU. Under Wayland that doesn't seem to be working.

I'm not sure there's much I can do about this. I've run into this issue numerous times. Firefox just isn't too happy about running on a non-default GPU, too much of the code is hardcoded to use the default one.

thesword53, 2022-12-05:
> ```
> [GFX1-]: No GPUs detected via PCI
> [GFX1-]: glxtest: process failed (received signal 11)
> ```

This is an issue with Firefox with some Wayland compositors. I have the same issue using Sway but not with KDE Plasma/Gnome with Wayland.

https://bugzilla.mozilla.org/show_bug.cgi?id=1759315

elFarto, 2022-12-19:
Closing this issue as the root issues was the hardcoded DRM device, and the secondary issues is the Optimus issue.

closed 2022-12-19 by elFarto