elFarto/nvidia-vaapi-driver #130: Error in vainfo after setting everything up as in the readme

gargoless, 2022-12-06:
Hello I've installed the driver and followed the setup as described in the readme with the goal of using hardware accel. in firefox and something's not working, the following logs are of running vainfo with NVD_LOG=1

```
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
[1119-1119] ../src/vabackend.c:1634       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55b5d7793140
[1119-1119] ../src/export-buf.c: 171          findCudaDisplay Found 3 EGL devices
[1119-1119] ../src/export-buf.c: 175          findCudaDisplay Got EGL_CUDA_DEVICE_NV value '0' from device 0
[1119-1119] ../src/export-buf.c: 123    checkModesetParameter Unable to retrieve render node, deriving render node from master node
[1119-1119] ../src/export-buf.c: 135    checkModesetParameter Checking device file: /dev/dri/renderD128
[1119-1119] ../src/export-buf.c: 210             initExporter Got EGLDisplay from CUDA device
[1119-1119] ../src/export-buf.c: 240             initExporter Driver doesn't support 16-bit surfaces
[1119-1119] ../src/export-buf.c:  95                reconnect Reconnecting to stream
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: VA-API NVDEC driver
Segmentation fault
```

SETUP
64bit
rtx 3060, driver 470.141.03, CUDA ver. 11.4
firefox 102.5.0esr
I have a bit of a weird set-up  because of old packages in Debian so I guess that's probably the cause, I'm aware of it and I come with low expectations:
Debian 11.5 (Bullseye)
The package nvidia-vaapi-driver(ver0.0.5-1) is on Debian testing but not on stable so I pulled it creating a simple backport. In this process I also had to pull the nvidia-codec-headers (also only in testing, also as a manual/simple backport) to build it.
I set up the configs in firefox and variables as described in the readme.

When option `nvidia-drm.modeset=1` is set firefox crashes on start-up.

elFarto, 2022-12-06:
We had a similar issue with #71, which seemed to be resolved by upgrading libva. 

I would also recommend upgrading the NVIDIA driver version to 520 if you can (I'd avoid 525 at the moment, we have issues with it), as the 470 series is only borderline supported as it doesn't have GBM support, which Firefox really wants.

philipl, 2022-12-09:
Should we expect `vainfo` to work with the direct backend when using an X11 display? I can only make it work with `--display drm`.

elFarto, 2022-12-09:
I checked a fix in for that this morning. I made a minor error assuming if a drm_state was passed in, it would have a valid fd.

elFarto, 2022-12-19:
Can you check if this issue still exists in the v0.0.8 version?

elFarto, 2023-01-07:
@gargoless Is this still an issue?

gargoless, 2023-01-07:
Sorry I had completely forgotten about this issue. My initial issue was with the version of libva I was running, it had a bug in it that was interfering. I never seriously tried anything beyond that since I was stuck with an nvidia drivers version that technically shouldn’t work.

closed 2023-01-07 by gargoless