elFarto/nvidia-vaapi-driver #131: Firefox - glxtest: VA-API test failed: failed to initialise VAAPI connection.

Jamstah, 2022-12-07:
Fedora 37, can get vainfo working fine, but can't get firefox to use it. Have tried all the debug steps I could find that seem to relate but I'm none the wiser.

Firefox 107.0.1 (64-bit)
Fedora 37
wayland
nvidia-vaapi-driver-0.0.7-2.fc37.x86_64
akmod-nvidia-520.56.06-1.fc37.x86_64

```
jammy:~$ vainfo
Trying display: wayland
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
jammy:~$ echo $MOZ_DISABLE_RDD_SANDBOX 
1
jammy:~$ echo $LIBVA_DRIVER_NAME 
nvidia
jammy:~$ echo $EGL_PLATFORM 
wayland
jammy:~$ firefox
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection. (t=0.207879) [GFX1-]: glxtest: VA-API test failed: failed to initialise VAAPI connection.
[2022-12-07T17:42:08Z ERROR glean_core::metrics::ping] Invalid reason code startup for ping background-update
[2022-12-07T17:42:13Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-07T17:42:13Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-07T17:42:13Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-07T17:42:13Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-07T17:42:36Z ERROR viaduct::backend::ffi] Missing HTTP status
[2022-12-07T17:42:36Z ERROR viaduct::backend::ffi] Missing HTTP status
[2022-12-07T17:42:36Z ERROR viaduct::backend::ffi] Missing HTTP status
```

[image]

Details logs: https://pastebin.com/cq5MWiu9

elFarto, 2022-12-07:
Looks like you might be suffering from issue #122. Can you try setting `media.rdd-ffvpx.enabled` to `false` within Firefox, and see if that helps?

Jamstah, 2022-12-08:
I've included that in the pastebin link at the end: https://pastebin.com/cq5MWiu9

elFarto, 2022-12-08:
Oh, my apologies, I completely missed that NVD_LOG=1 was set. There are no log statements from this driver in that output, which is concerning. It does seem like Firefox is crashing when attempting to load VA-API on start-up, which doesn't make much sense.

Jamstah, 2022-12-08:
Just ran again and also noticed these errors:
```
[Child 102966: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 102966: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 102966: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 102966: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 102966: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 102786: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Child 102786: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD128 error Permission denied
[Child 102966: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Child 102966: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD128 error Permission denied
```

Which is odd because:
```
jammy:~$ ls -la /dev/dri
total 0
drwxr-xr-x.  3 root root        140 Dec  7 17:58 .
drwxr-xr-x. 23 root root       4940 Dec  7 17:58 ..
drwxr-xr-x.  2 root root        120 Dec  7 17:58 by-path
crw-rw----+  1 root video  226,   0 Dec  7 17:58 card0
crw-rw----+  1 root video  226,   1 Dec  7 17:58 card1
crw-rw-rw-.  1 root render 226, 128 Dec  7 17:58 renderD128
crw-rw-rw-.  1 root render 226, 129 Dec  7 17:58 renderD129
jammy:~$ groups
jammy wheel render libvirt
```

But I don't know if that would be a red herring?

I'm also on an optimus system, I see you asked that before:
```
jammy:~$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation TigerLake-H GT1 [UHD Graphics] (rev 01)
01:00.0 VGA compatible controller: NVIDIA Corporation TU117GLM [T1200 Laptop GPU] (rev a1)
```

And here's the nvd.txt, which does imply its going to be dual video card related :(
```
jammy:~$ cat nvd.txt 
     55932.639334568 [104951-104951] ../src/vabackend.c:1781       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7ffb5df1f880 31
     55932.639367097 [104951-104951] ../src/vabackend.c:1796       __vaDriverInit_1_0 Invalid driver for DRM device: i915
```

elFarto, 2022-12-08:
Ok, you have a Optimus GPU setup, and Firefox is trying to run on the Intel card, while `LIBVA_DRIVER_NAME=nvidia` forces VA-API to use this driver. You can try adding `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json`, but lots of people have been having difficulty with this sort of setup.

thesword53, 2022-12-08:
Is `libva-intel-driver` or `intel-media-driver` installed on your system?

Jamstah, 2022-12-08:
Silly question, why not try it?

That works, and I now have non-jittery youtube, which is pretty awesome, thanks for the pointer!

Netflix won't play anything though, but I'm sure I'll work it out.

Jamstah, 2022-12-08:
OK, closing this as I have something that works for me, but if it would be useful for me to mess with anything optimus related I'm happy to put in some time and give back, but I'd need some guidance on what to try.

Thanks both for the help!

closed 2022-12-08 by Jamstah

hholst80, 2023-05-25:
Just FYI, I had the same problem on Wayland and Fedora 38. The package mentioned fixed the error message.