elFarto/nvidia-vaapi-driver #132: [drm.h] Compilation error on arch

ManuLinares, 2022-12-07:
```
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/user/scripts/nvidia-vaapi-driver/build
ninja: Entering directory `/home/user/scripts/nvidia-vaapi-driver/build'
[1/15] Compiling C object nvidia_drv_video.so.p/src_direct_nv-driver.c.o
FAILED: nvidia_drv_video.so.p/src_direct_nv-driver.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_nv-driver.c.o -MF nvidia_drv_video.so.p/src_direct_nv-driver.c.o.d -o nvidia_drv_video.so.p/src_direct_nv-driver.c.o -c ../src/direct/nv-driver.c
../src/direct/nv-driver.c:11:10: fatal error: drm/drm_fourcc.h: No such file or directory
   11 | #include <drm/drm_fourcc.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
[2/15] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
FAILED: nvidia_drv_video.so.p/src_h264.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_h264.c.o -MF nvidia_drv_video.so.p/src_h264.c.o.d -o nvidia_drv_video.so.p/src_h264.c.o -c ../src/h264.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:13,
                 from ../src/h264.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[3/15] Compiling C object nvidia_drv_video.so.p/src_hevc.c.o
FAILED: nvidia_drv_video.so.p/src_hevc.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_hevc.c.o -MF nvidia_drv_video.so.p/src_hevc.c.o.d -o nvidia_drv_video.so.p/src_hevc.c.o -c ../src/hevc.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:13,
                 from ../src/hevc.c:3:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[4/15] Compiling C object nvidia_drv_video.so.p/src_av1.c.o
FAILED: nvidia_drv_video.so.p/src_av1.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_av1.c.o -MF nvidia_drv_video.so.p/src_av1.c.o.d -o nvidia_drv_video.so.p/src_av1.c.o -c ../src/av1.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:13,
                 from ../src/av1.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[5/15] Compiling C object nvidia_drv_video.so.p/src_export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_export-buf.c.o -MF nvidia_drv_video.so.p/src_export-buf.c.o.d -o nvidia_drv_video.so.p/src_export-buf.c.o -c ../src/export-buf.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:13,
                 from ../src/export-buf.c:1:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
[6/15] Compiling C object nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I../subprojects/nv-codec-headers-11.1.5.1/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o -MF nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o.d -o nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o -c ../src/direct/direct-export-buf.c
In file included from ../nvidia-include/nvidia.h:9,
                 from ../src/direct/../direct/nv-driver.h:7,
                 from ../src/direct/../vabackend.h:13,
                 from ../src/direct/direct-export-buf.c:3:
../nvidia-include/nvidia-drm-ioctl.h:26:10: fatal error: drm/drm.h: No such file or directory
   26 | #include <drm/drm.h>
      |          ^~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

elFarto, 2022-12-07:
Ah, yes, I had forgotten about that. We choose between the kernel version and the libdrm version elsewhere in the code, but I hadn't done that in the new direct backend.

ManuLinares, 2022-12-07:
Pretty sure I already have the kernel headers.

```
~ $ uname -r
6.0.11-1-cachyos
~ $ yay -Q linux-cachyos-headers 
linux-cachyos-headers 6.0.11-1
~ $ yay -Q linux-cachyos
linux-cachyos 6.0.11-1
~ $ 

~ $ ls /usr/include/libdrm/
nouveau/      drm_mode.h      intel_debug.h  qxl_drm.h        radeon_bo_int.h  radeon_surface.h  via_drm.h
amdgpu_drm.h  drm_sarea.h     mach64_drm.h   r128_drm.h       radeon_cs_gem.h  savage_drm.h      virtgpu_drm.h
amdgpu.h      i915_drm.h      mga_drm.h      r600_pci_ids.h   radeon_cs.h      sis_drm.h         vmwgfx_drm.h
drm_fourcc.h  intel_aub.h     msm_drm.h      radeon_bo_gem.h  radeon_cs_int.h  tegra_drm.h
drm.h         intel_bufmgr.h  nouveau_drm.h  radeon_bo.h      radeon_drm.h     vc4_drm.h

```

elFarto, 2022-12-07:
The `/usr/include/libdrm` directory is the user-space libraries copy of those headers. The kernel ones live in `/usr/include/drm`.

khaneliman, 2022-12-07:
> I've just pushed a new change to master that should resolve this issue.

Worked for me, thanks.

ManuLinares, 2022-12-07:
excelent, thanks

closed 2022-12-07 by ManuLinares