elFarto/nvidia-vaapi-driver #135: Build issue direct backend

rkoot, 2022-12-08:
Hi,
I tried to build the latest revision on opensuse tumbleweed, nvidia-driver-525.60.11, cuda-11.8, gcc-12.2.1 , but it fails with the following compiler error:

~/devel/nvidia-vaapi-driver/build> ninja -j1
[1/13] Compiling C object nvidia_drv_video.so.p/src_av1.c.o
FAILED: nvidia_drv_video.so.p/src_av1.c.o 
ccache cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_av1.c.o -MF nvidia_drv_video.so.p/src_av1.c.o.d -o nvidia_drv_video.so.p/src_av1.c.o -c ../src/av1.c
In file included from /usr/include/EGL/eglplatform.h:134,
                 from /usr/include/EGL/egl.h:20,
                 from ../src/vabackend.h:6,
                 from ../src/av1.c:1:
../nvidia-include/nv-ioctl.h:74:14: error: two or more data types in declaration specifiers
   74 |     NvU32    Status;
      |              ^~~~~~
In file included from ../nvidia-include/nvidia.h:8,
                 from ../src/direct/nv-driver.h:7,
                 from ../src/vabackend.h:13:
../nvidia-include/nv-ioctl.h:74:20: warning: declaration does not declare anything
   74 |     NvU32    Status;
      |                    ^
../nvidia-include/nv-ioctl.h:83:14: error: two or more data types in declaration specifiers
   83 |     NvU32    Status;
      |              ^~~~~~
../nvidia-include/nv-ioctl.h:83:20: warning: declaration does not declare anything
   83 |     NvU32    Status;
      |                    ^
ninja: build stopped: subcommand failed.

elFarto, 2022-12-08:
That looks like you have a #define for the word Status, but I'm not sure where that's coming from. As a quick test you can edit lines 75 and 83 to have `status` rather than `Status`, and that should let it compile.

rkoot, 2022-12-09:
yeah, it seems to be Xlib.h that's the culprit.
It seems to build indeed, but on runtime I now get this:

NVD_BACKEND=direct NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     57501.420577702 [13083-13083] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x561a903af350 10
     57501.420580828 [13083-13083] ../src/vabackend.c:1878       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
     57501.420584605 [13083-13083] ../src/vabackend.c:1895       __vaDriverInit_1_0 Now have 0 (0 max) instances
     57501.420587410 [13083-13083] ../src/vabackend.c:1920       __vaDriverInit_1_0 Selecting Direct backend
     57501.423197688 [13083-13083] ../src/direct/direct-export-buf.c:  59      direct_initExporter Manually opened DRM device
     57501.423202777 [13083-13083] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
     57501.423205903 [13083-13083] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 700 1 2 6
     57501.423253323 [13083-13083] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
     57501.477279603 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477284011 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477286676 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477289090 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477293659 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477295953 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477299290 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477301233 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477303357 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477305562 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477306764 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477307916 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477310100 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

     57501.477311232 [13083-13083] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (1453599520)

vainfo: Supported profile and entrypoints
     57501.477316402 [13083-13083] ../src/vabackend.c:1843              nvTerminate Terminating 0x561a903af350
     57501.477528141 [13083-13083] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances

Not sure why it's failing on cuda errors.
I'm using a RTX3070 (8GB vram) btw.

elFarto, 2022-12-09:
Those are some odd errors. Can you run both `NVD_LOG=1 vainfo` and `NVD_BACKEND=direct NVD_LOG=1 vainfo --display drm` and paste the output here?

rkoot, 2022-12-09:
############ NVD_BACKEND unset ################
NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     61285.581041096 [17806-17806] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5572ab575350 10
     61285.581044041 [17806-17806] ../src/vabackend.c:1878       __vaDriverInit_1_0 Non-DRM display type detected, defaulting to GPU ID 0. Use NVD_GPU to pick a specific GPU.
     61285.581046476 [17806-17806] ../src/vabackend.c:1895       __vaDriverInit_1_0 Now have 0 (0 max) instances
     61285.581049732 [17806-17806] ../src/vabackend.c:1917       __vaDriverInit_1_0 Selecting EGL backend
     61285.583338047 [17806-17806] ../src/export-buf.c: 161       findGPUIndexFromFd Looking for GPU index: 0
     61285.584181167 [17806-17806] ../src/export-buf.c: 173       findGPUIndexFromFd Found 3 EGL devices
     61285.584229849 [17806-17806] ../src/export-buf.c: 182       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     61285.584234477 [17806-17806] ../src/export-buf.c: 134 checkModesetParameterFromFd Unable to check nvidia_drm modeset setting
     61285.584236070 [17806-17806] ../src/export-buf.c: 206       findGPUIndexFromFd Selecting EGLDevice 0
     61285.584240729 [17806-17806] ../src/export-buf.c: 210       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 1
     61285.584243414 [17806-17806] ../src/export-buf.c: 213       findGPUIndexFromFd No DRM device file for EGLDevice 2
     61285.584244386 [17806-17806] ../src/export-buf.c: 216       findGPUIndexFromFd No match found, falling back to default device
     61285.584892528 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AB24
     61285.584896306 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: XB24
     61285.584897488 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: R8  
     61285.584898420 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RG88
     61285.584901105 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: R16 
     61285.584902888 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RG32
     61285.584905703 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: GR32
     61285.584907427 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: UYVY
     61285.584908428 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AR12
     61285.584910492 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AB12
     61285.584912406 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RA12
     61285.584914129 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BA12
     61285.584915141 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: XR15
     61285.584917235 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: XB15
     61285.584918067 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RX15
     61285.584919960 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BX15
     61285.584920962 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AR15
     61285.584921834 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AB15
     61285.584922715 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RA15
     61285.584924539 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BA15
     61285.584925541 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AB4H
     61285.584927334 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: XB4H
     61285.584928246 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AR24
     61285.584929979 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AS24
     61285.584930981 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: XS24
     61285.584931863 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: XR24
     61285.584932744 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RGB8
     61285.584935159 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BGR8
     61285.584936992 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RG16
     61285.584938685 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BG16
     61285.584939497 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RG24
     61285.584941160 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BG24
     61285.584943254 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: AR30
     61285.584944887 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: RX24
     61285.584946620 [17806-17806] ../src/export-buf.c: 261         egl_initExporter Found format: BX24
[... cut ...]
     61463.834161374 [17984-17984] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (401023056)

     61463.834162586 [17984-17984] ../src/vabackend.c: 332      doesGPUSupportCodec CUDA ERROR '(null)' (401023056)

vainfo: Supported profile and entrypoints
     61463.834165912 [17984-17984] ../src/vabackend.c:1843              nvTerminate Terminating 0x55d06af27a30
     61463.834318821 [17984-17984] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances

elFarto, 2022-12-09:
Can you retry with the latest master version? That shouldn't require any header file changes.

rkoot, 2022-12-09:
Still the same.

rkoot, 2022-12-09:
Ahhh. it seems the above mess was completely my fault.
Latest master builds fine and also works fine.

elFarto, 2022-12-09:
What exactly was causing it? As I have no clue how you managed to break it in that way.

rkoot, 2022-12-09:
well... <blush>I was toying around with nvidia's videocodec 12.x and managed to copy the stub libs from that zip file in my system library path. those broke the lot. Not completely sure if it was the stub libs per se, or that it was the mismatch between said stubs and the header files (still 11.x), but the result was a broken nvdec/nvenc. removing the stub libraries got it working</blush>
So yeah, major cockup...