elFarto/nvidia-vaapi-driver #136: Failed to build in nvidia-340

xswqaz, 2022-12-08:
Failed to build with below error

The Meson build system
Version: 0.61.2
Source dir: /home/user/nvidia-vaapi-driver
Build dir: /home/user/nvidia-vaapi-driver/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 11.3.0 "cc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Library EGL found: YES
Dependency gstreamer-codecparsers-1.0 found: YES 1.20.3 (cached)
Dependency ffnvcodec found: YES 11.1.5.1 (cached)
Dependency libva found: YES 1.14.0 (cached)
Dependency threads found: YES unknown (cached)
Build targets in project: 1

ninja: Entering directory `/home/user/nvidia-vaapi-driver/build'
[1/1] Linking target nvidia_drv_video.so
FAILED: nvidia_drv_video.so 
cc  -o nvidia_drv_video.so nvidia_drv_video.so.p/src_av1.c.o nvidia_drv_video.so.p/src_export-buf.c.o nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o nvidia_drv_video.so.p/src_direct_nv-driver.c.o nvidia_drv_video.so.p/src_h264.c.o nvidia_drv_video.so.p/src_hevc.c.o nvidia_drv_video.so.p/src_jpeg.c.o nvidia_drv_video.so.p/src_mpeg2.c.o nvidia_drv_video.so.p/src_mpeg4.c.o nvidia_drv_video.so.p/src_vabackend.c.o nvidia_drv_video.so.p/src_vc1.c.o nvidia_drv_video.so.p/src_vp8.c.o nvidia_drv_video.so.p/src_list.c.o nvidia_drv_video.so.p/src_vp9.c.o -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,nvidia_drv_video.so /usr/lib/x86_64-linux-gnu/libva.so /usr/lib/x86_64-linux-gnu/libgstcodecparsers-1.0.so /usr/lib/x86_64-linux-gnu/libgstbase-1.0.so /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so /usr/lib/x86_64-linux-gnu/libgobject-2.0.so /usr/lib/x86_64-linux-gnu/libglib-2.0.so -lEGL -lm -ldl -Wl,--end-group -pthread
/usr/bin/ld: nvidia_drv_video.so.p/src_export-buf.c.o: in function `egl_initExporter':
/home/user/nvidia-vaapi-driver/build/../src/export-buf.c:237: undefined reference to `eglGetPlatformDisplay'
/usr/bin/ld: nvidia_drv_video.so.p/src_export-buf.c.o: in function `egl_destroyBackingImage':
/home/user/nvidia-vaapi-driver/build/../src/export-buf.c:329: undefined reference to `eglDestroyImage'
/usr/bin/ld: nvidia_drv_video.so.p/src_export-buf.c.o: in function `egl_allocateBackingImage':
/home/user/nvidia-vaapi-driver/build/../src/export-buf.c:470: undefined reference to `eglCreateImage'
/usr/bin/ld: /home/user/nvidia-vaapi-driver/build/../src/export-buf.c:474: undefined reference to `eglDestroyImage'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Could not rebuild /home/user/nvidia-vaapi-driver/build

Any clues?

elFarto, 2022-12-08:
The compile error is because you have a very old version of the EGL headers. But aside from that, this driver won't work with the 340 NVIDIA driver, it requires the 470 or 500+ series.

xswqaz, 2022-12-09:
Is eglDestroyImage function required for hwaccel in browser ( chrome / firefox ) ?
It will be for helpful for low-end card user, If there is option for old driver

elFarto, 2022-12-09:
I'm afraid so. Unfortunately the 470 series of drivers is the oldest we can support as those are the ones that gained the ability to export DMA-BUFs.