elFarto/nvidia-vaapi-driver #138: Failed to create drm render device

wael444, 2022-12-14:
```
 $ NVD_LOG=1 NVD_BACKEND=direct LIBVA_DRIVER_NAME=nvidia MOZ_LOG="Dmabuf:5" MOZ_DISABLE_RDD_SANDBOX=1 firefox
[Parent 17425: Main Thread]: D/Dmabuf We're missing DRM render device!
[Parent 17425: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 17425: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 17425: Main Thread]: D/Dmabuf Failed to create drm render device
```

wael444, 2022-12-14:
```
$ LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: x11
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

wael444, 2022-12-14:
yep.. however getting this now:
```
[RDD 359: MediaPDecoder #1]: D/Dmabuf Using DRM device /dev/dri/renderD129
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      8161.641573206 [359-534] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f3050009a40 31
      8161.641578636 [359-534] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
      8161.641580564 [359-534] ../src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
      8161.643658423 [359-534] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      8161.643664608 [359-534] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 0 0 0
      8161.643717657 [359-534] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
libva info: va_openDriver() returns 0
      8161.690294995 [359-534] ../src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
      8161.700004361 [359-534] ../src/vabackend.c: 578           nvCreateConfig got profile: 6 with 0 attributes
      8161.700013079 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      8161.700015089 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f3050a6d340 7
      8161.700539173 [359-534] ../src/vabackend.c: 578           nvCreateConfig got profile: 7 with 0 attributes
      8161.700542805 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      8161.700544399 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f3050a6d480 7
      8161.701080189 [359-534] ../src/vabackend.c: 578           nvCreateConfig got profile: 13 with 0 attributes
      8161.701083820 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      8161.701085360 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f3050a6d5c0 7
      8161.701606756 [359-534] ../src/vabackend.c: 578           nvCreateConfig got profile: 14 with 0 attributes
      8161.701610365 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      8161.701612054 [359-534] ../src/vabackend.c:1596 nvQuerySurfaceAttributes with 9 (8) 0x7f3050a6af00 7
Sandbox: attempt to open unexpected file /sys/devices/system/cpu/cpu0/cache/index2/size
Sandbox: attempt to open unexpected file /sys/devices/system/cpu/cpu0/cache/index3/size
Sandbox: attempt to open unexpected file /sys/devices/system/cpu/present
Sandbox: attempt to open unexpected file /sys/devices/system/cpu/possible
JavaScript error: , line 0: TypeError: NetworkError when attempting to fetch resource.
JavaScript error: , line 0: TypeError: NetworkError when attempting to fetch resource.
[Child 32659: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD129
[Child 32659: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD129 error Permission denied
```
despite:
```
$ stat /dev/dri/*
  File: /dev/dri/card0
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 5h/5d   Inode: 277         Links: 1     Device type: e2,0
Access: (0777/crwxrwxrwx)  Uid: (    0/    root)   Gid: (   13/   video)
Access: 2022-12-14 18:08:04.096716404 +0300
Modify: 2022-12-14 18:08:04.096716404 +0300
Change: 2022-12-14 18:47:30.720003904 +0300
  File: /dev/dri/card1
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 5h/5d   Inode: 283         Links: 1     Device type: e2,1
Access: (0660/crw-rw----)  Uid: (    0/    root)   Gid: (   13/   video)
Access: 2022-12-14 20:16:53.216566704 +0300
Modify: 2022-12-14 20:16:53.216566704 +0300
Change: 2022-12-14 20:16:53.226566703 +0300
  File: /dev/dri/renderD128
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 5h/5d   Inode: 276         Links: 1     Device type: e2,80
Access: (0777/crwxrwxrwx)  Uid: ( 1000/    wael)   Gid: ( 1000/    wael)
Access: 2022-12-14 18:08:04.096716404 +0300
Modify: 2022-12-14 18:08:04.096716404 +0300
Change: 2022-12-14 20:18:54.799897695 +0300
  File: /dev/dri/renderD129
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 5h/5d   Inode: 282         Links: 1     Device type: e2,81
Access: (0777/crwxrwxrwx)  Uid: ( 1000/    wael)   Gid: ( 1000/    wael)
Access: 2022-12-14 20:16:53.216566704 +0300
Modify: 2022-12-14 20:16:53.216566704 +0300
Change: 2022-12-14 20:18:54.799897695 +0300
```

elFarto, 2022-12-14:
Can you try `NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display drm --device /dev/dri/renderD129` and see what it outputs?

wael444, 2022-12-15:
```
Trying display: drm
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       251.503679214 [12013-12013] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x562e9830b8e0 31
       251.503683442 [12013-12013] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
       251.503684995 [12013-12013] ../src/vabackend.c:1916       __vaDriverInit_1_0 Selecting EGL backend
       251.505591286 [12013-12013] ../src/export-buf.c: 161       findGPUIndexFromFd Looking for DRM device index: 0
       251.515467710 [12013-12013] ../src/export-buf.c: 175       findGPUIndexFromFd Found 3 EGL devices
       251.515499298 [12013-12013] ../src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       251.515501247 [12013-12013] ../src/export-buf.c: 190       findGPUIndexFromFd Found drmDeviceIndex: 0
       251.515502832 [12013-12013] ../src/export-buf.c: 225       findGPUIndexFromFd Selecting EGLDevice 0
       251.515506304 [12013-12013] ../src/export-buf.c: 229       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 1
       251.515508115 [12013-12013] ../src/export-buf.c: 232       findGPUIndexFromFd No DRM device file for EGLDevice 2
       251.515509427 [12013-12013] ../src/export-buf.c: 235       findGPUIndexFromFd No match found, falling back to default device
       251.526538388 [12013-12013] ../src/export-buf.c: 289         egl_initExporter Driver supports 16-bit surfaces
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
       251.797243459 [12013-12013] ../src/vabackend.c:1843              nvTerminate Terminating 0x562e9830b8e0
       251.797247093 [12013-12013] ../src/export-buf.c:  65      egl_releaseExporter Releasing exporter, 0 outstanding frames
       251.797251128 [12013-12013] ../src/export-buf.c:  82      egl_releaseExporter Done releasing frames
       251.797252777 [12013-12013] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

wael444, 2022-12-15:
wait.. it is working!

but, just not specifically under youtube? works in something alternative like invidious. i suppose this should be closed as i don't use youtube anyway and that was for testing.

wael444, 2022-12-15:
another thing i have with this vaapi driver is that video playback is slow, for example if i pause and then wait exactly 5 seconds for
```
       525.677548732 [12550-13463] ../src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f6700e99620)
       525.677551661 [12550-13463] ../src/vabackend.c:1843              nvTerminate Terminating 0x7f6700d52cc0
       525.677975597 [12550-13463] ../src/vabackend.c:1857              nvTerminate Now have 1 (0 max) instances
```
to happen, playing the video again causes a delay.

closed 2022-12-15 by wael444

elFarto, 2022-12-15:
With regards to it not working on Youtube, it likes to use AV1 which your card doesn't support, so it's recommended to disable that in about:config.

The termination happening after 5 seconds is all Firefox's doing, there's nothing I can do about that.

wael444, 2022-12-15:
I'm also primarily doing i915 VA-API so that I have my igpu do something. I bought a new cpu with an igpu and I thought I can give it some use. Do you think its worth it? I can't get it working at all.

elFarto, 2022-12-15:
I have no knowledge of getting Intel VA-API working, so I can't comment on that.

wael444, 2022-12-15:
Nonono. I want your thoughts about it. Should I use my (presumably) but difficult to setup Intel iGPU for some hardware acceleration so the recording load can be taken off my CPU? While gaming and recording it can have an affect so..

elFarto, 2022-12-15:
Any hardware accelerated encoding will help take a load off the CPU so it probably worth it, but you're probably better getting NVENC working and using that to the any encoding you need to do. I should note that this library can't do encoding, you'll need to use NVENC directly from whatever application you're using for recording.

wael444, 2022-12-15:
so AFAIK, i can't really use VA-API with my iGPU, (and the only realistic use case): 
```
[Parent 1809: Main Thread]: D/Dmabuf We're missing DRM render device!
[Parent 1809: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 1809: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 1809: Main Thread]: D/Dmabuf Failed to create drm render device
```
however, `mpv` and `ffmpeg` can use the iGPU just fine, which i use quite rarely still. firefox might have the most benefit but it does not want to work..