elFarto/nvidia-vaapi-driver #141: [direct backend] nv_rm_control failed: 0 57

YidaozhanYa, 2022-12-17:
System information:
KDE Plasma 5.26.4 on X11
Integrated Graphics: Intel CometLake-H GT2 [UHD Graphics]
NVIDIA Graphics: NVIDIA GeForce GTX 1650 Ti Mobile
No optimus-manager or bumblebee is used

Related logs:
[vainfo.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10251278/vainfo.txt)
[vlc.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10251279/vlc.txt)
[mpv-vaapi.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10251277/mpv-vaapi.txt)
[chromium.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10251275/chromium.txt)
[firefox.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10251276/firefox.txt)

```
 ➜ vainfo
Trying display: wayland
Trying display: x11
      1584.561294865 [43611-43611] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x563808dea830 10
      1584.561303181 [43611-43611] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1584.561304615 [43611-43611] ../src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
      1584.564994600 [43611-43611] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      1584.565004049 [43611-43611] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 0 0 0 0
      1584.565075145 [43611-43611] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
      1584.565084783 [43611-43611] ../src/direct/nv-driver.c:  79            nv_rm_control nv_rm_control failed: 0 57
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      1584.638244179 [43611-43611] ../src/vabackend.c:1843              nvTerminate Terminating 0x563808dea830
      1584.638276164 [43611-43611] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

elFarto, 2022-12-17:
Are you using v0.0.8?

YidaozhanYa, 2022-12-17:
This is what I just cloned and compiled from this git repository.
At first, I thought it was a duplicate of #103, but after I turned on modeset, everything still didn't work as before.

elFarto, 2022-12-17:
Can you try running `NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129` (assuming renderD129 is your NVIDIA card).

YidaozhanYa, 2022-12-17:
```
 ➜ export NVD_BACKEND=direct; export NVD_LOG=1; export LIBVA_DRIVER_NAME=nvidia

 ➜ NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129
Trying display: drm
       392.954482676 [6501-6501] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5614413ac8e0 31
       392.954490591 [6501-6501] ../src/vabackend.c:1888       __vaDriverInit_1_0 Invalid driver for DRM device: i915
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
This time i got a different error.

YidaozhanYa, 2022-12-17:
D129 is indeed my NVIDIA card, but vainfo recognizes wrong device, like 2 weeks ago

YidaozhanYa, 2022-12-17:
(After setting modeset, NVIDIA card becomes D128)

elFarto, 2022-12-17:
I've just pushed a fix I think will help.

YidaozhanYa, 2022-12-17:
The new push works, the 0 57 error doesn't occur anymore, but I still can't use nvidia-vaapi-driver. Everything gave the same error, just the 0 57 error didn't occur.

(I can't figure out what's wrong and don't know how to change the issue title)

YidaozhanYa, 2022-12-17:
```
 ➜ vainfo --display drm --device /dev/dri/renderD128
Trying display: drm
      1502.502354997 [14528-14528] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55d3a4a3f8e0 31
      1502.502363876 [14528-14528] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1502.502366456 [14528-14528] ../src/vabackend.c:1920       __vaDriverInit_1_0 Selecting Direct backend
      1502.506445466 [14528-14528] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      1502.506457112 [14528-14528] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 2 6
      1502.506595613 [14528-14528] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      1502.576142214 [14528-14528] ../src/vabackend.c:1843              nvTerminate Terminating 0x55d3a4a3f8e0
      1502.576182934 [14528-14528] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

elFarto, 2022-12-17:
I think you just have an Optimus issue now. For this driver to work with Firefox, both have to be using the NVIDIA GPU, but trying to get Firefox to run on the GPU that isn't the GPU running X11 seems to be very difficult.

YidaozhanYa, 2022-12-17:
Not just Firefox, everything with vaapi doesn't work.
Using optimus-manager can make X11 run on nvidia card, which can make vaapi work, but it will cause huge screen delay and tearing, so vaapi is meaningless then.

elFarto, 2022-12-17:
Unfortunately I don't have a real Optimus system to test with (my desktop Intel+NVIDIA system doesn't seem to act the same as a Laptop for some reason), so resolving these Optimus issues is difficult.

marioroy, 2022-12-17:
I do not have an Optimus system to test. Perhaps the following may resolve the issue.
```bash
$ cat /etc/X11/xorg.conf.d/nvidia-drm-outputclass.conf 

Section "OutputClass"
    Identifier "nvidia"
    MatchDriver "nvidia-drm"
    Driver "nvidia"
    Option "AllowEmptyInitialConfiguration"
    Option "PrimaryGPU" "yes"
    Option "SLI" "Auto"
    Option "BaseMosaic" "on"
EndSection

Section "OutputClass"
    Identifier "intel"
    MatchDriver "i915"
    Driver "modesetting"
EndSection
```

YidaozhanYa, 2022-12-17:
> I do not have an Optimus system to test. Perhaps the following may resolve the issue. Basically, set NVIDIA as the primary GPU and have the Intel GPU use the modesetting driver. I saw this from an EL9 NVIDIA RPM.
> 
> ```shell
> $ cat /etc/X11/xorg.conf.d/nvidia-drm-outputclass.conf
> ```
> 
> ```
> Section "OutputClass"
>     Identifier "nvidia"
>     MatchDriver "nvidia-drm"
>     Driver "nvidia"
>     Option "AllowEmptyInitialConfiguration"
>     Option "PrimaryGPU" "yes"
>     Option "SLI" "Auto"
>     Option "BaseMosaic" "on"
> EndSection
> 
> Section "OutputClass"
>     Identifier "intel"
>     MatchDriver "i915"
>     Driver "modesetting"
> EndSection
> ```

This Xorg profile will make X run on the NVIDIA card with huge display lag, tearing and frame drops, so not a good solution.

marioroy, 2022-12-18:
I misunderstood your intention on wanting both OpenGL and HW decoding acceleration to occur on the NVIDIA GPU. Thank you for clarifying.

[image]

xtellaris, 2022-12-19:
After upgrading `v525` Nvidia driver, I reproduced this issue on my old Optimus laptop (MUXless) with Intel iGPU + Nvidia 1050ti dGPU (mobile ver.) and `nvidia-vaapi-driver v0.0.8` on `direct` backend.`vainfo` reacts as well as `v520` drvier, but somehow no application can use vaapi. However, it would work when I fallbacked to `v520` driver + default `EGL` backend.

<details>
  <summary>
System info:</summary>
```
OS: Arch Linux x86_64 
Host: 80WW Lenovo R720-15IKBN 
Kernel: 6.1.0-2-cachyos-tt-lto 
Resolution: 1920x1080 
DE: Plasma 5.26.4 
WM: KWin 
CPU: Intel i5-7300HQ (4) @ 3.500GHz 
GPU: Intel HD Graphics 630 
GPU: NVIDIA GeForce GTX 1050 Ti Mobile 
```
</details>

<details>
  <summary>
`vainfo` with `NVD_LOG=1`</summary>
```bash
$ NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
      3946.807369180 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x56222bdbf400 10
      3946.807378086 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3946.807382529 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
      3946.812762990 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/direct/direct-export-buf.c:  94      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      3946.812772203 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      3946.812796033 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 0 0
      3946.812909381 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      3946.898700073 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x56222bdbf400
      3946.898797690 [11902-11902] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances

```
</details>

According to the log above, I believe `nvidia-vaapi-driver v0.0.8` do find the right drm device, but I tested mannually as well.
<details>
  <summary>
vainfo --display drm --device /dev/dri/renderD129
</summary>
```
$ NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129
Trying display: drm
      4710.969159513 [12585-12585] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x5615b00728e0 31
      4710.969194533 [12585-12585] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
      4710.969211394 [12585-12585] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
      4710.974636477 [12585-12585] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      4710.974661048 [12585-12585] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 0 0
      4710.974802033 [12585-12585] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
vainfo: VA-API version: 1.16 (libva 2.16.0)
[... 84 lines cut ...]
[   0.194][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.194][d][vo/gpu/vaapi] Unsupported format: p010
[   0.196][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.196][d][vo/gpu/vaapi] Unsupported format: nv12
[   0.197][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.197][d][vo/gpu/vaapi] Unsupported format: p010
[   0.199][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.199][d][vo/gpu/vaapi] Unsupported format: nv12
[   0.200][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.200][d][vo/gpu/vaapi] Unsupported format: p010
[   0.203][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.203][d][vo/gpu/vaapi] Unsupported format: nv12
[   0.203][d][vo/gpu/vaapi] Failed to import surface in EGL: 12300
[   0.203][d][vo/gpu/vaapi] Unsupported format: p010
[   0.203][d][vo/gpu/vaapi] Supported formats:
[   0.203][v][vo/gpu/vaapi] Done probing surface formats.
[   0.229][v][vo/gpu] Loading failed.
[   0.229][v][vd] Could not create device.
[   0.229][v][vd] No hardware decoding available for this codec.
[   0.229][v][vd] Using software decoding.
...
```
</details>

I don't know how to cat the log of Firefox, but it also cannot use vaapi when playing an avc stream video, and `nvidia-pmon` reports 0 ultilization about Decoder.

elFarto, 2022-12-19:
@xtellaris Can you try the latest master version?

xtellaris, 2022-12-20:
> @xtellaris Can you try the latest master version?

Tried https://github.com/elFarto/nvidia-vaapi-driver/commit/7113c20184e32abeb6cb3857392a1549bbcd7a5a and `v525` Nvidia driver, vaapi still looks good with `vainfo` but broken with mpv or firefox. Logs also looks the same as which of `v0.0.8` would report.
---
By the way, I downgrade to `v520` Nvidia driver again with master version nvidia-vaapi-driver remaining, then remove `env` about `direct` backend and reboot. This time, the `EGL` backend is broken in applications as well ( `vainfo` report is good). `mpv` log : 
```
[   0.292][v][vd] Opening decoder vp9
[   0.292][v][vd] Looking at hwdec vp9-vaapi...
[   0.292][v][vo/gpu] Loading hwdec drivers for format: 'vaapi'
[   0.292][v][vo/gpu] Loading hwdec driver 'vaapi'
[   0.292][v][vo/gpu/vaapi] using EGL dmabuf interop
[   0.292][v][vo/gpu/vaapi] Trying to open a x11 VA display...
[   0.292][d][vo/gpu/vaapi/vaapi] libva: VA-API version 1.16.0
[   0.292][d][vo/gpu/vaapi/vaapi] libva: User environment variable requested driver 'nvidia'
[   0.292][d][vo/gpu/vaapi/vaapi] libva: Trying to open /usr/lib/dri/nvidia_drv_video.so
[   0.307][d][vo/gpu/vaapi/vaapi] libva: Found init function __vaDriverInit_1_0
[   0.355][d][vo/gpu/vaapi/vaapi] libva: va_openDriver() returns 0
[   0.355][v][vo/gpu/vaapi/vaapi] Initialized VAAPI: version 1.16
[   0.355][d][ffmpeg] AVHWDeviceContext: VAAPI driver: VA-API NVDEC driver [egl backend].
[   0.355][d][ffmpeg] AVHWDeviceContext: Driver not found in known nonstandard list, using standard behaviour.
[   0.355][v][vo/gpu/vaapi] Going to probe surface formats (may log bogus errors)...
[   0.382][d][vo/gpu/vaapi] Supported formats:
[   0.382][d][vo/gpu/vaapi]  nv12
[   0.382][d][vo/gpu/vaapi]  p010
[   0.382][v][vo/gpu/vaapi] Done probing surface formats.
[   0.382][v][vd] Trying hardware decoding via vp9-vaapi.
[   0.382][v][vd] Selected codec: vp9 (Google VP9)
[   0.382][v][vf] User filter list:
[   0.382][v][vf]   (empty)
[   0.382][v][cplayer] Starting playback...
[   0.383][v][vd] Pixel formats supported by decoder: cuda vaapi vdpau yuv420p
[   0.383][v][vd] Codec profile: Profile 0 (0x0)
[   0.409][v][vd] Requesting pixfmt 'vaapi' from decoder.
[   0.574][e][ffmpeg/video] vp9: Failed to create decode context: 2 (resource allocation failed).
[   0.574][e][ffmpeg/video] vp9: Failed setup for format vaapi: hwaccel initialisation returned error.
[   0.574][v][vd] Pixel formats supported by decoder: cuda vdpau yuv420p yuv420p
[   0.574][v][vd] Codec profile: Profile 0 (0x0)
[   0.574][v][vd] Requesting pixfmt 'yuv420p' from decoder.
[   0.576][d][osc] osc_init 
[   0.576][v][vd] Falling back to software decoding.
```
---
So I would suggest Optimus device user relys on `v520` Nvidia driver + `v0.0.8` nvidia-vaapi-driver for some time. And thanks for the good work !

marioroy, 2022-12-20:
I can confirm Mpv segfault with master `7113c20` on `v520.61.05`. All is well on `v520` Nvidia driver and the `v0.0.8` nvidia-vaapi-driver release.

marioroy, 2022-12-20:
Mpv playblak using Nvidia driver `v525.60.13` and `NVD_BACKEND=direct` works with commit [bb7f6fc](https://github.com/elFarto/nvidia-vaapi-driver/commit/bb7f6fcb66e595647bc3c1bc5a4592479c5d2e0b).

elFarto, 2022-12-20:
Ok, I've just tested on my system, and those versions work fine. So the issue is just for Optimus systems.

The `Failed to import surface in EGL` errors are likely mpv trying to import into an Intel EGL context, which won't work.

marioroy, 2022-12-20:
Oh yikes... :) My desktop system does not have Optimus.  I reverted to commit [bb7f6fc](https://github.com/elFarto/nvidia-vaapi-driver/commit/bb7f6fcb66e595647bc3c1bc5a4592479c5d2e0b).  Mpv `NVD_BACKEND=direct` segfaults with commit `7113c20` on `v525.60.13`.

YidaozhanYa, 2022-12-20:
For me, with the latest master commit (7113c20), things got different. mpv will now crash with a segmentation fault.
[mpv.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10268074/mpv.txt)
[prime-run mpv.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10268081/prime-run.mpv.txt)

elFarto, 2022-12-20:
Ah, it's the vaapi-copy that's doing it. I can reproduce it locally now so I can investigate.

elFarto, 2022-12-20:
Ok, give master a go. Looks like it was a small issue from the YUV444 work that only effected the copy path.

marioroy, 2022-12-20:
No segmentation fault this time. However, not working playing 10-bit media `mpv --hwdec=vaapi-copy --vo=xv`; e.g. https://youtu.be/Jz9TdfXlTgs format 699 (1920x1080 24 FPS 10 HDR).

```text
[ffmpeg/video] av1: Hardware does not support image size 1920x1080 (constraints: width 128-8192 height 1587019272-8192).
[ffmpeg/video] av1: Failed setup for format vaapi: hwaccel initialisation returned error.
[ffmpeg/video] av1: Your platform doesn't suppport hardware accelerated AV1 decoding.
[ffmpeg/video] av1: Failed to get pixel format.
```

No issues with commit [bb7f6fc](https://github.com/elFarto/nvidia-vaapi-driver/commit/bb7f6fcb66e595647bc3c1bc5a4592479c5d2e0b).

marioroy, 2022-12-20:
<details><summary>youtube-play-vaapi script</summary>
```bash
#!/usr/bin/env bash
###############################################################################
#
# A script for playing YouTube videos via the VA-API interface.
#
# by Mario Roy, August 2022
#    for testing video hardware acceleration in mpv
#
url=$1

# Display usage.
if [ -z "$url" -o "$url" = "-h" -o "$url" = "--help" ]; then
    echo 'The YouTube URL is "Best 8k HDR of 2020 Dolby Vision"'
    echo ""
    echo "Usage:"
    echo "  youtube-play-vaapi URL -F, --list-formats"
    echo "  youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs -F"
    echo "  youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs --list-formats"
    echo ""
    echo "  youtube-play-vaapi URL [VIDEO_FORMAT_CODE]"
    echo "  youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs      #  auto avc1"
    echo "  youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 136  #  720p avc1"
    echo "  youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 137  # 1080p avc1"
    echo ""
    exit 1
fi

# YouTube Downloader ##########################################################
#
# Installation: pip3 install --user --upgrade yt-dlp
#
youtube_dl=~/.local/bin/yt-dlp

if [ ! -f "$youtube_dl" ]; then
    echo "Please install yt-dlp as described in the README file."
    echo "To install, run: pip3 install --user --upgrade yt-dlp"
    exit 1
fi

# List all available formats of the requested video.
if [ "$2" = "-F" -o "$2" = "--list-formats" ]; then
    ${youtube_dl} --list-formats "$url"
    exit 0
fi

# Set the format code if given on the command line.
if [ -n "$2" ]; then
    video_format=$2
else
    video_format="bestvideo[height<=?1080][fps<=?30][ch!=?2]"
fi

# Fetch the title and audio+video URLs into an array.
# title: ${lines[0]}, audio: ${lines[1]}, video: ${lines[2]}

readarray -t lines < <(
    ${youtube_dl} --get-title -g -S "vcodec:avc1" \
        -f "bestaudio[ext=webm]+${video_format}" "$url"
)

# mpv #########################################################################
#
# Using NVIDIA graphics?
#   10-bit (HDR) media works with --hwdec=vaapi-copy and --vo={xv,x11} options.
#   VP9 HDR works well. For AV1 HDR, run youtube-play-nvidia.
#
mpv --hwdec=vaapi-copy --vo=xv --force-media-title="${lines[0]}" \
    --audio-file="${lines[1]}" "${lines[2]}"

stty echo # restore echo
```
</details>

xtellaris, 2022-12-20:
Tested `mpv --no-config --hwdec=vaapi-copy` to play a vp9 video with latest commit https://github.com/elFarto/nvidia-vaapi-driver/commit/f91a078eefd25ffd1253723050b3630f4833aaa5 on my Optimus machine with `v520` driver. `mpv` reported failed initialization.
```
[   0.072][v][vd] Opening decoder vp9
[   0.072][v][vd] Looking at hwdec vp9-vaapi-copy...
[   0.072][d][vaapi] libva: VA-API version 1.16.0
[   0.072][d][vaapi] libva: User environment variable requested driver 'nvidia'
[   0.072][d][vaapi] libva: Trying to open /usr/lib/dri/nvidia_drv_video.so
[   0.082][d][vaapi] libva: Found init function __vaDriverInit_1_0
[   0.082][e][vaapi] libva: /usr/lib/dri/nvidia_drv_video.so init failed
[   0.082][d][vaapi] libva: va_openDriver() returns 1
[   0.082][e][vaapi] Failed to initialize VAAPI: operation failed
[   0.082][v][vd] Could not create device.
[   0.082][v][vd] No hardware decoding available for this codec.
[   0.082][v][vd] Using software decoding.

```
And `mpv --no-config --hwdec=vaapi`(non-copy decoder) still reported broken like we talked before.

marioroy, 2022-12-20:
8-bit vp9 : `./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 243`

[image]

10-bit vp9.2 : `./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 332`

[image]

elFarto, 2022-12-20:
> Tested `mpv --no-config --hwdec=vaapi-copy` to play a vp9 video with latest commit

Can you run that again with NVD_LOG=1 set?

xtellaris, 2022-12-20:
> > Tested `mpv --no-config --hwdec=vaapi-copy` to play a vp9 video with latest commit
> 
> Can you run that again with NVD_LOG=1 set?

<details>
   <summary>vainfo</summary>

```sh-session
$ NVD_LOG=1 vainfo                    
Trying display: wayland
Trying display: x11
     18617.000468062 [21816-21816] ../src/vabackend.c:2079       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     18617.000481508 [21816-21816] ../src/vabackend.c:2088       __vaDriverInit_1_0 Now have 0 (0 max) instances
     18617.000487249 [21816-21816] ../src/vabackend.c:2111       __vaDriverInit_1_0 Selecting EGL backend
     18617.006142321 [21816-21816] ../src/export-buf.c: 137       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     18617.006152590 [21816-21816] ../src/export-buf.c: 154       findGPUIndexFromFd Looking for GPU index: 0
     18617.008503462 [21816-21816] ../src/export-buf.c: 166       findGPUIndexFromFd Found 4 EGL devices
     18617.008611529 [21816-21816] ../src/export-buf.c: 175       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     18617.008646480 [21816-21816] ../src/export-buf.c: 213       findGPUIndexFromFd Selecting EGLDevice 0
     18617.009883169 [21816-21816] ../src/export-buf.c: 284         egl_initExporter Driver supports 16-bit surfaces
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
     18617.100282926 [21816-21816] ../src/vabackend.c:2051              nvTerminate Terminating 0x5635f6688400
     18617.100290411 [21816-21816] ../src/export-buf.c:  66      egl_releaseExporter Releasing exporter, 0 outstanding frames
     18617.100295772 [21816-21816] ../src/export-buf.c:  83      egl_releaseExporter Done releasing frames
     18617.100300584 [21816-21816] ../src/vabackend.c:2065              nvTerminate Now have 0 (0 max) instances
```
</details>

---
Seems it didn't use the right DRM device:

<details>
    <summary>mpv with NVD_LOG=1</summary>

```sh-session
$ NVD_LOG=1 mpv --no-config \
--hwdec=vaapi-copy --log-file=/tmp/mpv.log \
"/home/xtellaris/Videos/Test/Mad Max - Fury Road - Official Retaliate Trailer [HD] [MonFNCgK4WE].webm"
 (+) Video --vid=1 (*) (vp9 1920x1080 23.976fps)
     18529.483018244 [21722-21722] ../src/vabackend.c:2079       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     18529.483030001 [21722-21722] ../src/vabackend.c:2088       __vaDriverInit_1_0 Now have 0 (0 max) instances
     18529.483035148 [21722-21722] ../src/vabackend.c:2111       __vaDriverInit_1_0 Selecting EGL backend
     18529.483053366 [21722-21722] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
[vaapi] libva: /usr/lib/dri/nvidia_drv_video.so init failed
[vaapi] Failed to initialize VAAPI: operation failed
VO: [gpu] 1920x1080 yuv420p
V: 00:00:11 / 00:02:32 (8%) Dropped: 4

Exiting... (Quit)
```
</details>

---
Try to specify the DRM device:
[... 53 lines cut ...]
     18841.285292610 [21924-21924] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x5578015457d0)
     18841.285296741 [21924-21924] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     18841.285301172 [21924-21924] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x557801545240)
     18841.285305191 [21924-21924] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     18841.285309760 [21924-21924] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x557801544cb0)
     18841.285313789 [21924-21924] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     18841.285318349 [21924-21924] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x557801544720)
     18841.285322451 [21924-21924] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     18841.285327018 [21924-21924] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x55780102bea0)
     18841.285331089 [21924-21924] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     18841.285335698 [21924-21924] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x55780102e420)
     18841.285339804 [21924-21924] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
[... cut ...]
     18841.287118723 [21924-21924] ../src/vabackend.c:2051              nvTerminate Terminating 0x5578010245f0
     18841.287129673 [21924-21924] ../src/export-buf.c:  66      egl_releaseExporter Releasing exporter, 0 outstanding frames
     18841.287134920 [21924-21924] ../src/export-buf.c:  83      egl_releaseExporter Done releasing frames
     18841.287139914 [21924-21924] ../src/vabackend.c:2065              nvTerminate Now have 0 (0 max) instances
VO: [gpu] 1920x1080 yuv420p
V: 00:00:02 / 00:02:32 (2%)

Exiting... (Quit)

```
</details>

elFarto, 2022-12-20:
NVD_GPU isn't meant to be used like that, it's meant to just be the index of the GPU you want. The driver will pick the first NVIDIA GPU it finds without that variable being set. 

You can use mpv's `--vaapi-device=/dev/...` parameter to set what device to use for the vaapi-copy hwdec (it doesn't work with the vaapi hwdec option for some reason).

However, the out of memory error is puzzling. Your card has more than enough VRAM to handle 12 surfaces, so I'm not certain what's going on there.

xtellaris, 2022-12-20:
> NVD_GPU isn't meant to be used like that, it's meant to just be the index of the GPU you want. The driver will pick the first NVIDIA GPU it finds without that variable being set.

 I believe most Optimus laptop have only one Nvidia GPU. In my case, `nvidia-smi -L` says it is "GPU 0", So the value for `NVD_GPU` should be '0' ?

I tried with `--vaapi-device`, it reported "out of memery" again.
<details>
    <summary> mpv log </summary>

```sh-session
$ NVD_LOG=1 mpv --no-config \ 
--hwdec=vaapi-copy --vaapi-device=/dev/dri/renderD129 --log-file=/tmp/mpv.log \
"/home/xtellaris/Videos/Win-Videos/Test/Mad Max - Fury Road - Official Retaliate Trailer [HD] [MonFNCgK4WE].webm"
 (+) Video --vid=1 (*) (vp9 1920x1080 23.976fps)
     20802.199994481 [23262-23262] ../src/vabackend.c:2079       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     20802.200004913 [23262-23262] ../src/vabackend.c:2088       __vaDriverInit_1_0 Now have 0 (0 max) instances
     20802.200027402 [23262-23262] ../src/vabackend.c:2111       __vaDriverInit_1_0 Selecting EGL backend
     20802.200048239 [23262-23262] ../src/export-buf.c: 152       findGPUIndexFromFd Looking for DRM device index: 1
     20802.200905322 [23262-23262] ../src/export-buf.c: 166       findGPUIndexFromFd Found 4 EGL devices
     20802.201058964 [23262-23262] ../src/export-buf.c: 175       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     20802.201066887 [23262-23262] ../src/export-buf.c: 181       findGPUIndexFromFd Found drmDeviceIndex: 1
     20802.201089728 [23262-23262] ../src/export-buf.c: 213       findGPUIndexFromFd Selecting EGLDevice 0
     20802.202242830 [23262-23262] ../src/export-buf.c: 284         egl_initExporter Driver supports 16-bit surfaces
     20802.267166942 [23262-23262] ../src/vabackend.c:1393      nvQueryImageFormats In nvQueryImageFormats
     20802.294746056 [23262-23262] ../src/vabackend.c: 637           nvCreateConfig got profile: 19 with 0 attributes
     20802.294761722 [23262-23262] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (8) (nil) 0
     20802.294785442 [23262-23262] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (8) 0x55a272864fc0 7
     20802.296616659 [23262-23262] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (8) (nil) 0
     20802.296625985 [23262-23262] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (8) 0x55a272959b40 7
     20802.298363312 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a27234f9e0)
     20802.298373924 [23262-23262] ../src/vabackend.c:1487            nvDeriveImage In nvDeriveImage
     20802.298381289 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a27234f520)
     20802.298386967 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a27234d960)
     20802.298392198 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a27295ccd0)
     20802.298397476 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a272865ce0)
     20802.298402502 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a272866270)
     20802.298407518 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a272866800)
     20802.298412534 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a272866d90)
     20802.298417511 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a272867320)
     20802.298422433 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a2728678b0)
     20802.298427469 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a272867e40)
     20802.298432448 [23262-23262] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x55a2728683d0)
     20802.323672366 [23262-23262] ../src/vabackend.c: 637           nvCreateConfig got profile: 19 with 0 attributes
     20802.323687919 [23262-23262] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (8) (nil) 0
     20802.323693729 [23262-23262] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (8) 0x55a272865780 7
     20802.326394406 [23262-23262] ../src/vabackend.c: 979          nvCreateContext with 12 render targets, 12 surfaces, at 1920x1080
     20802.367015798 [23262-23262] ../src/vabackend.c:1032          nvCreateContext CUDA ERROR 'out of memory' (2)

[ffmpeg/video] vp9: Failed to create decode context: 2 (resource allocation failed).
[ffmpeg/video] vp9: Failed setup for format vaapi: hwaccel initialisation returned error.
     20802.367089748 [23262-23262] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x55a2728683d0)
     20802.367094786 [23262-23262] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     20802.367099545 [23262-23262] ../src/vabackend.c: 949        nvDestroySurfaces Destroying surface -1 (0x55a272867e40)
     20802.367104123 [23262-23262] ../src/export-buf.c: 359 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
[... cut ...]
     20802.368981441 [23262-23262] ../src/vabackend.c:2051              nvTerminate Terminating 0x55a2723509f0
     20802.368995471 [23262-23262] ../src/export-buf.c:  66      egl_releaseExporter Releasing exporter, 0 outstanding frames
     20802.369001006 [23262-23262] ../src/export-buf.c:  83      egl_releaseExporter Done releasing frames
     20802.369006569 [23262-23262] ../src/vabackend.c:2065              nvTerminate Now have 0 (0 max) instances
VO: [gpu] 1920x1080 yuv420p
V: 00:00:39 / 00:02:32 (26%)

Exiting... (Quit)

```
</details>

Let me try it with `v0.0.8` nvidia-vaapi-driver.

xtellaris, 2022-12-20:
> Let me try it with v0.0.8 nvidia-vaapi-driver.

A very long log, as it was decoding.
<details>
    <summary>mpv log</summary>
```sh-sesstion
$ NVD_LOG=1 mpv --no-config --start=10 --end=11 \
--hwdec=vaapi-copy --vaapi-device=/dev/dri/renderD129 --log-file=/tmp/mpv.log \
"/home/xtellaris/Videos/Win-Videos/Test/Mad Max - Fury Road - Official Retaliate Trailer [HD] [MonFNCgK4WE].webm"
 (+) Video --vid=1 (*) (vp9 1920x1080 23.976fps)
     21981.079456921 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x56122b1976c0 31
     21981.079470807 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
     21981.079493416 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1916       __vaDriverInit_1_0 Selecting EGL backend
     21981.079529490 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 161       findGPUIndexFromFd Looking for DRM device index: 1
     21981.080440524 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 175       findGPUIndexFromFd Found 4 EGL devices
     21981.080613104 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 184       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     21981.080621485 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 190       findGPUIndexFromFd Found drmDeviceIndex: 1
     21981.080644776 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 225       findGPUIndexFromFd Selecting EGLDevice 0
     21981.080654118 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 229       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 1
     21981.080660368 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 229       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 2
     21981.080665702 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 232       findGPUIndexFromFd No DRM device file for EGLDevice 3
     21981.080669930 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 235       findGPUIndexFromFd No match found, falling back to default device
     21981.081984143 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 289         egl_initExporter Driver supports 16-bit surfaces
     21981.129622561 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
     21981.158063502 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 19 with 0 attributes
     21981.158078556 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) (nil) 0
     21981.158102673 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) 0x56122b7aec40 7
     21981.159874986 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) (nil) 0
     21981.159884648 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) 0x56122b6b84c0 7
     21981.161636802 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b19fce0)
     21981.161646723 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1314            nvDeriveImage In nvDeriveImage
     21981.161671862 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b1a0160)
     21981.161677630 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6b8c90)
     21981.161683017 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6b9220)
     21981.161688244 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6b97b0)
     21981.161693010 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6b9d40)
     21981.161698184 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6ba2d0)
     21981.161702881 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6ba860)
     21981.161707936 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6badf0)
     21981.161713121 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6bb380)
     21981.161718202 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6bb910)
     21981.161723267 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 742        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56122b6bbea0)
     21981.188724204 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 19 with 0 attributes
[... cut ...]
     21982.336292884 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c: 340  egl_destroyBackingImage Destroying BackingImage: 0x7f37480068e0
     21982.336412813 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c:  65      egl_releaseExporter Releasing exporter, 0 outstanding frames
     21982.337487406 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/export-buf.c:  82      egl_releaseExporter Done releasing frames
     21982.337802787 [24202-24202] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances

Exiting... (End of file)

```

YidaozhanYa, 2022-12-25:
Now mpv with vaapi-copy works for me, but vlc and firefox still don't work. vlc reports `glconv_vaapi_x11 gl error: vaDeriveImage: operation failed`, Firefox reports `[GFX1-]: glxtest: VA-API test failed: failed to initialize VAAPI connection.` on start.
[vlc.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10299572/vlc.txt)

elFarto, 2022-12-26:
VLC doesn't work with this library as it required vaDeriveImage, which we don't support.
The Firefox issue is the normal Optimus issue.

closed 2025-08-07 by elFarto