elFarto/nvidia-vaapi-driver #142: Build fails with ERROR: C shared or static library 'EGL' not found

respectablenoob-source, 2022-12-17:
I am trying to build driver on Opensuse Tumbleweed and am facing this error. I have installed meson, gstreamer-plugins-bad, ffnvcodec, and gstreamer-plugins-vaapi from yast.

Here is the log:

`compiler stderr:

Running command: cc -E -dM -
-----
Detecting linker via: cc -Wl,--version
linker returned <Popen: returncode: 0 args: ['cc', '-Wl,--version']>
linker stdout:
GNU ld (GNU Binutils; openSUSE Tumbleweed) 2.39.0.20220810-1
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

linker stderr:
collect2 version 12.2.1 20221020 [revision 0aaef83351473e8f4eb774f8f999bbe87a4866d7]
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld -plugin /usr/lib64/gcc/x86_64-suse-linux/12/liblto_plugin.so -plugin-opt=/usr/lib64/gcc/x86_64-suse-linux/12/lto-wrapper -plugin-opt=-fresolution=/tmp/cc4PVmCi.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --build-id --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 /usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/crt1.o /usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/crti.o /usr/lib64/gcc/x86_64-suse-linux/12/crtbegin.o -L/usr/lib64/gcc/x86_64-suse-linux/12 -L/usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/lib -L/usr/lib64/gcc/x86_64-suse-linux/12/../../.. --version -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/lib64/gcc/x86_64-suse-linux/12/crtend.o /usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/crtn.o

Sanity testing C compiler: cc
Is cross compiler: False.
Sanity check compiler command line: cc sanitycheckc.c -o sanitycheckc.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/sanitycheckc.exe
C compiler for the host machine: cc (gcc 12.2.1 "cc (SUSE Linux) 12.2.1 20221020 [revision 0aaef83351473e8f4eb774f8f999bbe87a4866d7]")
C linker for the host machine: cc ld.bfd 2.39.0.20220810-1
-----
Detecting compiler via: cc --version
compiler returned <Popen: returncode: 0 args: ['cc', '--version']>
compiler stdout:
cc (SUSE Linux) 12.2.1 20221020 [revision 0aaef83351473e8f4eb774f8f999bbe87a4866d7]
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

compiler stderr:

Running command: cc -E -dM -
-----
Detecting linker via: cc -Wl,--version
linker returned <Popen: returncode: 0 args: ['cc', '-Wl,--version']>
linker stdout:
GNU ld (GNU Binutils; openSUSE Tumbleweed) 2.39.0.20220810-1
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

linker stderr:
collect2 version 12.2.1 20221020 [revision 0aaef83351473e8f4eb774f8f999bbe87a4866d7]
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld -plugin /usr/lib64/gcc/x86_64-suse-linux/12/liblto_plugin.so -plugin-opt=/usr/lib64/gcc/x86_64-suse-linux/12/lto-wrapper -plugin-opt=-fresolution=/tmp/cccGN6yw.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --build-id --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 /usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/crt1.o /usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/crti.o /usr/lib64/gcc/x86_64-suse-linux/12/crtbegin.o -L/usr/lib64/gcc/x86_64-suse-linux/12 -L/usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/lib -L/usr/lib64/gcc/x86_64-suse-linux/12/../../.. --version -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/lib64/gcc/x86_64-suse-linux/12/crtend.o /usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/crtn.o

Sanity testing C compiler: cc
Is cross compiler: False.
Sanity check compiler command line: cc sanitycheckc.c -o sanitycheckc.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/sanitycheckc.exe
C compiler for the build machine: cc (gcc 12.2.1 "cc (SUSE Linux) 12.2.1 20221020 [revision 0aaef83351473e8f4eb774f8f999bbe87a4866d7]")
C linker for the build machine: cc ld.bfd 2.39.0.20220810-1
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu: x86_64
Running compile:
Working directory:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmppryevdev
Command line:  cc /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmppryevdev/testfile.c -o /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmppryevdev/output.exe -D_FILE_OFFSET_BITS=64 -O0 -Wl,--start-group -lm -Wl,--end-group -Wl,--allow-shlib-undefined 

Code:
 int main(void) { return 0; }

Compiler stdout:
 
Compiler stderr:
 
Library m found: YES
Running compile:
Working directory:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpnkwziz4y
Command line:  cc /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpnkwziz4y/testfile.c -o /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpnkwziz4y/output.exe -D_FILE_OFFSET_BITS=64 -O0 -Wl,--start-group -ldl -Wl,--end-group -Wl,--allow-shlib-undefined 

Code:
 int main(void) { return 0; }

Compiler stdout:
 
Compiler stderr:
 
Library dl found: YES
Running compile:
Working directory:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpjdsal889
Command line:  cc /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpjdsal889/testfile.c -o /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpjdsal889/output.exe -D_FILE_OFFSET_BITS=64 -O0 -Wl,--start-group -lEGL -Wl,--end-group -Wl,--allow-shlib-undefined 

Code:
 int main(void) { return 0; }

Compiler stdout:
 
Compiler stderr:
 /usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: cannot find -lEGL: No such file or directory
collect2: error: ld returned 1 exit status

Running compile:
Working directory:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpu0i1ubd4
Command line:  cc /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpu0i1ubd4/testfile.c -o /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmpu0i1ubd4/output.exe -D_FILE_OFFSET_BITS=64 -O0 

Code:
 #include<stdio.h>
        
        int main(void) {
            printf("%ld\n", (long)(sizeof(void *)));
            return 0;
        }
Compiler stdout:
 
Compiler stderr:
 
Program stdout:

8

Program stderr:

Running compile:
Working directory:  /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmp2ha_08zg
Command line:  cc /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmp2ha_08zg/testfile.c -o /home/ali/nvidia-vaapi-driver-0.0.8/build/meson-private/tmp2ha_08zg/output.obj -c -D_FILE_OFFSET_BITS=64 -O0 --print-search-dirs 

Code:
 
Compiler stdout:
 install: /usr/lib64/gcc/x86_64-suse-linux/12/
programs: =/usr/lib64/gcc/x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/:/usr/lib64/gcc/x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/
libraries: =/usr/lib64/gcc/x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/lib/x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/lib/../lib64/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../x86_64-suse-linux/12/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../../lib64/:/lib/x86_64-suse-linux/12/:/lib/../lib64/:/usr/lib/x86_64-suse-linux/12/:/usr/lib/../lib64/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/lib/:/usr/lib64/gcc/x86_64-suse-linux/12/../../../:/lib/:/usr/lib/

Compiler stderr:
 

meson.build:18:0: ERROR: C shared or static library 'EGL' not found
`

System info:
64 bit, NVidia driver version:470.161.03

elFarto, 2022-12-18:
You'll also need to make sure you have the EGL devel library install. Not sure what it's called on Opensuse, but it's `libglvnd-devel` on Fedora.

respectablenoob-source, 2022-12-18:
> You'll also need to make sure you have the EGL devel library install. Not sure what it's called on Opensuse, but it's `libglvnd-devel` on Fedora.

Thank you, I fixed the EGL error but now I have a new build error:

`[1/13] Compiling C object nvidia_drv_video.so.p/src_av1.c.o
FAILED: nvidia_drv_video.so.p/src_av1.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_av1.c.o -MF nvidia_drv_video.so.p/src_av1.c.o.d -o nvidia_drv_video.so.p/src_av1.c.o -c ../src/av1.c
In file included from ../src/av1.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[2/13] Compiling C object nvidia_drv_video.so.p/src_h264.c.o
FAILED: nvidia_drv_video.so.p/src_h264.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_h264.c.o -MF nvidia_drv_video.so.p/src_h264.c.o.d -o nvidia_drv_video.so.p/src_h264.c.o -c ../src/h264.c
In file included from ../src/h264.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[3/13] Compiling C object nvidia_drv_video.so.p/src_export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_export-buf.c.o -MF nvidia_drv_video.so.p/src_export-buf.c.o.d -o nvidia_drv_video.so.p/src_export-buf.c.o -c ../src/export-buf.c
In file included from ../src/export-buf.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[4/13] Compiling C object nvidia_drv_video.so.p/src_vp8.c.o
FAILED: nvidia_drv_video.so.p/src_vp8.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_vp8.c.o -MF nvidia_drv_video.so.p/src_vp8.c.o.d -o nvidia_drv_video.so.p/src_vp8.c.o -c ../src/vp8.c
In file included from ../src/vp8.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[5/13] Compiling C object nvidia_drv_video.so.p/src_vc1.c.o
FAILED: nvidia_drv_video.so.p/src_vc1.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_vc1.c.o -MF nvidia_drv_video.so.p/src_vc1.c.o.d -o nvidia_drv_video.so.p/src_vc1.c.o -c ../src/vc1.c
In file included from ../src/vc1.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[6/13] Compiling C object nvidia_drv_video.so.p/src_mpeg4.c.o
FAILED: nvidia_drv_video.so.p/src_mpeg4.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_mpeg4.c.o -MF nvidia_drv_video.so.p/src_mpeg4.c.o.d -o nvidia_drv_video.so.p/src_mpeg4.c.o -c ../src/mpeg4.c
In file included from ../src/mpeg4.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[7/13] Compiling C object nvidia_drv_video.so.p/src_mpeg2.c.o
FAILED: nvidia_drv_video.so.p/src_mpeg2.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_mpeg2.c.o -MF nvidia_drv_video.so.p/src_mpeg2.c.o.d -o nvidia_drv_video.so.p/src_mpeg2.c.o -c ../src/mpeg2.c
In file included from ../src/mpeg2.c:1:
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[8/13] Compiling C object nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o
FAILED: nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o -MF nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o.d -o nvidia_drv_video.so.p/src_direct_direct-export-buf.c.o -c ../src/direct/direct-export-buf.c
In file included from ../src/direct/direct-export-buf.c:3:
../src/direct/../vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[9/13] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
FAILED: nvidia_drv_video.so.p/src_vabackend.c.o 
[... 10 lines cut ...]
../src/vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
[... cut ...]
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -std=c11 -O0 -g -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_nv-driver.c.o -MF nvidia_drv_video.so.p/src_direct_nv-driver.c.o.d -o nvidia_drv_video.so.p/src_direct_nv-driver.c.o -c ../src/direct/nv-driver.c
In file included from ../src/direct/nv-driver.c:19:
../src/direct/../vabackend.h:6:10: fatal error: EGL/egl.h: No such file or directory
    6 | #include <EGL/egl.h>
      |          ^~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Could not rebuild /home/ali/nvidia-vaapi-driver-0.0.8/build
`

What could be causing this?

elFarto, 2022-12-18:
That's looking for the header files for libEGL. I thought they should have been in the -devel package you installed.

After a quick search around on pkgs.org, `Mesa-libEGL-devel` is the package that has the EGL header files.

respectablenoob-source, 2022-12-18:
> That's looking for the header files for libEGL. I thought they should have been in the -devel package you installed.
> 
> After a quick search around on pkgs.org, `Mesa-libEGL-devel` is the package that has the EGL header files.

This issue has been solved, thanks.

closed 2022-12-18 by respectablenoob-source