elFarto/nvidia-vaapi-driver #143: Availability Request: 0.0.8 on RPMFusion

cprin21, 2022-12-18:
Version 0.0.7 is the latest version uploaded to the RPMFusion Nonfree repository, barring access to hardware acceleration for Fedora 37 users on Nvidia 525.

elFarto, 2022-12-18:
I'm not the person who packages it for RPMFusion. You'd need to talk to the maintainer, @kwizart.

closed 2022-12-18 by elFarto

kwizart, 2022-12-18:
Thanks for the report and mention.

FYI, the package for Fedora, RHEL and derivatives is located at https://github.com/rpmfusion/nvidia-vaapi-driver (github mirror)
And the preferred method to report issue is to use bugzilla.rpmfusion.org

cprin21, 2022-12-18:
Thank you for your responses, I was unaware as the online browser of the repos didn't mention a listed maintainer and as such I defaulted to this github. Rightfully closed, glad to see a conclusive response!