elFarto/nvidia-vaapi-driver #145: Not works on GTX 1660S

skbeh, 2022-12-20:
There is zero usage of `decoder` and the CPU usage is relatively high when playing.
Test video: https://www.youtube.com/watch?v=9JlLeYk4U_c, 1440p.
I am using latest Arch Linux and GTX 1660S 6G. The GPU memory usage is around 20%.
Seted environment variables:
```
LIBVA_DRIVER_NAME=nvidia
NVD_BACKEND=direct
MOZ_DISABLE_RDD_SANDBOX=1
```

<details><summary>Firefox log</summary>

```
3279.616143224 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0cf722ed40 31
3279.616150064 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
3279.616153154 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
3279.622185022 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
3279.622195212 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 2600 1 2 6
3279.622269695 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
3279.668949282 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
3279.686946014 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 6 with 0 attributes
3279.686955284 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
3279.686958214 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f0cd99a3d00 7
3279.688280595 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 7 with 0 attributes
3279.688285865 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
3279.688288525 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f0cd99a3880 7
3279.689529324 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 13 with 0 attributes
3279.689534434 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
3279.689536874 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f0cd99a3280 7
3279.690683990 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 14 with 0 attributes
3279.690689720 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 9 (8) (nil) 0
3279.690692360 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 9 (8) 0x7f0cd99a3700 7
3279.691841476 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 19 with 0 attributes
3279.691846986 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10(8) (nil) 0
3279.691849487 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10(8) 0x7f0cd99a3e80 7
3279.692996302 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 21 with 0 attributes
3279.693001493 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10(8) (nil) 0
3279.693004193 [18759-18772] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10(8) 0x7f0cd99a3280 7
3280.706972199 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0cf722ef20 31
3280.706982880 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 1 (0 max) instances
3280.706986380 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
3280.707700342 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
3280.707706762 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 2600 1 2 6
3280.707786575 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
3280.744839972 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
3280.745003217 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0cf722ef20
3280.745145781 [18759-18806] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 1 (0 max) instances
3280.768125189 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0cf722ef20 31
3280.768134109 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 1 (0 max) instances
3280.768135929 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
3280.768776639 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
3280.768783469 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 2600 1 2 6
3280.768868272 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
3280.803362759 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
3280.803532995 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0cf722ef20
3280.803761622 [18759-18825] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 1 (0 max) instances
[2022-12-20T03:10:57Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-20T03:10:57Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-20T03:10:57Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2022-12-20T03:10:57Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
3282.020914062 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0cf722ef20 31
3282.020923322 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 1 (0 max) instances
3282.020926773 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
3282.021824711 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
3282.021830881 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 2600 1 2 6
3282.021941934 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
3282.060047215 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
3282.060212170 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0cf722ef20
3282.060374965 [18759-18858] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 1 (0 max) instances
3282.391534024 [18759-18897] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0cf722ef20 31
[... 225 lines cut ...]
3913.770174804 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0cbc52d8e0)
3913.770176894 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0cbc52db60)
3913.770179104 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0cbc52dac0)
3913.770181294 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0cbc52d3e0)
[... cut ...]
3913.770212475 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0cbc518ce0)
3913.770483185 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0cf722ef20
3913.770570198 [18759-20022] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 1 (0 max) instances
```

</details>

`about:support`: https://paste.debian.net/hidden/da5db96a/

`vainfo -a`: https://paste.debian.net/hidden/0af21b58/

elFarto, 2022-12-20:
That's very odd. What video did you watch before that one you posted? The one that had the resolution 1600x1504. Did an advert play or something? That's a very odd resolution for Youtube to have and it wanted a large amount of surfaces allocated.

skbeh, 2022-12-20:
There was no advert because I am using an ad blocker

elFarto, 2022-12-20:
So any idea what the video it tried to play about 10 minutes before the one you posted? I'm interested because that seems to have been rejected by NVDEC, which I've not really seen before.

skbeh, 2022-12-20:
I'm not sure. Maybe they are some embedded videos in YouTube homepage.

skbeh, 2022-12-21:
When using latest build from master branch, the decoder is being used and the CPU consuming is much less. It seems that the problem is solved.

closed 2022-12-21 by skbeh