elFarto/nvidia-vaapi-driver #146: VRAM Leak on direct backend

scorpion-26, 2022-12-20:
Reproduction: 
- Open stream on twitch on firefox
- Repeatedly pause, unfocus twitch tab, refocus tab and restart the stream

After the unfocus ~10 MB are freed, though when restarting ~80 MB are allocated again.
When closing the tab, 300 MB is freed, though previously leaked memory(Which can be in th GB range) still persists. 
Only when the browser is closed, or RDD Process is killed, the VRAM gets freed

Oddly enough, nvidia-smi reports the high total VRAM usage, though the applications list only a fraction of that when combined. 
Trying to utilize the ghost memory when VRAM is almost full with vulkan results in an error, so the total reading is probably accurate

nvidia-smi, sorry for bad formatting 
`Tue Dec 20 22:16:24 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.11    Driver Version: 525.60.11    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:06:00.0 Off |                  N/A |
|  0%   54C    P2    35W / 200W |   3214MiB /  8192MiB |     10%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A    293789      G   /usr/local/bin/Hyprland          1058MiB |
|    0   N/A  N/A    379223      G   kitty                              87MiB |
|    0   N/A  N/A    873979      G   kitty                              51MiB |
|    0   N/A  N/A    941383      G   kitty                             155MiB |
|    0   N/A  N/A   1012775      G   firefox                           196MiB |
|    0   N/A  N/A   2304944    C+G   ...local/lib/firefox/firefox      278MiB |
|    0   N/A  N/A   2874510      G   ...RendererForSitePerProcess        7MiB |
|    0   N/A  N/A   2974832      G   /usr/lib/Xorg                      32MiB |
|    0   N/A  N/A   3904467      G   kitty                              21MiB |
|    0   N/A  N/A   3905865      G   ...RendererForSitePerProcess        8MiB |
|    0   N/A  N/A   3905872      G   Xwayland                            3MiB |
+-----------------------------------------------------------------------------+`

elFarto, 2022-12-21:
Thanks for reporting that. The is a known issue with memory allocated by the direct backend not being accounted for in nvidia-smi, I'm not sure what I have to do to get that to show up.

I'm not seeing such a large leak, if any on my machine. Playing then pausing the video and switching to another tab causes the total memory usage to return to the pre-playing usage after a few seconds.

I'll see if I can get the memory to actually show properly in nvidia-smi, that might help spot any issue better.

scorpion-26, 2022-12-21:
I noticed, that I was still on the latest commit of the direct-backend branch. 
Switching to latest master fixed the issue. 

Sorry for the inconvience

closed 2022-12-21 by scorpion-26