elFarto/nvidia-vaapi-driver #147: Typo in README

rkoot, 2022-12-24:
I assume you meant to say from Maxwell onward? That would make it >= Maxwell  instead of <=

Maybe a link to wikipedia's or nvidia's nvenc matrix so people can easily check what their gpu should be capable of?
https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new
https://en.wikipedia.org/wiki/Nvidia_NVDEC
Just a thought.

crimist, 2022-12-25:
When the table was first created @elFarto was unsure whether the direct backend was compatible with all architectures. At this point it seems that it likely works on all cards supporting this driver so perhaps we should remove it?

crimist, 2022-12-26:
Ah and to answer your question, no this is intentional. I didn't want to create a very large table and I'm unsure of the minimum supported architecture.

rkoot, 2022-12-26:
A link would suffice, don't you think? And the ' Generational compatibility' table now tells you that *all* gpus, predating Maxwell work, which they don't. from Maxwell up it's supported, so >= Maxwell instead of <=Maxwell

elFarto, 2022-12-26:
Yeah, that's sort of my fault. I added in the <= Maxwell one in there as I've tested it on my 760 which is a Kepler card, without thinking too much what that made the table imply. I'll see if I can clean it up a bit.

rkoot, 2023-01-05:
Maybe a useful addition to the README:
Whenever vaapi gets triggered to do its stuff, your syslog will get spammed with a few lines like this

libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0

Ie, if you go to youtube, the generation of the thumbnails results in a whole slew of repeated lines like above.

Solution: to keep libva from spamming your syslog, create a file named /etc/libva.conf with contents LIBVA_MESSAGING_LEVEL=1
Now libva keeps quiet.
Thought this was handy to maybe mention.

elFarto, 2023-01-07:
I've updated the README to better explain the compatibility of the direct backend, and added the LIBVA_MESSAGING_LEVEL information.

closed 2023-01-07 by elFarto