elFarto/nvidia-vaapi-driver #148: Mpv 10-bit media failing with master @ 31d4a34, but okay @ bb7f6fc

marioroy, 2022-12-27:
NVIDIA proprietary driver v520.61.05. The youtube-play-vaapi script can be found [here](https://github.com/elFarto/nvidia-vaapi-driver/issues/141#issuecomment-1359308861).

```bash
$ ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699
[ffmpeg/video] av1: Hardware does not support image size 1920x1080 (constraints: width 128-8192 height 1881003120-8192).
[ffmpeg/video] av1: Failed setup for format vaapi: hwaccel initialisation returned error.
[ffmpeg/video] av1: Your platform doesn't suppport hardware accelerated AV1 decoding.
[ffmpeg/video] av1: Failed to get pixel format.
Error while decoding frame (hardware decoding)!
```

Strange is the constraints message above. This occurs randomly.
```text
[ffmpeg/video] av1: Hardware does not support image size 1920x1080 (constraints: width 128-8192 height 1881003120-8192).
```

Often, no constraints message or errors, but green playback.

[image]

No issues with master @ bb7f6fc, the commit after the 0.0.8 release.

<details><summary>nvd.log</summary>
<p>
```text
$ NVD_LOG=1 ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699
 (+) Video --vid=1 (*) (av1 1920x1080 23.976fps)
 (+) Audio --aid=1 --alang=eng (*) 'videoplayback?expire=1672156963&ei=w8KqY86JIIenlu8PrqOPoAI&ip=24.175.72.178&id=o-AMIwyim7B9s4GRyzYcZEtwldob_WimLv_uNWiuxHUDwn&itag=251&source=youtube&requiressl=yes&mh=ut&mm=31%2C29&mn=sn-q4flrnsk%2Csn-q4fl6nlz&ms=au%2Crdu&mv=m&mvi=4&pl=17&initcwndbps=2088750&spc=zIddbA4kLg7NdNaa-nrz02R0nlirY_k&vprv=1&svpuc=1&mime=audio%2Fwebm&gir=yes&clen=4401290&dur=244.281&lmt=1663500126935604&mt=1672135052&fvip=2&keepalive=yes&fexp=24001373%2C24007246&c=ANDROID&txp=4432434&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cspc%2Cvprv%2Csvpuc%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&sig=AOq0QJ8wRgIhAK699i_xEjX4BK1K_A9JFGWAFgTruzAiuZqenYP7V0gEAiEAk7KH0jkY2HhxUuX7j9jys6vMzbsrSryNKefFC-jBL9I%3D&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRAIgFTF9byhyrRHT4qo89mKL5ZR_JFXfPTNJptFXvpJZvn4CIBOixUlfplGf9XB_lgT6DCuh0mAcaxxknvsJotbdO5cM' (opus 2ch 48000Hz) (external)
[vo/xv] Warning: this legacy VO has bad quality and performance, and will in particular result in blurry OSD and subtitles. You should fix your graphics drivers, or not force the xv VO.
       901.595402299 [14052-14052] ../src/vabackend.c:2079       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       901.595411559 [14052-14052] ../src/vabackend.c:2088       __vaDriverInit_1_0 Now have 0 (0 max) instances
       901.595416410 [14052-14052] ../src/vabackend.c:2111       __vaDriverInit_1_0 Selecting EGL backend
       901.597810336 [14052-14052] ../src/export-buf.c: 152       findGPUIndexFromFd Looking for DRM device index: 0
       901.598625128 [14052-14052] ../src/export-buf.c: 166       findGPUIndexFromFd Found 1 EGL devices
       901.598688149 [14052-14052] ../src/export-buf.c: 175       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       901.598692059 [14052-14052] ../src/export-buf.c: 181       findGPUIndexFromFd Found drmDeviceIndex: 0
       901.598693829 [14052-14052] ../src/export-buf.c: 213       findGPUIndexFromFd Selecting EGLDevice 0
       901.599383679 [14052-14052] ../src/export-buf.c: 284         egl_initExporter Driver supports 16-bit surfaces
       901.653480453 [14052-14052] ../src/vabackend.c:1393      nvQueryImageFormats In nvQueryImageFormats
       901.668517149 [14052-14052] ../src/vabackend.c: 637           nvCreateConfig got profile: 32 with 0 attributes
       901.668526600 [14052-14052] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 11 (8) (nil) 0
       901.668529140 [14052-14052] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 11 (8) 0x56002ea3a140 7
       901.669410453 [14052-14052] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 11 (8) (nil) 0
       901.669415993 [14052-14052] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 11 (8) 0x56002ea3c700 7
       901.670270616 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e944100)
       901.670277166 [14052-14052] ../src/vabackend.c:1487            nvDeriveImage In nvDeriveImage
       901.670281636 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e944900)
       901.670284346 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e944ea0)
       901.670286786 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e945440)
       901.670289166 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e9459e0)
       901.670291646 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e945f80)
       901.670295156 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e946520)
       901.670297996 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e946ac0)
       901.670301356 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e947060)
       901.670304906 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e947600)
       901.670307426 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e947ba0)
       901.670310537 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e948140)
       901.670312347 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e9486e0)
       901.670315257 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e948c80)
       901.670318357 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e949220)
       901.670321587 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e9497c0)
       901.670324247 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e949d60)
       901.670327687 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e94a320)
       901.670329747 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e94a8c0)
       901.670332857 [14052-14052] ../src/vabackend.c: 915        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x56002e94ae60)
       901.683235321 [14052-14052] ../src/vabackend.c: 637           nvCreateConfig got profile: 32 with 0 attributes
       901.683241761 [14052-14052] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 11 (8) (nil) 0
       901.683244361 [14052-14052] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 11 (8) 0x56002e949880 7
       901.684060453 [14052-14052] ../src/vabackend.c: 979          nvCreateContext with 20 render targets, 20 surfaces, at 1920x1080
       901.691419514 [14052-14069] ../src/vabackend.c: 378          resolveSurfaces [RT] Resolve thread for 0x56002e9adb00 started
[... 109 lines cut ...]
       901.891748829 [14052-14052] ../src/vabackend.c:1429            nvCreateImage created image id: 208
       901.892309827 [14052-14069] ../src/vabackend.c: 411          resolveSurfaces Mapped surface 1 to 205200000 (2048)
       901.892326897 [14052-14069] ../src/vabackend.c: 415          resolveSurfaces Surface 1 exported
       901.933360165 [14052-14052] ../src/vabackend.c:1295             nvEndPicture Decoded frame successfully to idx: 2 (0x56002e94a320)
       901.933370605 [14052-14069] ../src/vabackend.c: 391          resolveSurfaces Reading from queue: 10 11
       901.933373515 [14052-14069] ../src/vabackend.c: 407          resolveSurfaces Mapping surface 2
       901.933377405 [14052-14052] ../src/vabackend.c:1429            nvCreateImage created image id: 227
       901.933936874 [14052-14069] ../src/vabackend.c: 411          resolveSurfaces Mapped surface 2 to 205200000 (2048)
       901.933952424 [14052-14069] ../src/vabackend.c: 415          resolveSurfaces Surface 2 exported
[... cut ...]
       902.016676658 [14052-14069] ../src/vabackend.c: 407          resolveSurfaces Mapping surface 4
       902.016678908 [14052-14052] ../src/vabackend.c:1429            nvCreateImage created image id: 265
       902.017311687 [14052-14069] ../src/vabackend.c: 411          resolveSurfaces Mapped surface 4 to 205200000 (2048)
       902.017327838 [14052-14069] ../src/vabackend.c: 415          resolveSurfaces Surface 4 exported
       902.058326005 [14052-14052] ../src/vabackend.c:1295             nvEndPicture Decoded frame successfully to idx: 5 (0x56002e949220)
```
</p>
</details>

marioroy, 2022-12-27:
The above was playing AV1 media. I tried VP9.2. There is no hardware acceleration for VP9.2 10-bit media using master @ 31d4a34, but okay @ bb7f6fc.

```bash
$ NVD_LOG=1 ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 335
 (+) Video --vid=1 (*) (vp9 1920x1080 23.976fps)
 (+) Audio --aid=1 --alang=eng (*) 'videoplayback?expire=1672159700&ei=dM2qY8_tHeunir4PvcCP2AU&ip=24.175.72.178&id=o-AOQuXxn09MF9jpCGYU-37xQKHU9bzUfLSqc4w_hDKGqZ&itag=251&source=youtube&requiressl=yes&mh=ut&mm=31%2C26&mn=sn-q4flrnsk%2Csn-a5msen7s&ms=au%2Conr&mv=m&mvi=4&pl=17&initcwndbps=2058750&spc=zIddbASGj5tJyBjQTTMxsbAjmYWxAkM&vprv=1&svpuc=1&mime=audio%2Fwebm&gir=yes&clen=4401290&dur=244.281&lmt=1663500126935604&mt=1672137677&fvip=4&keepalive=yes&fexp=24001373%2C24007246&c=ANDROID&txp=4432434&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cspc%2Cvprv%2Csvpuc%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&sig=AOq0QJ8wRgIhAKUxq1RO8EppBaq7WmvXwB7wGTE6NII2WrCmljwxasbIAiEA1GSwS2aZe1wVycrP0Z0ZxWVbcI5o8T1OUIFxm5y_biE%3D&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRQIgQW2PyWpSM7t7b7yXaYk-wSIUamHqGfPy201Cn95voZkCIQC9H05Y7tD50SVgZp9d2A9H12b4RDArN8L_igmuxxNUlA%3D%3D' (opus 2ch 48000Hz) (external)
[vo/xv] Warning: this legacy VO has bad quality and performance, and will in particular result in blurry OSD and subtitles. You should fix your graphics drivers, or not force the xv VO.
      3638.789017340 [16260-16260] ../src/vabackend.c:2079       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      3638.789026710 [16260-16260] ../src/vabackend.c:2088       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3638.789031420 [16260-16260] ../src/vabackend.c:2111       __vaDriverInit_1_0 Selecting EGL backend
      3638.791408625 [16260-16260] ../src/export-buf.c: 152       findGPUIndexFromFd Looking for DRM device index: 0
      3638.792229368 [16260-16260] ../src/export-buf.c: 166       findGPUIndexFromFd Found 1 EGL devices
      3638.792292889 [16260-16260] ../src/export-buf.c: 175       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
      3638.792297029 [16260-16260] ../src/export-buf.c: 181       findGPUIndexFromFd Found drmDeviceIndex: 0
      3638.792298729 [16260-16260] ../src/export-buf.c: 213       findGPUIndexFromFd Selecting EGLDevice 0
      3638.792972889 [16260-16260] ../src/export-buf.c: 284         egl_initExporter Driver supports 16-bit surfaces
      3638.836537625 [16260-16260] ../src/vabackend.c:1393      nvQueryImageFormats In nvQueryImageFormats
      3638.849635220 [16260-16260] ../src/vabackend.c: 637           nvCreateConfig got profile: 21 with 0 attributes
      3638.849644290 [16260-16260] ../src/vabackend.c:1747 nvQuerySurfaceAttributes with 10 (10) (nil) 0
      3638.849647210 [16260-16260] ../src/vabackend.c:1763 nvQuerySurfaceAttributes 16 bits surfaces not supported: 1
[ffmpeg] AVHWDeviceContext: Failed to query surface attributes: 4 (invalid VAConfigID).
[ffmpeg/video] vp9: Failed setup for format vaapi: hwaccel initialisation returned error.
      3638.850903888 [16260-16260] ../src/vabackend.c:2051              nvTerminate Terminating 0x5630ec3dcce0
      3638.850909378 [16260-16260] ../src/export-buf.c:  66      egl_releaseExporter Releasing exporter, 0 outstanding frames
      3638.850911379 [16260-16260] ../src/export-buf.c:  83      egl_releaseExporter Done releasing frames
      3638.850914819 [16260-16260] ../src/vabackend.c:2065              nvTerminate Now have 0 (0 max) instances
[autoconvert] Converting yuv420p10 -> yuv420p
AO: [pulse] 48000Hz stereo 2ch float
[autoconvert] Converting yuv420p10 -> yuv420p
VO: [xv] 1920x1080 yuv420p
AV: 00:00:04 / 00:04:04 (2%) A-V:  0.000 Cache: 6.1s/1MB

Exiting... (Quit)
```

marioroy, 2022-12-27:
There are no issues playing VP9 8-bit media with master @ 31d4a34.

elFarto, 2022-12-27:
That constraints bug is odd. I've checked both this library's code and ffmpegs and I can't see any issue. It looks like an uninitalised value, but that shouldn't happen. I've added some extra logging to see exactly what we're returning.

The second issue should be fixed. Looks like there was a mix up with the YUV444 code that made it require YUV444 support for just 16-bit surfaces.

marioroy, 2022-12-27:
FFmpeg version 5.1.git-3ab11dc; nvidia-vaapi-driver @ 10a7a2d

```bash
$ ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699
```
Now using hardware acceleration.  Though, green playback like the prior screenshot.
[https://termbin.com/wq3l](https://termbin.com/wq3l)

elFarto, 2022-12-27:
I think I see the issue. There's a slight mismatch between VA-API and NVDEC regarding VP9 and AV1 10 and 12-bit codecs. I'll probably have to take a good look at the whole VAConfig code to see if we can defer all the surface/chroma/bit-depth until the last possible moment.

I've added some logging to see if this is what's occurring.

marioroy, 2022-12-27:
FFmpeg version 5.1.git-3ab11dc; nvidia-vaapi-driver @ 2938a59.

```text
$ NVD_LOG=1 ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699
...
551.092723363 [4277-4277] ../src/vabackend.c: 700           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
...
551.106511030 [4277-4277] ../src/vabackend.c: 700           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
...
```

[https://termbin.com/fbhcq](https://termbin.com/fbhcq)

marioroy, 2022-12-30:
@thesword53 @elFarto  If helpful to know which commit, 10-bit playback via Mpv has been broken since @ 7113c20.

thesword53, 2023-01-03:
> FFmpeg version 5.1.git-3ab11dc; nvidia-vaapi-driver @ [10a7a2d](https://github.com/elFarto/nvidia-vaapi-driver/commit/10a7a2d1beeb3f3ea215dc1793800bbed9b44a12)
> 
> VP9.2 10-bit playback: now okay and HW accelerated.
> 
> ```shell
> $ ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 335
> ```
> 
> AV1 10-bit playback: green playback like the prior screenshot, but HW accelerated. Okay @ [bb7f6fc](https://github.com/elFarto/nvidia-vaapi-driver/commit/bb7f6fcb66e595647bc3c1bc5a4592479c5d2e0b).
> 
> ```shell
> $ NVD_LOG=1 ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699
> ```

It's maybe related to 7113c20184e32abeb6cb3857392a1549bbcd7a5a. I only have a RTX 2080 SUPER and a GTX 1060 for testing and both doesn't support AV1 decoding.

According to your logs, the driver uses NV12 surface instead of P010 for the AV1 10 bits video.

marioroy, 2023-01-03:
> It's maybe related to `7113c20`.

Yes, AV1 10 bits media played well before the `7113c20` commit.

elFarto, 2023-01-03:
I did have a look into this, and @thesword53 is correct. Since the base AV1 profile can be 8-bit or 10-bit we need some way to distinguish which one to use. NVDEC requires this choice at context creation, where-as VA-API doesn't.

I wasn't fully able to determine why FFmpeg was picking NV12 instead of P010. @philipl any ideas?

philipl, 2023-01-03:
In https://github.com/elFarto/nvidia-vaapi-driver/commit/90413a73cc98178f49a35ced8f03307f6ea322ab there are two heuristics for picking the decoder configuration. I did not spend any time looking at how ffmpeg or mpv behave because you should just use nvdec directly (and that's my statement here too, BTW). It's possible we need a third heuristic to handle that case if we care to handle it.

philipl, 2023-01-03:
At https://github.com/elFarto/nvidia-vaapi-driver/commit/90413a73cc98178f49a35ced8f03307f6ea322ab#diff-3dde7031705b1f52ddb04a226565772944b957892c5a2c6a546b56a199c4d8adR617 it only looks at the first attribute for the RTFormat. Maybe it comes from ffmpeg as a subsequent attribute?

marioroy, 2023-01-04:
Using master, AV1 10-bit playback does not enter the if statement.
```c
    if (drv->supports16BitSurface) {
        switch(cfg->profile) {
        case VAProfileHEVCMain10:
            cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
            cfg->bitDepth = 10;
            break;
        case VAProfileHEVCMain12:
            cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
            cfg->bitDepth = 12;
            break;
        case VAProfileVP9Profile2:
        case VAProfileAV1Profile0:
printf("here1\n");
fflush(stdout); 
            // If the user provides an RTFormat, we can use that to identify which decoder
            // configuration is appropriate. If a format is not required here, the caller
            // must pass render targets to createContext so we can use those to establish
            // the surface format and bit depth.
            if (num_attribs > 0 && attrib_list[0].type == VAConfigAttribRTFormat) {
printf("here2 %ld attrib_list[0].value\n", attrib_list[0].value);
printf("here2 %ld va_rt_format_yuv420_12\n", VA_RT_FORMAT_YUV420_12);
printf("here2 %ld va_rt_format_yuv420_10\n", VA_RT_FORMAT_YUV420_10);
fflush(stdout);
                switch(attrib_list[0].value) {
                case VA_RT_FORMAT_YUV420_12:
                    cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
                    cfg->bitDepth = 12;
                    break;
                case VA_RT_FORMAT_YUV420_10:
                    cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
                    cfg->bitDepth = 10;
                    break;
                default:
                    break; 
                }
            } else { 
printf("here3\n");
fflush(stdout);
                if (cfg->profile == VAProfileVP9Profile2) {  
                    cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
                    cfg->bitDepth = 10;
                } else {
                    LOG("Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.");
                }
            }
        default:
            break;
        }
    }
```

Recent yt-dlp errors using the youtube-play-vaapi script. I updated the [script](https://github.com/elFarto/nvidia-vaapi-driver/issues/141#issuecomment-1359308861).
```bash
# Fetch the title and audio+video URLs into an array.
# title: ${lines[0]}, audio: ${lines[1]}, video: ${lines[2]}

readarray -t lines < <(
    ${youtube_dl} --get-title -g -S "vcodec:avc1" \
        -f "bestaudio+${video_format}" "$url"
)
```

Running Mpv.
```text
$ ./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699
 (+) Video --vid=1 (*) (av1 1920x1080 23.976fps)
 (+) Audio --aid=1 --alang=eng (*) 'videoplayback?expire=1672830788&ei=5Aq1Y-62JcWjir4P3oqkkAE&ip=24.175.72.178&id=o-AF-eVz2SGjhvQpIkeu8A-fzUOp2J_-C87blNNyTBFHPL&itag=251&source=youtube&requiressl=yes&mh=ut&mm=31%2C29&mn=sn-q4fl6nlz%2Csn-q4flrnsk&ms=au%2Crdu&mv=m&mvi=2&pl=17&initcwndbps=1956250&spc=zIddbPPH2UibJoZDnNnEwRi_RC3w6as&vprv=1&svpuc=1&mime=audio%2Fwebm&gir=yes&clen=4401290&dur=244.281&lmt=1663500126935604&mt=1672808939&fvip=4&keepalive=yes&fexp=24007246&c=ANDROID&txp=4432434&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cspc%2Cvprv%2Csvpuc%2Cmime%2Cgir%2Cclen%2Cdur%2Clmt&sig=AOq0QJ8wRQIhAPby7Kg-IYwmY_blcLlTwZaRbLCk5qTKQa6v85imioTUAiBI_t1dd_TZzP9V3XWV58LensnAZcZ_unzRPVhi4TSwCg%3D%3D&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRAIgBcAJMSxFB3q0KrcB1gfCvNndDmD8_Uuo3wFvpONY8AkCIB3rziEbH_YZzS-wdwOPRUY6QteKpHbFxnjMorMpjbhl' (opus 2ch 48000Hz) (external)
[vo/xv] Warning: this legacy VO has bad quality and performance, and will in particular result in blurry OSD and subtitles. You should fix your graphics drivers, or not force the xv VO.
here1
here3
here1
here3
Using hardware decoding (vaapi-copy).
[autoconvert] Converting nv12 -> yuv420p
AO: [pulse] 48000Hz stereo 2ch float
[autoconvert] Converting nv12 -> yuv420p
VO: [xv] 1920x1080 yuv420p
AV: 00:00:08 / 00:04:04 (3%) A-V:  0.000 Cache: 4.5s/1MB

Exiting... (Quit)
```

philipl, 2023-01-04:
Doesn't that proposed change break 8bit playback?

marioroy, 2023-01-04:
> Doesn't that proposed change break 8bit playback?

Negative :) 8-bit playback is working. I tested AV1 8-bit media in Firefox.

The check for `cfg->profile` is done above via switch statement. Therefore, the following diff.
```diff
--- vabackend.c.orig	2023-01-03 23:34:32.957224573 -0600
+++ vabackend.c	2023-01-03 23:44:09.894484204 -0600
@@ -693,12 +693,8 @@
                     break;
                 }
             } else {
-                if (cfg->profile == VAProfileVP9Profile2) {
-                    cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
-                    cfg->bitDepth = 10;
-                } else {
-                    LOG("Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.");
-                }
+                cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
+                cfg->bitDepth = 10;
             }
         default:
             break;
```

marioroy, 2023-01-04:
The following is passing. Here the code in entirety.
```c
    if (drv->supports16BitSurface) {
        switch(cfg->profile) {
        case VAProfileHEVCMain10:
            cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
            cfg->bitDepth = 10;
            break;
        case VAProfileHEVCMain12:
            cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
            cfg->bitDepth = 12;
            break;
        case VAProfileVP9Profile2:
        case VAProfileAV1Profile0:
            // If the user provides an RTFormat, we can use that to identify which decoder
            // configuration is appropriate. If a format is not required here, the caller
            // must pass render targets to createContext so we can use those to establish
            // the surface format and bit depth.
            if (num_attribs > 0 && attrib_list[0].type == VAConfigAttribRTFormat) {
                switch(attrib_list[0].value) {
                case VA_RT_FORMAT_YUV420_12:
                    cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
                    cfg->bitDepth = 12;
                    break;
                case VA_RT_FORMAT_YUV420_10:
                    cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
                    cfg->bitDepth = 10;
                    break;
                default:
                    break;
                }
            } else {
                cfg->surfaceFormat = cudaVideoSurfaceFormat_P016;
                cfg->bitDepth = 10;
            }
        default:
            break;
        }
    }
```

marioroy, 2023-01-04:
> Doesn't that proposed change break 8bit playback?

Yes. I found a video with av1 8-bit format (Big Buck Bunny).
```bash
./youtube-play-vaapi https://youtu.be/aqz-KE-bpKQ 399
```

marioroy, 2023-01-04:
The following tests are passing using the `av1-10-bit-fix` branch.

```bash
./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699  # av1 10-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting p010 -> yuv420p

./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 335  # vp9.2 10-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting p010 -> yuv420p

./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 248  # vp9 8-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting nv12 -> yuv420p

./youtube-play-vaapi https://youtu.be/aqz-KE-bpKQ 399  # av1 8-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting nv12 -> yuv420p
```

anonish, 2023-01-04:
+1 confirm branch av1-10-bit-fix working on 3090ti (with av1 hw accel)

thesword53, 2023-01-04:
> @thesword53 Can you remember the exact reason for that change?

I tried to add more constraint to supported formats as the intel media driver does in [`MediaLibvaCapsG12::QuerySurfaceAttributes`](https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L1274).

elFarto, 2023-01-06:
I've tided up the fix and committed it to master. Ideally we'd have some better logic there, but I don't think we have the information to be able to provide an exact list at the moment.

marioroy, 2023-01-06:
I verified that master [681f12f](https://github.com/elFarto/nvidia-vaapi-driver/commit/681f12f3d588a1d98189cc11f109a53b99320544) is passing.
```bash
./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 699  # av1 10-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting p010 -> yuv420p

./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 335  # vp9.2 10-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting p010 -> yuv420p

./youtube-play-vaapi https://youtu.be/Jz9TdfXlTgs 248  # vp9 8-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting nv12 -> yuv420p

./youtube-play-vaapi https://youtu.be/aqz-KE-bpKQ 399  # av1 8-bit
   Using hardware decoding (vaapi-copy).
   [autoconvert] Converting nv12 -> yuv420p
```

elFarto, 2023-01-07:
Thanks for the testing. I'll close this issue now.

closed 2023-01-07 by elFarto