elFarto/nvidia-vaapi-driver #149: No video playback starts on direct backend and driver 525

Spanfile, 2022-12-27:
Commit: 2938a5910315a920dd220cf63047b3cb25ba522b 
OS: Arch Linux, kernel 6.1.1
Nvidia: 525.60.11, RTX 3080 Ti

Attempting to start any video on any site buffers a bit of the video but never starts playing. I had to do what #126 described after 525 and it worked back then, but after that something changed (probably Firefox or the kernel, I don't recall an nvidia update) and now no video works. This could just be a Firefox issue but I'm too stupid to figure Mozilla's bug tracker out if something like this has been reported to them.

<details>
<summary>vainfo</summary>

```
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

</details>

<details>
<summary>NVD_LOG=1 NVD_BACKEND=direct firefox</summary>

```
[2022-12-27T18:30:27Z ERROR glean_core::metrics::ping] Invalid reason code startup for ping background-update
      5211.808759487 [44042-44055] ../src/vabackend.c:2083       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      5211.808766837 [44042-44055] ../src/vabackend.c:2092       __vaDriverInit_1_0 Now have 0 (0 max) instances
      5211.808770484 [44042-44055] ../src/vabackend.c:2118       __vaDriverInit_1_0 Selecting Direct backend
      5211.813022797 [44042-44055] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      5211.813033341 [44042-44055] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 2 6
      5211.813101862 [44042-44055] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.60.11
      5211.860651276 [44042-44055] ../src/vabackend.c:1395      nvQueryImageFormats In nvQueryImageFormats
      5211.875866840 [44042-44055] ../src/vabackend.c: 637           nvCreateConfig got profile: 6 with 0 attributes
      5211.875876432 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      5211.875879104 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) 0x7fb688470e80 7
      5211.876717032 [44042-44055] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      5211.876751996 [44042-44055] ../src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
      5211.876756017 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      5211.876758183 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) 0x7fb688470040 7
      5211.878038084 [44042-44055] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      5211.878075230 [44042-44055] ../src/vabackend.c: 637           nvCreateConfig got profile: 13 with 0 attributes
      5211.878079502 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      5211.878081284 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 4 (8) 0x7fb688470d00 7
      5211.879374634 [44042-44055] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      5211.879412633 [44042-44055] ../src/vabackend.c: 637           nvCreateConfig got profile: 14 with 0 attributes
      5211.879417692 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      5211.879419874 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 9 (8) 0x7fb688470040 7
      5211.880836730 [44042-44055] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      5211.880869616 [44042-44055] ../src/vabackend.c: 637           nvCreateConfig got profile: 19 with 0 attributes
      5211.880873402 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      5211.880875135 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 10 (8) 0x7fb688470c40 7
      5211.881993717 [44042-44055] ../src/vabackend.c:1807 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      5211.882025872 [44042-44055] ../src/vabackend.c: 637           nvCreateConfig got profile: 32 with 0 attributes
      5211.882028828 [44042-44055] ../src/vabackend.c: 700           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      5211.882031847 [44042-44055] ../src/vabackend.c:1749 nvQuerySurfaceAttributes with 11 (8) (nil) 0
[... 272 lines cut ...]
      5222.118525611 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface 1 (0x7fb6882fa2a0)
      5222.118528279 [44042-44168] ../src/direct/direct-export-buf.c: 248      destroyBackingImage Destroying BackingImage: 0x7fb688afdbb0
      5222.118628715 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface 4 (0x7fb6882fa8e0)
      5222.118631713 [44042-44168] ../src/direct/direct-export-buf.c: 248      destroyBackingImage Destroying BackingImage: 0x7fb6b35e6e20
      5222.118775777 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface 3 (0x7fb6882faa20)
      5222.118778422 [44042-44168] ../src/direct/direct-export-buf.c: 248      destroyBackingImage Destroying BackingImage: 0x7fb6897b9d90
      5222.118931480 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface 2 (0x7fb6882fa340)
      5222.118933884 [44042-44168] ../src/direct/direct-export-buf.c: 248      destroyBackingImage Destroying BackingImage: 0x7fb6897b8fc0
      5222.119033575 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa660)
      5222.119036268 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fac00)
      5222.119037704 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fb380)
      5222.119038973 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa3e0)
      5222.119055741 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa5c0)
      5222.119057034 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fb240)
      5222.119058335 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa520)
      5222.119060688 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa160)
      5222.119062242 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa480)
      5222.119063647 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6882fa200)
      5222.119065685 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6880b3ce0)
      5222.119067499 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6880b3740)
      5222.119069247 [44042-44168] ../src/vabackend.c: 951        nvDestroySurfaces Destroying surface -1 (0x7fb6880b3a60)
      5222.119072633 [44042-44168] ../src/vabackend.c:2055              nvTerminate Terminating 0x7fb6b352eb60
      5222.119119077 [44042-44168] ../src/vabackend.c:2069              nvTerminate Now have 0 (0 max) instances
```
</details>

elFarto, 2022-12-27:
I can't see anything in those log outputs that indicates a problem. I also don't see any issues in the Mozilla Bugzilla that indicate a problem.

Can you run Firefox with this environment variable set aswell: `MOZ_LOG="PlatformDecoderModule:5"`?

Spanfile, 2022-12-27:
This is with `MOZ_LOG="PlatformDecoderModule:5" NVD_LOG=1 NVD_BACKEND=direct firefox`: https://termbin.com/8qi7x

Spanfile, 2022-12-27:
I figured it out; my sound card was misbehaving and for some reason prevented video playback as well.

closed 2022-12-27 by Spanfile