elFarto/nvidia-vaapi-driver #151: Firefox Uses VAAPI for WebGL but refuses to use it for video

BeepBoopBang, 2022-12-28:
I have all about:config's set, and all environment variables set. VAINFO gives good output, but when watching a video nvidia-settings shows no Video Decoding, and nvidia-smi does show a process for firefox but only type "G" 

I have a GTX 1070 and am on NVIDIA Driver Version 510.108.03. I am using X11.

elFarto, 2022-12-29:
Judging by your closed issue #150, it doesn't seem to want to use EGL. This might be caused by Debian having too old a version of libEGL. Also, if Firefox says you're using the 470 drivers you probably are, as those version numbers come from NVIDIA's OpenGL driver directly, so shouldn't be effected by any packaging system.

BeepBoopBang, 2022-12-29:
Hi, I closed that issue because I realized I was completely wrong about the graphics driver versioning, and the version I had earlier was actually even older than the typical Nvidia Driver in Debian stable. So I backported driver version 510.108.03 myself, and now the eGL error has gone away, and my driver is now correctly listed as "4.6.0 NVIDIA 510.108.03". Everything to me points to VAAPI working, it just seems to not want to use it for video. I tried using "H264ify" to see if that helped but it had no effect.

evelez7, 2022-12-30:
I'm having a similar issue. Firefox uses WebRender (according to `about:support`), so it is hardware accelerated, but `HARDWARE_VIDEO_DECODING` says "blocklisted by env: Blocklisted by gfxInfo." I have also verified by `nvidia-settings` and `nvidia-smi` that there is no video engine use while watching video.

I did some digging and it seems that Firefox automatically disables all NVIDIA video hardware acceleration on Linux. See the Firefox [source code for `GfxInfo.cpp`](https://searchfox.org/mozilla-central/source/widget/gtk/GfxInfo.cpp) on line 866-872. It is appending information to a block list, hence the text under the video decoding field.

My only guess is it would work if we could disable the block lists.

BeepBoopBang, 2022-12-30:
Thanks evelez7... your comment seems to have only given me more questions though. I took a look at the source for firefox-esr (what I am using) and it appears to have no hardware decode support? At least the blocklist you refer to is not present in the "GfxInfo.cpp" file for firefox-esr 102.6.0esr. After getting the latest firefox binary it appears video acceleration is working. 

While I know firefox-esr is less feature complete I guess I assumed that since the version was higher than 98 it would work, and the about:config entries appear to point to there being support for hardware video acceleration. My bad.

The thing is in the latest firefox binaries about:support, hardware video acceleration does say it is block listed, but testing in nvidia-settings and nvidia-smi appears to show it working after I correctly set my about:config options.

morfeus77, 2023-01-02:
Just wanted to chime in on this issue.  I couldn't make this work for some time either - everything seemed set correctly yet Video Engine Utilization in Nvidia settings stayed at 0%.  I was almost giving up on it, but since today it seems to be working: Video Utilization does show it being used.  I'm not exactly sure what made it worked.  It's either one of these 3 things: clearing the startup cache (in about:support), reboot or installation of the libgstreamer-plugins-bad1.0-dev optional package for additional codecs and rebuilding the shared library.  

PS: about:support still gives me this info for hardware video decoding but it seems the user pref overrules the blocklist

HARDWARE_VIDEO_DECODING	
available by default
force_enabled by user: Force enabled by pref
blocklisted by env: Blocklisted by gfxInfo

marioroy, 2023-01-02:
Confirmation: The libgstreamer-plugins-bad1.0-dev package is a requirement to build nvidia-vaapi-driver, not optional.

It has been a while since I ran Ubuntu. The `libgstreamer-plugins-bad1.0-0` is the runtime package to utilize the NVIDIA video engine. That's the one; possibly a package dependency when installing `libgstreamer-plugins-bad1.0-dev`.

The following script installs minimum dependencies on Ubuntu for building libva, nv-codec-headers, nvidia-vaapi-driver, and vdpau-va-driver. Ubuntu 20.04 requires more recent libva (okay on 22.04, not sure 21.10). The vdpau-va-driver is used by Google Chrome and derivatives.

```bash
#!/bin/bash
# Usage: sudo bash install-dependencies

# Get installed packages; append leading and trailing spaces.
installed_packages=$(dpkg --list | grep '^ii ' | cut -f1 -d: | awk '{ print "", $2, "" }')

# Append list of packages to install.
add_list=""
count=0

for package_name in \
    build-essential autoconf cmake curl git libtool meson ninja-build vainfo \
    libavcodec-extra libdrm-dev libegl-dev libgl-dev libglx-dev \
    libgstreamer1.0-dev libgstreamer-plugins-bad1.0-0 libvdpau-dev \
    libwayland-dev libxcb-dri3-dev libxcb-present-dev \
    libxext-dev libxfixes-dev \
;do
    if [[ ! ${installed_packages} =~ " ${package_name} " ]]; then
        add_list="${add_list} ${package_name}"
    else
        count=$(( ${count} + 1 ))
    fi
done

# Install missing packages.
if [[ -n ${add_list} ]]; then
    add_list=${add_list## } # remove leading whitespace
    echo "apt update -y && apt install -y ${add_list}"
    apt update -y && apt install -y ${add_list}
fi

if [[ ${count} -gt 0 ]]; then
    echo "${count} packages were already installed"
fi
```

elFarto, 2023-01-03:
> Thanks evelez7... your comment seems to have only given me more questions though. I took a look at the source for firefox-esr (what I am using) and it appears to have no hardware decode support? At [...]

It's good that it's working with the latest Firefox atleast.

> Confirmation: The libgstreamer-plugins-bad1.0-dev package is a requirement (for video engine utilization) to build nvidia-vaapi-driver, not optional.

This shouldn't be the case. The -bad package is only needed for VP9 support.

elFarto, 2023-01-03:
I should add that setting `media.ffmpeg.vaapi.enabled` to `true` will override the blocklist. It is a bit confusing how Firefox displays this information as it's not clear which one has priority.

drod3763, 2023-01-06:
I have the same problem although I'm not sure how to prove firefox is using it other than the about:config switches. `vainfo` works fine but Video Engine Utilization is 0.

```
Trying display: wayland
Trying display: x11
      7302.893982934 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55a50d947470 10
      7302.893988894 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
      7302.893990390 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
      7302.897443434 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/direct/direct-export-buf.c:  94      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      7302.897447641 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
      7302.897450401 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 0 0 0
      7302.897514023 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.78.01
vainfo: VA-API version: 1.17 (libva 2.17.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      7302.955149299 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x55a50d947470
      7302.955185083 [43058-43058] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

Here is firefox when I try to load a youtube video:
```
 NVD_LOG=1 firefox-developer-edition
7973.805507153 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7fae5ba304c0 31
7973.805516237 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
7973.805518549 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
7973.810118433 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
7973.810126973 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 0 0 0
7973.810195855 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.78.01
7973.856379154 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
7973.868821096 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 6 with 0 attributes
7973.868831108 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
7973.868833358 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7fae30272700 7
7973.869585593 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 7 with 0 attributes
7973.869590778 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
7973.869592699 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7fae30272340 7
7973.870344973 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 13 with 0 attributes
7973.870350025 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
7973.870351985 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7fae30272040 7
7973.871093508 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 14 with 0 attributes
7973.871098523 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 9 (8) (nil) 0
7973.871100477 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 9 (8) 0x7fae30272a00 7
7973.872154211 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 19 with 0 attributes
7973.872160780 [46147-46543] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 10 (8) (nil) 0
[... cut ...]
7978.730857143 [46147-46865] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7fae316d2e20)
7978.730858994 [46147-46865] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7fae316d2ce0)
7978.730861788 [46147-46865] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7fae5ba30e20
7978.730903270 [46147-46865] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instancesh
```

elFarto, 2023-01-07:
@drod3763 This line is very concerning:
```
...             alloc_image DRM_IOCTL_NVIDIA_GEM_IMPORT_NVKMS_MEMORY failed: -1 0
```
I've added more logging to the master branch to figure out what error is being thrown, could you retest with that and post the log output here?

GudPiggeh, 2023-01-11:
I had a very similar error log to @drod3763 . I installed `nvidia-vaapi-driver-git` (AUR), and suddenly `vainfo` told me I needed `nvidia_drm.modeset=1` in my grub, which it wasn't telling me before; It turns out I accidentally had a typo when I put that in earlier, and after fixing it, the driver now seems to work fine.

closed 2023-01-28 by BeepBoopBang