elFarto/nvidia-vaapi-driver #154: VP9 not hardware accelerated on Piped with RTX 2070 (Question)

Gert-dev, 2023-01-03:
Hardware acceleration works for me on sites such as Invidious using h.264, but on Piped when I select VP9, it doesn't seem to kick in. I'm not entirely sure if this _should_ be working for me on an RTX 2070 Super Mobile Max-Q since I don't exactly know what profile I would need support for for this to work, i.e. `vainfo` says:

```
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

`Profile0` and `Profile2` are listed, but perhaps the profile required for Piped VP9 is different? I'm testing using [this video](https://piped.video/watch?v=BHACKCNDMW8) and forcing VP9 and h.264 through the preferences in the upper-right corner; I clearly see `C+G` in `nvidia-smi` and `Video Engine Utilization` in the NVIDIA control panel going up when forcing h.264, but not when forcing VP9.

I'm also using the direct backend on version 0.0.8 currently.

elFarto, 2023-01-03:
VAProfileVP9Profile0 is the main VP9 profile that's used, VAProfileVP9Profile2 is for HDR (10 and 12-bit surfaces), so you have the hardware to decode that video.

Can you run firefox with `NVD_LOG=1` set, attempt to play back the VP9 video and post the output here?

thesword53, 2023-01-04:
> vainfo: Driver version: VA-API NVDEC driver [egl backend]

According to the `vainfo` output you're using EGL backend.

Gert-dev, 2023-01-04:
I think this can be closed as invalid. I tried creating a new profile and it works there. There are no errors for my current profile, but I am using [arkenfox](https://github.com/arkenfox) so I'm assuming that it's one of those settings that is interfering for some reason.

> According to the vainfo output you're using EGL backend.

Yes, the output is the same for the direct back end (I was only setting the necessary environment variable for Firefox and executed `vainfo` in a new shell).

closed 2023-01-04 by Gert-dev