elFarto/nvidia-vaapi-driver #155: Can't get Firefox video hardware decoding to work

IvanDSM, 2023-01-04:
I'm on Arch Linux, with a GTX 950 using the latest version of the libva-nvidia-driver package, Firefox Nightly 110.0a1 (2022-12-24), NVIDIA drivers 525.60.11 running Wayland with the labwc compositor. I have the about:config and envvars set according to the README, have configured my GRUB cmdline default to set nvidia-drm.modeset to 1 and I've have tried setting NVD_BACKEND to direct. No matter what I do, Firefox outputs this on its log:

```
[RDD 277288: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: Initialising VA-API FFmpeg decoder
[RDD 277288: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   codec vp9 : Google VP9
[RDD 277288: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Can't get DRM VA-API display.
[RDD 277288: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   Failed to create VA-API device context
```

Setting NVD_LOG=1 doesn't result in any output from the vaapi driver.

I'm happy to provide any other necessary details.

rkoot, 2023-01-04:
Did you install gstreamer-codecparsers-1.0 ? this is required for VP9.

IvanDSM, 2023-01-04:
I have the following gstreamer packages installed:

```
gst-libav 1.20.5-1
gst-plugin-pipewire 1:0.3.63-3
gst-plugin-va 1.20.5-1
gst-plugins-bad 1.20.5-1
gst-plugins-bad-libs 1.20.5-1
gst-plugins-base 1.20.5-1
gst-plugins-base-libs 1.20.5-1
gst-plugins-good 1.20.5-1
gst-plugins-ugly 1.20.5-1
gstreamer 1.20.5-1
```
The `gstreamer-1.0/gst/codecparsers/gstvp9parser.h` header is present in the gst-plugins-bad-libs package, so it seems that I have it installed.

elFarto, 2023-01-04:
Can you paste the output of `vainfo`?

IvanDSM, 2023-01-04:
Yeah! Here you go:

With NVD_BACKEND=egl:

```
Trying display: wayland
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
```

With NVD_BACKEND=direct:

```
Trying display: wayland
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
```

elFarto, 2023-01-04:
Can you try setting `media.rdd-ffvpx.enabled` to `false` in `about:config` and restarting Firefox?

IvanDSM, 2023-01-05:
I tried doing that but no dice - same error output. Do you think VP9 specifically is the issue? I could try installing h264ify to see if h264 gives me a different result.

elFarto, 2023-01-05:
The key error here seems to be `Can't get DRM VA-API display`. This indicates that Firefox's call to vaGetDisplayDRM failed. Normally this would be because libva is unable to work out which driver to use. That was normally fixed by setting `LIBVA_DRIVER_NAME=nvidia`, but that's not needed with the version of libva you have (2.16).

libva should be printing some logging while it's trying to load the driver, and the fact you're not seeing that means there's something wrong with Firefox calling libva.

If I get some time later I'll try running the nightly build to see if something has changed.

elFarto, 2023-01-07:
Could you try running Firefox with these environment variables set: `MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" NVD_LOG=1`. That should indicate which DRM device it's using:

```
[RDD 10817: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX: Initialising VA-API FFmpeg decoder
[RDD 10817: MediaPDecoder #1]: D/PlatformDecoderModule FFVPX:   codec vp9 : Google VP9
[RDD 10817: MediaPDecoder #1]: D/Dmabuf Using DRM device /dev/dri/renderD128
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      4424.808943454 [10817-10825] ../src/vabackend.c:2088       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
```

marioroy, 2023-01-07:
> That was normally fixed by setting LIBVA_DRIVER_NAME=nvidia, but that's not needed with the version of libva you have (2.16).

This may no longer the case on x11, but okay on Wayland. I have `libva 2.17` and requires setting `LIBVA_DRIVER_NAME=nvidia`.

### x11

```text
$ vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.17.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit

$ LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.17.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

### wayland

```text
$ vainfo
Trying display: wayland
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.17.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

marioroy, 2023-01-07:
Interestingly (back on x11), `firefox 108.0.2` and `mpv 0.35.0` picks the `nvidia` driver without setting `LIBVA_DRIVER_NAME`.

elFarto, 2023-01-07:
I'm on X11, and vainfo picks the correct driver without `LIBVA_DRIVER_NAME` set, but that's with libva 2.16. I assume you're running a nightly build of libva, as 2.17 hasn't been released yet. If that's the case it may be worth raising an issue in their bug tracker for that.

marioroy, 2023-01-07:
vainfo does not load the nvidia va driver on x11, but okay on wayland
https://github.com/intel/libva/issues/675

evelikov, 2023-01-10:
The libva frontend uses nvctrl X extension to obtain the nvidia driver name for X11 and (X)wayland. For va/drm the detection works like the open-source drivers - via the drm module name and should just work.

If it were me, I would tweak the driver to work support `VA_DISPLAY_DRM`, which should (lead at least in theory) lead to gst and others falling back to va/drm.

If @elFarto is happy with that option I can open a PR.

elFarto, 2023-01-10:
Support isn't the issue here, the driver can work regardless of how it's opened. We ignore any X11 information passed in as we have no way to determine the GPU from that. The issue is that libva (without nvctrl support) can't figure out which driver to use.

evelikov, 2023-01-12:
> We ignore any X11 information passed in as we have no way to determine the GPU from that

IMHO this sounds like a good enough reason to disable `VA_DISPLAY_X11` within the driver.

elFarto, 2023-01-12:
> > We ignore any X11 information passed in as we have no way to determine the GPU from that
> 
> IMHO this sounds like a good enough reason to disable `VA_DISPLAY_X11` within the driver.

I'm not sure I follow. We don't explicitly say anywhere in the driver what we support or don't support. Even if we add something to fail on X11 initialisation, we'd never get there as libva has no idea _which_ driver to load.

adamnejm, 2023-04-08:
Setting `media.rdd-ffvpx.enabled` to `false` worked for me on X11 with `nvidia 530.41.03-3`
I love you **elFarto**! :heart:

closed 2023-06-16 by elFarto