elFarto/nvidia-vaapi-driver #158: Firefox's RDD Process crashes randomly

guihkx, 2023-01-08:
- Firefox 108.0.2
- nvidia-vaapi-driver d628720416812b8db9d62519892b3fdb31076ece
- NVIDIA drivers: 470.161.03
- Session: X11

The crash is reproducible by F5'ing any supported video (I'm using [this one](https://docs.evostream.com/sample_content/assets/bun33s.mp4)), but the crash itself is random. The hw decoding still seems to work just fine though, so I'm not sure if it's actually caused by Firefox...

Logs:

- [firefox-108.0.2-rdd-process-crash.txt](firefox-108.0.2-rdd-process-crash.txt) (generated with: `NVD_MAX_INSTANCES=3 NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" firefox 2>&1`)
- Firefox crash report: https://crash-stats.mozilla.org/report/index/45584a18-2eb2-4881-bc66-2b4880230108

---

Aside:

Do you know why `vainfo` stopped working? It used to work at some point in the past...

```
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
Trying display: x11
     40856.602870613 [23253-23253] ../src/vabackend.c:2088       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     40856.602882418 [23253-23253] ../src/vabackend.c:2097       __vaDriverInit_1_0 Now have 0 (0 max) instances
     40856.602885375 [23253-23253] ../src/vabackend.c:2120       __vaDriverInit_1_0 Selecting EGL backend
     40856.606382686 [23253-23253] ../src/export-buf.c: 137       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     40856.606394477 [23253-23253] ../src/export-buf.c: 154       findGPUIndexFromFd Looking for GPU index: 0
     40856.607632054 [23253-23253] ../src/export-buf.c: 166       findGPUIndexFromFd Found 1 EGL devices
     40856.607925220 [23253-23253] ../src/export-buf.c: 175       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     40856.607945378 [23253-23253] ../src/export-buf.c: 224       findGPUIndexFromFd No match found, falling back to default device
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

elFarto, 2023-01-08:
The log you've posted doesn't actually seem to be using this driver for decode. It initialises it, but nothing is logged by this library for each frame decode. The crash report doesn't have this library in the stack trace, so it doesn't at least appear to be directly caused by us. This looks like a regular Firefox bug.

vainfo should still be working, but it might have gotten broken for 470 series drivers. I've push a small fix that should resolve that.

guihkx, 2023-01-08:
>The log you've posted doesn't actually seem to be using this driver for decode.

You're actually right. I got confused because I was recording a video of the crash, and the 'Video Engine Utilization' field on NVIDIA X Server Settings was actually showing some activity, but it was being caused by the software I was using to record. 🤡 

> I've push a small fix that should resolve that.

Thank you, it works now:

```
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
Trying display: x11
     43651.547763939 [29054-29054] ../src/vabackend.c:2088       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     43651.547777039 [29054-29054] ../src/vabackend.c:2097       __vaDriverInit_1_0 Now have 0 (0 max) instances
     43651.547780148 [29054-29054] ../src/vabackend.c:2120       __vaDriverInit_1_0 Selecting EGL backend
     43651.551879295 [29054-29054] ../src/export-buf.c: 137       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     43651.551893372 [29054-29054] ../src/export-buf.c: 154       findGPUIndexFromFd Looking for GPU index: 0
     43651.553195455 [29054-29054] ../src/export-buf.c: 166       findGPUIndexFromFd Found 1 EGL devices
     43651.553302933 [29054-29054] ../src/export-buf.c: 175       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     43651.553316568 [29054-29054] ../src/export-buf.c: 213       findGPUIndexFromFd Selecting EGLDevice 0
     43651.554947277 [29054-29054] ../src/export-buf.c: 286         egl_initExporter Driver doesn't support 16-bit surfaces
vainfo: VA-API version: 1.17 (libva 2.17.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
     43651.595506900 [29054-29054] ../src/vabackend.c:2060              nvTerminate Terminating 0x5635b8cee5f0
     43651.595515014 [29054-29054] ../src/export-buf.c:  66      egl_releaseExporter Releasing exporter, 0 outstanding frames
     43651.595525835 [29054-29054] ../src/export-buf.c:  83      egl_releaseExporter Done releasing frames
     43651.595530150 [29054-29054] ../src/vabackend.c:2074              nvTerminate Now have 0 (0 max) instances
```

>This looks like a regular Firefox bug.

Oh... :(

I guess I'll open a bug report on their tracker, then. Thank you for your time!

guihkx, 2023-01-08:
Okay, so I tried to reproduce the crash on a vey old commit (more specifically, dbfb6083809941c3f2257b5f402964fba55b6cfe), but it wouldn't crash anymore.

So I git bisected and 41bb944a01bef1cfeac70ef0cc01b993e264aade seems to be the one that introduced the crashes:

```
41bb944a01bef1cfeac70ef0cc01b993e264aade is the first bad commit
commit 41bb944a01bef1cfeac70ef0cc01b993e264aade
Author: Stephen <elfarto@elfarto.com>
Date:   Wed Jan 4 08:11:30 2023 +0000

    wip for fixing av1 10-bit via mpv

 src/vabackend.c | 35 ++++++++++++++++++++---------------
 1 file changed, 20 insertions(+), 15 deletions(-)
```

Bisect log:

```
git bisect start
# status: waiting for both good and bad commits
# bad: [038cde361d71f22d6b09f1f9eae2e508d664e9c8] made checkModeset return true if there's no fd to check
git bisect bad 038cde361d71f22d6b09f1f9eae2e508d664e9c8
# status: waiting for good commit(s), bad commit known
# good: [dbfb6083809941c3f2257b5f402964fba55b6cfe] Update README.md
git bisect good dbfb6083809941c3f2257b5f402964fba55b6cfe
# good: [2cb7c03e46f9613c451f5b2d32777cfa08422e1b] replaced logic that calculates log2GobsPerBlockX/Y/Z with something simpler that works for videos less than 128px in height
git bisect good 2cb7c03e46f9613c451f5b2d32777cfa08422e1b
# good: [01dca0cc5aaf3c7c1647d552d6320669cd8cec41] Merge pull request #144 from crimist/yuv444-docs
git bisect good 01dca0cc5aaf3c7c1647d552d6320669cd8cec41
# good: [0fac0d684c0b996c6aa17f072fb5132f0ce3112d] Merge pull request #152 from lengau/patch-1
git bisect good 0fac0d684c0b996c6aa17f072fb5132f0ce3112d
# bad: [681f12f3d588a1d98189cc11f109a53b99320544] Merge pull request #157 from elFarto/av1-10bit-fix
git bisect bad 681f12f3d588a1d98189cc11f109a53b99320544
# good: [cefe34f449d1835f01132431493fd4fdf6fe8ec6] Merge pull request #156 from kwizart/fedora_rhel
git bisect good cefe34f449d1835f01132431493fd4fdf6fe8ec6
# bad: [43fe35d6c73a2631f72091ef0e614584460a3734] tidied up av1 10-bit fix
git bisect bad 43fe35d6c73a2631f72091ef0e614584460a3734
# bad: [41bb944a01bef1cfeac70ef0cc01b993e264aade] wip for fixing av1 10-bit via mpv
git bisect bad 41bb944a01bef1cfeac70ef0cc01b993e264aade
# first bad commit: [41bb944a01bef1cfeac70ef0cc01b993e264aade] wip for fixing av1 10-bit via mpv
```

And sure enough, building on the parent commit (0fac0d684c0b996c6aa17f072fb5132f0ce3112d) stops the crashing...

elFarto, 2023-01-08:
Well, I can't reproduce this on my machine, but do see an issue with that commit that would be triggered by a driver not supporting 16-bit surfaces, which the 470 driver doesn't.

I've committed a fix that should fix it.

guihkx, 2023-01-08:
Just tested it and crash is gone, nice!

Thanks a lot for the quick fix!

closed 2023-01-08 by guihkx