elFarto/nvidia-vaapi-driver #159: Colour corruption (artifacts) in small VP8 video

Gert-dev, 2023-01-09:
Hardware acceleration is working well for me in pretty much all places, but I encountered a short VP8 video of a low resolution that has distorted colours.

It looks a bit like colour information is read from the wrong offset or something with a planar format or stride issue of some sort; the artifacts seem to remain the same for me across the length of the video.:

[image]

`NVD_LOG=1` reveals nothing to note for me, but a more experienced eye might see more: [VP8VideoColourCorruptionLog.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/10376003/VP8VideoColourCorruptionLog.txt)

This is the video that reproduces it for me, grabbed from a search engine for convenience (copyright belongs to whoever is the owner, no infringement intended, but merely for debugging purposes here).

[VP8VideoColourCorruptionVideo.webm](https://user-images.githubusercontent.com/24809863/211384439-5980dd83-4fe4-4228-9411-2599ecba7a3d.webm)

# Version Info
The usual version details:

* Version: 0.0.8
* GPU: RTX 2070 Super Mobile Max-Q
* NVIDIA driver version: 525.78.01-1.1
* Back-end: direct
* Firefox version: 108.0.2

`vainfo`:

```
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.17 (libva 2.17.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

elFarto, 2023-01-09:
I can't actually test this as my card doesn't support VP8, but I know where the problem lies.

[This](https://github.com/elFarto/nvidia-vaapi-driver/blob/7d4a27a99e54632867cb5a326efca86b41e257a2/src/direct/nv-driver.c#L431) line controls a special number that's passed to Firefox, that passes it to NVIDIA's EGL implementation. It tells EGL how the memory was allocated. I'm pretty sure it should it's being set incorrectly for the chroma plane (the one that controls colour).

You can try changing the `88` on that line to `91`, and see if that fixes it. If it does, then I'm a bit stuck as I've had videos that require it set to 4 for 88px and having a 90px one that needs 3 would leave me unable to compute it correctly.

Gert-dev, 2023-01-10:
I guess I should have tested latest master first, because it appears to be fixed there :stuck_out_tongue: . I'm using 0.0.8, and, looking at the commits, b0edcec294abcf412ffb3b7719498e95a1d464a3 was added after the release, which is a value that seems to work for me, at least for this video.

closed 2023-01-10 by Gert-dev