elFarto/nvidia-vaapi-driver #160: libva error: /usr/lib/dri/nvidia_drv_video.so init failed with AMD/nvidia (Nvidia Prime) Wayland

Arjun31415, 2023-01-12:
- System Details -
```
Distro: Garuda Linux
    base: Arch Linux
Kernel: 6.1.4-zen2-1-zen arch: x86_64 bits: 64 compiler: gcc v: 12.2.0
    parameters: BOOT_IMAGE=/@/boot/vmlinuz-linux-zen
    root=UUID=e52cff4d-7ee9-4bf9-9114-598c5db0fba7 rw rootflags=subvol=@
    quiet quiet splash rd.udev.log_priority=3 vt.global_cursor_default=0
    loglevel=3 ibt=off
Desktop: Hyprland 
CPU:
  Info: model: AMD Ryzen 7 4800H with Radeon Graphics bits: 64 type: MT MCP
    arch: Zen 2 gen: 3 level: v3 note: check built: 2020-22
    process: TSMC n7 (7nm) family: 0x17 (23) model-id: 0x60 (96) stepping: 1
    microcode: 0x8600103
Graphics:
  Device-1: NVIDIA TU106M [GeForce RTX 2060 Mobile] vendor: Hewlett-Packard
    driver: nvidia v: 525.78.01 alternate: nouveau,nvidia_drm non-free: 525.xx+
    status: current (as of 2022-12) arch: Turing code: TUxxx
    process: TSMC 12nm FF built: 2018-22 pcie: gen: 1 speed: 2.5 GT/s lanes: 8
    link-max: gen: 3 speed: 8 GT/s lanes: 16 ports: active: none off: HDMI-A-1
    empty: DP-1,DP-2 bus-ID: 01:00.0 chip-ID: 10de:1f15 class-ID: 0300
  Device-2: AMD Renoir vendor: Hewlett-Packard driver: amdgpu v: kernel
    arch: GCN-5.1 code: Vega-2 process: TSMC n7 (7nm) built: 2018-21 pcie:
    gen: 4 speed: 16 GT/s lanes: 16 ports: active: eDP-1 empty: none
    bus-ID: 06:00.0 chip-ID: 1002:1636 class-ID: 0300 temp: 42.0 C
```
- `mkinitcpio.conf`
```
MODULES=(crc32c intel_agp i915 amdgpu radeon nouveau nvidia nvidia_modeset nvidia_uvm nvidia_drm)
BINARIES=()
FILES=""
HOOKS="base udev autodetect modconf block keyboard keymap consolefont plymouth filesystems grub-btrfs-overlayfs"
```

- `nvidia-smi` -

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.78.01    Driver Version: 525.78.01    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   42C    P8     8W /  80W |    120MiB /  6144MiB |     39%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A       967      G   Hyprland                            1MiB |
|    0   N/A  N/A      1611      G   ...056829623301762679,131072       96MiB |
+-----------------------------------------------------------------------------+
```
- Nvidia Drivers -
`sudo pacman -Qq | grep "nvidia"`
```
garuda-nvidia-config
garuda-nvidia-prime-config
hyprland-nvidia-hidpi-git
lib32-nvidia-utils
lib32-opencl-nvidia
mhwd-nvidia
mhwd-nvidia-390xx
mhwd-nvidia-470xx
nvidia-dkms
nvidia-prime
nvidia-settings
nvidia-utils
opencl-nvidia
```

- VAAPI/ VDPAU -
`sudo pacman -Qq | grep -i "mesa"`
```
lib32-libva-mesa-driver
lib32-mesa
lib32-mesa-demos
lib32-mesa-vdpau
lib32-vulkan-mesa-layers
libva-mesa-driver
mesa
mesa-demos
mesa-utils
mesa-vdpau
vulkan-mesa-layers
```
`sudo pacman -Qq | grep -i "vdpau"`

```
lib32-libva-vdpau-driver
lib32-libvdpau
lib32-mesa-vdpau
libva-vdpau-driver
libvdpau
libvdpau-va-gl
mesa-vdpau
vdpauinfo
```

Nvidia-vaapi-driver installed and compiled from git source

- `vainfo` with `nouveau` 
```
vainfo
Trying display: wayland
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: Mesa Gallium driver 22.3.2 for AMD Radeon Graphics (renoir, LLVM 14.0.6, DRM 3.49, 6.1.4-zen2-1-zen)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
```
```
vdpauinfo
display: :0   screen: 0
Xlib:  extension "NV-GLX" missing on display ":0".
Xlib:  extension "NV-GLX" missing on display ":0".
Error creating VDPAU device: 1
```
-  With nvidia driver

`LIBVA_DRIVER_NAME=nvidia vainfo`
```
Trying display: wayland
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

`VDPAU_DRIVER=nvidia vdpauinfo`
```
display: :0   screen: 0
Xlib:  extension "NV-GLX" missing on display ":0".
Xlib:  extension "NV-GLX" missing on display ":0".
Error creating VDPAU device: 1
```

I tried the same without  `nvidia-prime` , no change in result. With `nouveau` it keeps using iGPU

Arjun31415, 2023-01-12:
` /sys/module/nvidia_drm/parameters/modeset` -
```
Y
```

rkoot, 2023-01-12:
it looks like your nvidia driver installation isn't properly initialized. vdpauinfo shouldn't spit out 'extension "NV-GLX" missing on display ":0".'. This is an indication your drivers isn't working properly. Maybe reinstall nvidia drivers and reboot.
Also, I noticed you have the libva-vdpau-driver (both 32bit and 64 bit) packages installed. Remove those as they tend to confuse libva.

Arjun31415, 2023-01-12:
Sure Ill reinstall nvidia drivers and try again. Also do i have to remove all lib32 vdpau drivers ?. Or only `lib32-libva-vdpau-driver`

rkoot, 2023-01-12:
Just the lib32-libva-vdpau-driver

Arjun31415, 2023-01-12:
I'm afraid even after reinstalling the driver, i am getting NV-GLX missing, is there some info i could provide for this

Arjun31415, 2023-01-12:
but without the nvdia-dkms driver there was no error, and it detected only 1 screen when there were 2 monitors

Arjun31415, 2023-01-12:
Also i noticed glxinfo does not print any missing NV-GLX errors

elFarto, 2023-01-14:
So this line is suspect:
```
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
```
Can you run `NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo`, and paste the output here?

I think the NV-GLX errors are expected, as  you're running under Wayland, so I don't think that X extension is exposed with Xwayland.

Arjun31415, 2023-01-14:
here -
```
Trying display: wayland
        67.310946676 [4264-4264] ../src/vabackend.c:2090       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
        67.310965034 [4264-4264] ../src/vabackend.c:2099       __vaDriverInit_1_0 Now have 0 (0 max) instances
        67.310972991 [4264-4264] ../src/vabackend.c:2125       __vaDriverInit_1_0 Selecting Direct backend
        67.320207965 [4264-4264] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: amdgpu
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
Seems like it is detecting `amdgpu` . I am not sure what i have done wrong for it to be detecting that

elFarto, 2023-01-14:
You've not done anything wrong, but running this driver on Optimus setups (with either Intel or AMD as the other card) is not simple, and I'm not sure anyone has got it to run correctly. By default libva is using the same card that your compositor is using.

You can try running `NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display drm --device /dev/dri/renderD128` (or 129) to tell libva you want it to use a different card, but getting Firefox to use a different card is a whole lot harder.

Arjun31415, 2023-01-14:
alright let me try it out. Maybe if i uninstall all amd drivers that will happen, but again I don't want to run the card on performance mode later to keep up

Arjun31415, 2023-01-14:
```
NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display drm --device /dev/dri/renderD128
Trying display: drm
      1410.403110448 [6012-6012] ../src/vabackend.c:2090       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1410.403133286 [6012-6012] ../src/vabackend.c:2099       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1410.403140550 [6012-6012] ../src/vabackend.c:2125       __vaDriverInit_1_0 Selecting Direct backend
      1410.412283743 [6012-6012] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: amdgpu
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

```
NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo --display drm --device /dev/dri/renderD129 
Trying display: drm
      1417.643330330 [6049-6049] ../src/vabackend.c:2090       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1417.643349746 [6049-6049] ../src/vabackend.c:2099       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1417.643356101 [6049-6049] ../src/vabackend.c:2125       __vaDriverInit_1_0 Selecting Direct backend
      1417.652494825 [6049-6049] ../src/direct/nv-driver.c: 219            init_nvdriver Initing nvdriver...
      1417.652518082 [6049-6049] ../src/direct/nv-driver.c: 224            init_nvdriver Got dev info: 100 1 2 6
      1417.652788438 [6049-6049] ../src/direct/nv-driver.c: 285            init_nvdriver NVIDIA kernel driver version: 525.78.01
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      1417.828787690 [6049-6049] ../src/vabackend.c:2062              nvTerminate Terminating 0x555dfc00f8e0
      1417.828897550 [6049-6049] ../src/vabackend.c:2076              nvTerminate Now have 0 (0 max) instances
```
Seems like 129 is working but it does terminate at the end

elFarto, 2023-01-14:
```
...it does terminate at the end with 0 instances, i don't know if that is correct
```
That is correct. So the driver is installed and working correctly. 

The issue is, I'm not sure Firefox can be convinced to use it. You can read of issue #11, which covers the biggest problems with Optimus setups.

Arjun31415, 2023-01-15:
Yeah I am unable to make it work with firefox 😢 Will have to switch to chromium and nouveau (to use radeon). 

Thanks to both of you for your time and help

closed 2023-01-15 by elFarto

sergiomb2, 2023-10-14:
Hi, 

NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia nwjs_app 

what does mean `findGPUIndexFromFd No support for EGL_EXT_device_enumeration`  ? 

Thank you 

```
15849.447107182 [32605-32605] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
15849.447125462 [32605-32605] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
15849.447129046 [32605-32605] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
15849.447136683 [32605-32605] ../src/export-buf.c: 127       findGPUIndexFromFd No support for EGL_EXT_device_enumeration
15849.447139672 [32605-32605] ../src/vabackend.c:2231       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
[32605:32605:1014/183939.726559:ERROR:vulkan_swap_chain.cc(404)] vkQueuePresentKHR() failed: -1000001004
[32605:32605:1014/183939.726675:ERROR:gpu_service_impl.cc(1054)] Exiting GPU process because some drivers can't recover from errors. GPU process will restart shortly.
[32572:32572:1014/183939.754690:ERROR:gpu_process_host.cc(960)] GPU process exited unexpectedly: exit_code=8704
15850.606552011 [32701-32701] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
15850.606565060 [32701-32701] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
15850.606568730 [32701-32701] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
15850.606575525 [32701-32701] ../src/export-buf.c: 127       findGPUIndexFromFd No support for EGL_EXT_device_enumeration
15850.606578713 [32701-32701] ../src/vabackend.c:2231       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
```

sergiomb2, 2023-10-14:
the vainfo 

NVD_LOG=1 vainfo
```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
16450.780303271 [33161-33161] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
16450.780308080 [33161-33161] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
16450.780324982 [33161-33161] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
16450.787442142 [33161-33161] ../src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specificCUDA GPU
16450.787452569 [33161-33161] ../src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
16450.791062551 [33161-33161] ../src/export-buf.c: 161       findGPUIndexFromFd Found 4 EGL devices
16450.791138722 [33161-33161] ../src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
16450.791143718 [33161-33161] ../src/export-buf.c: 191       findGPUIndexFromFd Selecting EGLDevice 0
16450.792372490 [33161-33161] ../src/export-buf.c: 260         egl_initExporter Driver supports 16-bit surfaces
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.18 (libva 2.18.2)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
VAProfileMPEG2Simple            : VAEntrypointVLD
VAProfileMPEG2Main              : VAEntrypointVLD
VAProfileVC1Simple              : VAEntrypointVLD
VAProfileVC1Main                : VAEntrypointVLD
VAProfileVC1Advanced            : VAEntrypointVLD
VAProfileH264Main               : VAEntrypointVLD
VAProfileH264High               : VAEntrypointVLD
VAProfileH264ConstrainedBaseline: VAEntrypointVLD
VAProfileHEVCMain               : VAEntrypointVLD
VAProfileVP9Profile0            : VAEntrypointVLD
VAProfileHEVCMain10             : VAEntrypointVLD
VAProfileHEVCMain12             : VAEntrypointVLD
VAProfileVP9Profile2            : VAEntrypointVLD
16450.864398393 [33161-33161] ../src/vabackend.c:2081              nvTerminate Terminating 0x55c4bd00c470
16450.864403030 [33161-33161] ../src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
16450.864413400 [33161-33161] ../src/export-buf.c:  78      egl_releaseExporter Done releasing frames
16450.864418155 [33161-33161] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances ```

elFarto, 2023-10-14:
The EGL backend currently doesn't work, please use `NVD_BACKEND=direct`.

talentlessguy, 2023-11-20:
This doesn't work with `NVD_BACKEND=direct`

```
NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia NVD_BACKEND=direct vainfo --display drm --device /dev/dri/renderD129
Trying display: drm
     11831.206977494 [56551-56551] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 130                     init CUDA ERROR 'unknown error' (999)

     11831.207046812 [56551-56551] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     11831.207052655 [56551-56551] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
     11831.207056705 [56551-56551] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
     11831.216221135 [56551-56551] ../nvidia-vaapi-driver-0.0.11/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
     11831.216231837 [56551-56551] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2200       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

talentlessguy, 2023-11-20:
I'm hitting "unknown error"

```
❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia NVD_BACKEND=direct vainfo --display drm --device /dev/dri/renderD128
Trying display: drm
     12635.386724561 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 130                     init CUDA ERROR 'unknown error' (999)

     12635.386746594 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     12635.386748447 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
     12635.386749931 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
     12635.391743865 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
     12635.391761935 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 545.29.02, major version: 545, minor version: 29
     12635.391764402 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 2 6
     12635.391806107 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  23       findGPUIndexFromFd CUDA ERROR 'initialization error' (3)

     12635.391808335 [63882-63882] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2205       __vaDriverInit_1_0 CUDA ERROR 'initialization error' (3)

libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

spectre10, 2024-04-16:
If anybody is using wayland, you can get it to work on optimus via `NVD_BACKEND=direct LIBVA_DRIVER_NAME=nvidia WLR_DRM_DEVICES=/dev/dri/by-path/{pci-*-card} MOZ_DISABLE_RDD_SANDBOX=1 prime-run firefox`. (pci-*-card, choose the nvidia one) (also assuming you have libva-nvidia-driver installed)

other envs which I had set for running wayland session which might also affect this:
`GBM_BACKEND=nvidia-drm
__GLX_VENDOR_LIBRARY_NAME=nvidia
__GL_VRR_ALLOWED=1
WLR_DRM_NO_ATOMIC=1
__GL_GSYNC_ALLOWED=0
__GL_VRR_ALLOWED=0
`

littleblack111, 2024-10-11:
> on wayland, you can get it to work on optimus via NVD_BACKEND=direct LIBVA_DRIVER_NAME=nvidia WLR_DRM_DEVICES=/dev/dri/by-path/{pci-*-card} MOZ_DISABLE_RDD_SANDBOX=1 prime-run firefox. (pci-*-card, choose the nvidia one) (also assuming you have libva-nvidia-driver installed)

the WLR_DRM_DEVICES doesnt seem to work:
```
╰─$ WLR_DRM_DEVICES=/dev/dri/renderD129 vainfo --display drm                                            ─╯
Trying display: drm
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
-> ERROR: Command Faild to run.

╭─   system@Arch-Linux    ~ ·············································· 3 ✘  11:43:58 AM   ─╮
╰─$ vainfo --display drm --device /dev/dri/renderD129                                                   ─╯
Trying display: drm
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
```

littleblack111, 2024-10-11:
> I'm hitting "unknown error"
>
> ```
> ❯ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia NVD_BACKEND=direct vainfo --display drm --device /dev/dri/renderD128
> Trying display: drm
>      12635.386724561 [63882-63882] [...]

see if u have a /dev/dri/renderD129, maybe that one is ur nvidia gpu if u have a igpu or other gpus