elFarto/nvidia-vaapi-driver #161: Can't build / EGL missing (525-driver)

Michas0206, 2023-01-13:
Hi, in short - 
I don't get it. How do I build nvidia-vaapi-driver with direct-backend - where do I have to set the environment-variable for "direct backend" exactly, or how?

Tried `export NVD_BACKEND=direct `and re-running "meson setup build" did not work, still the same Egl-error:

[image]

Many thanks in advance

elFarto, 2023-01-14:
Both the `egl` and `direct` backend are built at compile time, you don't need to choose between them there. Only when running an application that makes use of VA-API, do you need to choose using the `NVD_BACKEND` environment variable.

The error you're getting is likely because you're missing the EGL development libraries/headers. This is provided by the `libglvnd-devel` package on my Fedora system, not sure what it'll be on your distro.

Michas0206, 2023-01-14:
Many thanks, I'll check this and try it again later today.

Michas0206, 2023-01-14:
Made some steps, build looked successful, but still without success....any ideas what I am missing?

[image]
This setting was made to Grub and also updated Grub+restarted afterward:

[image]

elFarto, 2023-01-14:
Can you run firefox with `NVD_LOG=1 MOZ_LOG="Dmabuf:5,PlatformDecoderModule:5" firefox`? Do you have `MOZ_DISABLE_RDD_SANDBOX` set aswell?

Michas0206, 2023-01-14:
Yes, MOZ_DISABLE_RDD_SANDBOX is 'true' currently.
[image]

Michas0206, 2023-01-14:
Ah, I think I missed the part "In addition the following environment variables need to be set." - I set NVD_BACKEND only.....

Michas0206, 2023-01-14:
Now with 
[image]
I get no error on startup anymore:
[image]

about:support now says
[image]

Is it active now? I'll try to figure out...

Michas0206, 2023-01-14:
Hm, not sure, but blocklisted does not sound as "allowed" or everything is fine....

If I start firefox with your command from above now, it tells:
[image]

elFarto, 2023-01-14:
You've missed the N from `NVD_LOG`. However, I can see that it's failing to open /dev/dri/renderD128, which is possibly a permissions problem. You can try running `NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD128`.

The about:support page is correct for when it's enabled. The wording on that page is confusing, but you're user pref is overriding the blocklist.

Michas0206, 2023-01-14:
[image]

Michas0206, 2023-01-14:
Currently tried `NVD_LOG=1 MOZ_LOG="Dmabuf:5,PlatformDecoderModule:5" firefox` again:
[image]
and Ff is starting.

> The about:support page is correct for when it's enabled. The wording on that page is confusing, but you're user pref is overriding the blocklist.
Many thanks for your efforts, beside the "blocklisted" everything looks fine atm, I'll try around now a bit and play a bit with Firefox...

elFarto, 2023-01-14:
Starting firefox when you already have a firefox window open will just cause that process to exit, so you need to make sure you exit all instances before you start another one from the command line.

Michas0206, 2023-01-14:
Thanks, youre right: 
[image]

closed 2023-03-11 by elFarto